unitreerobotics/xr_teleoperate #3: Problem in running python teleop_hand_and_arm.py

bethalageetachandraraju, 2024-08-25:
Im getting this below error when i run python teleop_hand_and_arm.py.

DDS wrapper is properly installed.

owstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.

silencht, 2024-08-27:
Please use the `cyclonedds ps` and `cyclonedds subscribe rt/lowstate` commands to check if you can receive data from the `rt/lowstate` topic.

A possible reason for this warning is that DDS communication has not been properly created between the robot and the host computer. Maybe you can check the router ip segment configuration, try `ping` the robot pc ip, etc.

bethalageetachandraraju, 2024-08-27:
i tested both cyclonedds ps and cyclonedds subscribe rt/lowstate 
robot i able to communicate.
 

*****************************************
cyclonedds ps 

   0:00:01 ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ Entities discovered: 85
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃ Host       ┃ Application               ┃ Pid  ┃ Participants                         ┃ Topics                                       ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│ unitree-h1 │ basic_service             │ 1061 │ 01103f27-8c1e-03e6-798f-8ab8000001c1 │ rt/lowcmd                                    │
│            │                           │      │                                      │ rt/lowstate                                  │
│            │                           │      │                                      │ rt/lf/lowstate                               │
│ unitree-h1 │ python3                   │ 1090 │ 0110d820-cf74-b2db-18f5-636a000001c1 │ rt/api/bashrunner/response                   │
│            │                           │      │                                      │ rt/api/bashrunner/request                    │
│            │                           │      │                                      │ rt/selftest                                  │
│ unitree-h1 │ h1_loco_server            │ 1104 │ 01103669-4d1b-5000-4f3c-ab83000001c1 │ rt/arm_sdk                                   │
│            │                           │      │                                      │ rt/api/loco/request                          │
│            │                           │      │                                      │ rt/api/loco/response                         │
│            │                           │      │                                      │ rt/loco_sdk                                  │
│ unitree-h1 │ task_social               │ 1144 │ 011099d5-2c3a-26a1-b6b9-d639000001c1 │ rt/loco_sdk                                  │
│            │                           │      │                                      │ rt/lowstate                                  │
│            │                           │      │                                      │ rt/arm_sdk                                   │
│ unitree-h1 │ motion_switcher_service   │ 1265 │ 011044d8-1606-aa04-1731-c4a9000001c1 │ rt/api/motion_switcher/request               │
│            │                           │      │                                      │ rt/api/motion_switcher/response              │
│            │                           │      │                                      │ rt/lf/lowstate                               │
│ unitree-h1 │ python3                   │ 1287 │ 011045cc-0296-a012-3d5e-12b0000001c1 │ rt/public_network_status                     │
│            │                           │      │                                      │ rt/4g_traffic_report                         │
│ unitree-h1 │ robot_state_service       │ 1311 │ 0110e251-b398-ada6-1537-44fe000001c1 │ rt/api/robot_state/request                   │
│            │                           │      │                                      │ rt/api/sport/response                        │
│            │                           │      │                                      │ rt/api/vui/response                          │
│            │                           │      │                                      │ rt/multiplestate                             │
│            │                           │      │                                      │ rt/servicestate                              │
│            │                           │      │                                      │ rt/api/obstacles_avoid/response              │
│            │                           │      │                                      │ rt/public_network_status                     │
│            │                           │      │                                      │ rt/api/sport/request                         │
│            │                           │      │                                      │ rt/api/obstacles_avoid/request               │
│            │                           │      │                                      │ rt/webrtcres                                 │
│            │                           │      │                                      │ rt/config_change_status                      │
│            │                           │      │                                      │ rt/api/uwbswitch/response                    │
│            │                           │      │                                      │ rt/webrtcreq                                 │
│            │                           │      │                                      │ rt/api/uwbswitch/request                     │
│            │                           │      │                                      │ rt/api/config/request                        │
│            │                           │      │                                      │ rt/api/robot_state/response                  │
│            │                           │      │                                      │ rt/api/vui/request                           │
[... cut ...]
/home/airlab_g/miniconda3/envs/tv/lib/python3.8/site-packages/numpy/lib/twodim_base.py:295: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  v = asanyarray(v)
[2024-08-27 14:53:33.239] [SAPIEN] [warning] A second engine will share the same internal structures with the first one. Arguments passed to constructor will be ignored.
Initialize H1ArmController...
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.

bethalageetachandraraju, 2024-08-27:
Yes, It was continuous, its updating and data keep scrolling

silencht, 2024-08-27:
It could be caused by the robot H1's DDS not matching the version of this library.
So, does your output after executing the `cyclonedds typeof rt/lowstate` command is the same as the following output?
```bash
unitree@unitree:~$ cyclonedds typeof rt/lowstate

0:00:01 ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ Entities discovered: 2
As defined in participant(s) 0110e781-a445-2936-f483-45b6000001c1, 011044c3-8840-dbd2-687a-6d0d000001c1
module unitree_hg {                                                                                                                                                                                         
    module msg {                                                                                                                                                                                            
        module dds_ {                                                                                                                                                                                       
            @final                                                                                                                                                                                          
            struct IMUState_ {                                                                                                                                                                              
                float quaternion[4];                                                                                                                                                                        
                float gyroscope[3];                                                                                                                                                                         
                float accelerometer[3];                                                                                                                                                                     
                float rpy[3];                                                                                                                                                                               
                short temperature;                                                                                                                                                                          
            };                                                                                                                                                                                              
            @final                                                                                                                                                                                          
            struct MotorState_ {                                                                                                                                                                            
                octet mode;                                                                                                                                                                                 
                float q;                                                                                                                                                                                    
                float dq;                                                                                                                                                                                   
                float ddq;                                                                                                                                                                                  
                float tau_est;                                                                                                                                                                              
                short temperature[2];                                                                                                                                                                       
                float vol;                                                                                                                                                                                  
                unsigned long sensor[2];                                                                                                                                                                    
                unsigned long motorstate;                                                                                                                                                                   
                unsigned long reserve[4];                                                                                                                                                                   
            };                                                                                                                                                                                              
[... cut ...]
            };                                                                                                                                                                                              
        };                                                                                                                                                                                                  
    };                                                                                                                                                                                                      
};
```

bethalageetachandraraju, 2024-08-27:
Yes, similar to this.
[image]

silencht, 2024-08-27:
Also, close other motion control programs before starting this program.

And this is my `cyclonedds subscribe rt/lowstate` output:
```bash
LowState_(
    version=[0, 0],
    mode_pr=0,
    mode_machine=4,
    tick=274535,
    imu_state=IMUState_(
        quaternion=[0.9984849095344543, -0.0011377931805327535, 0.02254619635641575, 0.050182636827230453],
        gyroscope=[-0.0010652643395587802, -0.010652642697095871, -0.0031957929022610188],
        accelerometer=[-0.2585737705230713, -0.17238251864910126, 9.639056205749512],
        rpy=[-9.292894901591353e-06, 0.04515361040830612, 0.1004328653216362],
        temperature=78
    ),
    motor_state=[
        MotorState_(mode=10, q=-0.017512915655970573, dq=-7.337105489568785e-05, ddq=0.0, tau_est=-0.64453125, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=-0.08713893592357635, dq=0.00015671794244553894, ddq=0.0, tau_est=0.52734375, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=0.03188125416636467, dq=-0.00016748870257288218, ddq=0.0, tau_est=-0.703125, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=0.1788046509027481, dq=-0.000284747191471979, ddq=0.0, tau_est=-0.375, temperature=[29, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=1, q=-0.7946598529815674, dq=0.010806652717292309, ddq=0.0, tau_est=0.0064470842480659485, temperature=[41, 40], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 4, 0]),
        MotorState_(mode=1, q=-0.05474500730633736, dq=0.013466499745845795, ddq=0.0, tau_est=-0.2626223862171173, temperature=[40, 40], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 4, 0]),
        MotorState_(mode=10, q=-0.019302602857351303, dq=0.0003071847604587674, ddq=0.0, tau_est=0.234375, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=-0.07703980058431625, dq=-0.0005694311112165451, ddq=0.0, tau_est=0.1171875, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=0.003655344247817993, dq=0.00035212573129683733, ddq=0.0, tau_est=0.17578125, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=10, q=0.15811192989349365, dq=-0.00018150504911318421, ddq=0.0, tau_est=-0.09375, temperature=[29, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=1, q=-0.6082499027252197, dq=-0.007785136811435223, ddq=0.0, tau_est=-0.0016371047822758555, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 528, 4, 0]),
        MotorState_(mode=1, q=-0.013805490918457508, dq=-0.023477254435420036, ddq=0.0, tau_est=-0.2718762159347534, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 4, 0]),
        MotorState_(mode=10, q=0.23313918709754944, dq=0.00026044860715046525, ddq=0.0, tau_est=0.05859375, temperature=[30, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 529, 2, 0]),
        MotorState_(mode=1, q=0.017922401428222656, dq=-0.0010908307740464807, ddq=0.0, tau_est=-0.263671875, temperature=[41, 40], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=0.02056407928466797, dq=-0.01636246219277382, ddq=0.0, tau_est=-0.263671875, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=0.7142688035964966, dq=0.001714162528514862, ddq=0.0, tau_est=0.1677911877632141, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=1.3286914825439453, dq=-0.003272492438554764, ddq=0.0, tau_est=-0.17578125, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=0.45136356353759766, dq=-0.016394419595599174, ddq=0.0, tau_est=-0.046783626079559326, temperature=[43, 42], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=0.22690916061401367, dq=0.002049302449449897, ddq=0.0, tau_est=-0.046783626079559326, temperature=[43, 41], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 264, 4, 0]),
        MotorState_(mode=1, q=0.3101973533630371, dq=0.002049302449449897, ddq=0.0, tau_est=-0.046783626079559326, temperature=[45, 43], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 265, 4, 0]),
        MotorState_(mode=1, q=-0.008990645408630371, dq=-0.0054541537538170815, ddq=0.0, tau_est=0.263671875, temperature=[42, 41], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 265, 4, 0]),
        MotorState_(mode=1, q=0.005070209503173828, dq=0.013089969754219055, ddq=0.0, tau_est=-0.3515625, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 265, 4, 0]),
        MotorState_(mode=1, q=-0.5894694328308105, dq=0.00857081264257431, ddq=0.0, tau_est=0.22372159361839294, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 265, 4, 0]),
        MotorState_(mode=1, q=1.406294345855713, dq=0.0021816615480929613, ddq=0.0, tau_est=0.3515625, temperature=[40, 39], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 265, 4, 0]),
[... cut ...]
        MotorState_(mode=0, q=0.0, dq=0.0, ddq=0.0, tau_est=0.0, temperature=[0, 0], vol=0.0, sensor=[0, 0], motorstate=0, reserve=[0, 0, 0, 0])
    ],
    wireless_remote=b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00',
    reserve=[0, 0, 0, 0],
    crc=3576080590
)
```

There is a difference in the output, so I think it's the dds (idl) version that makes the difference.
Make sure your robot hardware is H1_2, then ask unitree tech support for more info about dds.

bethalageetachandraraju, 2024-08-28:
We have the H1 robot with inspire hand not h1_2 can you guide us and provide resources for running in H1_inspire robot

bethalageetachandraraju, 2024-08-28:
There is one dds wrapper right, i cant find other versions of this dds.

linClubs, 2024-09-05:
Is it running successfully on H1 robot with inspire?  @bethalageetachandraraju

bethalageetachandraraju, 2024-09-05:
not yet, there are anythings to be addressed
I request unitree to provide support for H1 robot as well for avp_teleop.

silencht, 2024-09-05:
https://github.com/unitreerobotics/avp_teleoperate/issues/4#issuecomment-2331646569

bethalageetachandraraju, 2024-09-25:
How to adjust the dds version, and support team mentioned H1 cant be controlled by python. This is so confusing.
can you please provide a complete solution for actual implementation on H1. The solution you mentioned in the above comment to replace the dds version yourself from unitree sdk2 python, what does this mean.
Please provide a detailed and complete solution please.

This is consuming lot of our time, please provide complete solution.

linClubs, 2024-10-11:
@bethalageetachandraraju I controlled the h1 with ros2 successfully. you can try it.

bethalageetachandraraju, 2024-10-11:
> @bethalageetachandraraju I controlled the h1 with ros2 successfully. you can try it.

can i have your email or linkdin, and are your using the pre loaded ROS2 in H1 or you installed a new one?

silencht, 2024-11-04:
A version using [unitree_sdk2_python](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/low_level/lowlevel_control.py) has been implemented in the [g1 branch](https://github.com/unitreerobotics/avp_teleoperate/tree/g1) for your reference.
The H1 series is still a while to be implemented.

closed 2024-11-04 by silencht