unitreerobotics/xr_teleoperate #4: adjust h1 not h1_2

linClubs, 2024-09-05:
Thanks for open source on robots-h1_2 with Inspire hand, I am trying to use this project to adapt it on the h1 not h1_2. 
1. How should I modify the inverse kinematics of the wrist? Just change the urdf？
2.  [teleop_hand_and_arm.py](https://github.com/unitreerobotics/avp_teleoperate/blob/master/teleop/teleop_hand_and_arm.py) line 154：
`if right_pose is not None and left_qpos is not None:` ， Is right_pose modified to right_qpos?

linClubs, 2024-09-09:
感谢您提供的h1版本，我将h1的逆解代码中做了2点修改：
1. self.reduced_robot.model.addFrame的平移改成0向量，
2. L_tf_target.translation的增量改成只增加y方向， 
然后可视化显示不对，代码如下：
~~~python
 self.reduced_robot.model.addFrame(
            pin.Frame('L_ee',
                      self.reduced_robot.model.getJointId('left_elbow_joint'),
                      pin.SE3(np.eye(3),
                              np.array([0, 0, 0]).T),
                      pin.FrameType.OP_FRAME)
        )
        
        self.reduced_robot.model.addFrame(
            pin.Frame('R_ee',
                      self.reduced_robot.model.getJointId('right_elbow_joint'),
                      pin.SE3(np.eye(3),
                              np.array([0, 0, 0]).T),
                      pin.FrameType.OP_FRAME)
        )
...
L_tf_target.translation += np.array([0.000,  0.001, 0.000])
R_tf_target.translation += np.array([0.000, -0.001, 0.000])
...
~~~

可视化如下：

[image]
rviz显示的L_tf_target轨迹跟机械臂逆解然后运动的轨迹，相差箭头所示的差距，我应该怎么修改这个参数，保证L_tf_target轨迹跟逆解后的left_elbow_joint运动是在同一个原点。是urdf的问题么？

silencht, 2024-09-10:
```c++
self.reduced_robot.model.addFrame(
    pin.Frame('L_ee',
              self.reduced_robot.model.getJointId('left_elbow_joint'),
              pin.SE3(np.eye(3),
                      np.array([0.2605 + 0.05,0,0]).T),
              pin.FrameType.OP_FRAME)
)

self.reduced_robot.model.addFrame(
    pin.Frame('R_ee',
              self.reduced_robot.model.getJointId('right_elbow_joint'),
              pin.SE3(np.eye(3),
                      np.array([0.2605 + 0.05,0,0]).T),
              pin.FrameType.OP_FRAME)
)
```
为什么要去掉平移量呢？
`0.2605 + 0.05` 这个偏移目的是将 end-effector 从 elbow_joint 移动至 wrist_joint 附近。如果去掉该偏移量，那么 ee 直接位于 elbow_joint，此时臂逆解时自由度进一步退化为 3。

linClubs, 2024-09-10:
感谢您答复，已经成功复现了
[image]

closed 2024-09-10 by silencht

bethalageetachandraraju, 2024-09-25:
> Hi,
> 
> 1. We have released a new branch called h1. The following changes were made to this branch:
>    
>    1. imported the URDF files of h1 and modified related something;
>    2. modified inverse kinematics: mainly ignored rotation in SE(3). You can test it with the `python avp_teleoperate/teleop/robot_control/robot_arm_ik.py` command. The code is not commented meshcat visualization.
>       > Be careful: comment out the visualization code when actually controlling the robot! Otherwise visualization may slow down the control frequency
>    3. dds versions for h1 and h1_2 are different. So you have to replace the dds in the h1 branch yourself now. please refer to [unitree_sdk2_python](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/low_level/lowlevel_control.py)
> 2. Yes, thanks for pointing out the bugs!

Please provide a detailed solution for dds version update for H1.

NirajPudasaini, 2024-09-30:
@silencht Can you follow up on releasing dds version release for H1?  Your 1) iii is unclear

qc-1, 2025-12-24:
> 感谢您提供的h1版本，我将h1的逆解代码中做了2点修改：
> 
>     1. self.reduced_robot.model.addFrame的平移改成0向量，
> 
>     2. L_tf_target.translation的增量改成只增加y方向，
>        然后可视化显示不对，代码如下：
> 
> 
>  self.reduced_robot.model.addFrame(
>             pin.Frame('L_ee',
>                       self.reduced_robot.model.getJointId('left_elbow_joint'),
>                       pin.SE3(np.eye(3),
>                               np.array([0, 0, 0]).T),
>                       pin.FrameType.OP_FRAME)
>         )
>         
>         self.reduced_robot.model.addFrame(
>             pin.Frame('R_ee',
>                       self.reduced_robot.model.getJointId('right_elbow_joint'),
>                       pin.SE3(np.eye(3),
>                               np.array([0, 0, 0]).T),
>                       pin.FrameType.OP_FRAME)
>         )
> ...
> L_tf_target.translation += np.array([0.000,  0.001, 0.000])
> R_tf_target.translation += np.array([0.000, -0.001, 0.000])
> ...
> 
> 可视化如下：
> 
> [image] rviz显示的L_tf_target轨迹跟机械臂逆解然后运动的轨迹，相差箭头所示的差距，我应该怎么修改这个参数，保证L_tf_target轨迹跟逆解后的left_elbow_joint运动是在同一个原点。是urdf的问题么？

请问您这里提到的机械臂逆解然后运动的轨迹是指真机移动的轨迹吗，还是只是说将机械臂逆解运动的轨迹跑在了 rviz 里面