unitreerobotics/xr_teleoperate #5: some problems in running teleop_hand.py

lmj250, 2024-09-12:
I can run teleop_hand.py smoothly but I can't open the 'localhost:8012' in the brower of Quest3. 

I found that in the command printed after run teleop_hand.py, there is no 'Websocket is connected. id:'  'default socket worker is up ,adding clientEvents' 'Uplink task running. id:'

It seems that WebSocket did not start successfully.

Thanks for your work, want to know how to solve this problem can enter VR in the browser of Quest3.

lmj250, 2024-09-12:
Simply put, the socket connection is not successful, this is the need to make some changes to the original code?

silencht, 2024-09-13:
hi, we have not tested Meta Quest device with this repo. 
you might could refer [OpenTeleVision](https://github.com/OpenTeleVision/TeleVision)'s related issues, such as https://github.com/OpenTeleVision/TeleVision/issues/12.

closed 2024-09-24 by le-wei