unitreerobotics/xr_teleoperate #11: Problem whe trying to run teleop_hand_and_arm.py

Kshitij-Sky, 2024-11-12:
Hi I'm getting a strange problem while running the code. I can move the robot and run examples using the sdk2 on the H1. Could you please tell me what the problem is and how I can solve it?

`python teleop_hand_and_arm.py 
Serving file:///home/robot/humanoid-remote-control/avp_teleoperate/teleop at /static
Visit: https://vuer.ai?grid=False
Process Process-2:
Traceback (most recent call last):
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/robot/humanoid-remote-control/avp_teleoperate/teleop/TeleVision.py", line 78, in run
    self.app.run()
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/site-packages/vuer/server.py", line 766, in run
    super().run()
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/site-packages/vuer/base.py", line 136, in run
    event_loop.run_until_complete(init_server())
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/robot/miniconda3/envs/tv/lib/python3.8/site-packages/vuer/base.py", line 124, in init_server
    ssl_context.load_cert_chain(certfile=self.cert, keyfile=self.key)
FileNotFoundError: [Errno 2] No such file or directory
/home/robot/miniconda3/envs/tv/lib/python3.8/site-packages/numpy/lib/twodim_base.py:295: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  v = asanyarray(v)
[2024-11-12 22:22:02.989] [SAPIEN] [warning] A second engine will share the same internal structures with the first one. Arguments passed to constructor will be ignored.
Initialize H1ArmController...
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.
lowstate_subscriber is not ok! Please check dds.`

silencht, 2024-11-13:
Hi, currently only recommended to refer h1 branch for inverse kinematics [related code](https://github.com/unitreerobotics/avp_teleoperate/blob/h1/teleop/robot_control/robot_arm_ik.py). 

The dds under this branch is the stale version and currently requires the user to modify the dds configuration under the [control](https://github.com/unitreerobotics/avp_teleoperate/blob/h1/teleop/robot_control/robot_arm.py) [files](https://github.com/unitreerobotics/avp_teleoperate/blob/h1/teleop/robot_control/robot_hand.py) themselves. That is, replace the temporary [unitree_dds_wrapper](https://github.com/unitreerobotics/unitree_dds_wrapper) with the official version [unitree_sdk2_python](https://github.com/unitreerobotics/unitree_sdk2_python). The dds example code for the H1 robot you can find [here](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/h1/low_level/h1_low_level_example.py).

p.s. Because H1_2 and G1 robots use almost the same dds configuration. So you can learn from the dds changes by comparing the contents of control files for the [H1_2](https://github.com/unitreerobotics/avp_teleoperate/blob/h1_2/teleop/robot_control/robot_arm.py) and [G1](https://github.com/unitreerobotics/avp_teleoperate/blob/g1/teleop/robot_control/robot_arm.py) branches. (*Currently the H1_2 branch uses unitree_dds_wrapper, while the G1 branch uses the latest unitree_sdk2_python.*)

KReguieg, 2024-11-18:
Hi,

we have the same problem. We tried solving it by replacing the unitree_dds_wrapper with unitree_python_sdk2, but it did not work. We need further assistance with that. Maybe easier, smaller examples. When will there be updates on the h1 and h1_2 branch with working examples.
So far we only made the dexterous hands example work by default. At this point I'm not even sure if we received the h1 or the h1_2. How can we tell, is there a config file on the robot pc that we can ask?

Thanks in advance.

silencht, 2024-11-19:
> At this point I'm not even sure if we received the h1 or the h1_2. How can we tell, is there a config file on the robot pc that we can ask?

On this [H1_developer Webpage](https://support.unitree.com/home/en/H1_developer), there are introductions to H1 and H1_2, which you can read to distinguish your robot version. You can make a simple distinction by the freedom of the robot's arms (H1: 4DoF, H1_2: 7DoF).

> We tried solving it by replacing the unitree_dds_wrapper with unitree_python_sdk2, but it did not work. We need further assistance with that. 

Once you have verified the robot version, you can test the dds communication. First, you can start by learning the [example code](https://github.com/unitreerobotics/unitree_sdk2_python/tree/master/example) for the [unitree_sdk2_python](https://github.com/unitreerobotics/unitree_sdk2_python) library. If your robot version is H1, you can use this [example](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/h1/low_level/h1_low_level_example.py) to test the subscription and publication of dds messages. The same applies to the H1_2 [example](https://github.com/unitreerobotics/unitree_sdk2_python/blob/master/example/g1/low_level/g1_ankle_swing_example.py). 

When you successfully execute and review the example code, you will see: To control the robot using dds, several steps are required.

1. According to the robot version, [import the header files needed by dds](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L4)
2. [Initialize the dds environment](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L191)
3. [Initialize some necessary parameters](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L77), such as kp,kd,mode_machine_, etc.
4. [Initialize subscribers and publishers](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L89) according to topic name and message type.
5. [Subscription thread](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L118) constantly subscribe to status messages.
6. [Publishing thread](https://github.com/unitreerobotics/unitree_sdk2_python/blob/c52f48b0b55f29dbc6ed3f67303bbd25732dc069/example/g1/low_level/g1_ankle_swing_example.py#L130C9-L130C22) assembles the msg control message (low_cmd) and then publishes it.

The python version is very similar to the c++ version, and you can refer to the [dds documentation for the c++ version](https://support.unitree.com/home/en/H1_developer/Basic_Services_Interface) as well. If you still have questions about using the above dds example scripts, please contact our technical support.

PranavDhulipala, 2024-11-19:
We are facing an issue with the arm inverse kinematics implementation for the H1 robot equipped with the Inspire hands. We had to make significant modifications to the codebase, including adapting the URDF due to relative path issues. The following is the initialization code for the robot model:

self.robot = pin.RobotWrapper.BuildFromURDF(
    '../assets/h1_description/urdf/h1_with_hand.urdf',
    '../assets/h1_description/urdf'
)

The setup worked up to a certain point, but now we encounter the following runtime error:

Traceback (most recent call last):
  File "teleop_hand_and_arm.py", line 143, in <module>
    sol_q, tau_ff, flag = arm_ik.ik_fun(left_pose, right_pose, armstate, armv)
  File "/root/avp_teleoperate/teleop/robot_control/robot_arm_ik.py", line 203, in ik_fun
    self.opti.set_initial(self.var_q, self.init_data)
  File "/opt/miniconda/envs/tv/lib/python3.8/site-packages/casadi/casadi.py", line 47835, in set_initial
    return _casadi.Opti_set_initial(self, *args)
RuntimeError: Error in Opti::set_initial [OptiNode] at .../casadi/core/optistack.cpp:125:
.../casadi/core/matrix_impl.hpp:252: Dimension mismatch. lhs is 8-by-1, while rhs is [14,1]

The error appears to stem from a dimension mismatch between self.var_q (8-by-1) and self.init_data (14-by-1). This likely occurs because the reduced robot model we defined has fewer degrees of freedom (DOFs) than the original robot. Specifically:

self.var_q corresponds to the active DOFs in the reduced model.
self.init_data seems to include states for both active and locked joints, causing a mismatch.

Could you please provide guidance on:

How to handle this dimension mismatch effectively? Should we explicitly map motorstate to the reduced model's DOFs?
Whether there is a more standardized way to adapt the URDF and initialize the model paths for the H1 robot with Inspire hands and are we even using the correct URDF in the first place ?

silencht, 2024-11-22:
> Whether there is a more standardized way to adapt the URDF and initialize the model paths for the H1 robot with Inspire hands and are we even using the correct URDF in the first place ?

The control file is currently used for two ways, one is for unit testing alone and the other is for importing teleop_hand_and_arm as part of the main program.
In h1 branch, we modify the URDF path selection for different uses by manual comments:
```python
# for teleop_hand_and_arm.py use
self.robot = pin.RobotWrapper.BuildFromURDF('../assets/h1_description/urdf/h1_with_hand.urdf', '../assets/h1_description/')
# for unit test
self.robot = pin.RobotWrapper.BuildFromURDF('../../assets/h1_description/urdf/h1_with_hand.urdf', '../../assets/h1_description/') 
```
In g1 branch, we adapt to URDF paths by initializing parameters with classes:
```python
class G1_29_ArmIK:
    def __init__(self, Unit_Test = False, Visualization = False):
        self.Unit_Test = Unit_Test
        if not self.Unit_Test:
            self.robot = pin.RobotWrapper.BuildFromURDF('../assets/g1/g1_body29_hand14.urdf', '../assets/g1/')
        else: # for unit test
            self.robot = pin.RobotWrapper.BuildFromURDF('../../assets/g1/g1_body29_hand14.urdf', '../../assets/g1/')
```
If your URDF path is wrong, then the pin.RobotWrapper.BuildFromURDF code should throw an error. 
If you have a better way you are welcome to discuss it :)

> How to handle this dimension mismatch effectively? Should we explicitly map motorstate to the reduced model's DOFs?

In your error informations,
```bash
Traceback (most recent call last):
File "teleop_hand_and_arm.py", line 143, in
sol_q, tau_ff, flag = arm_ik.ik_fun(left_pose, right_pose, armstate, armv)
File "/root/avp_teleoperate/teleop/robot_control/robot_arm_ik.py", line 203, in ik_fun
self.opti.set_initial(self.var_q, self.init_data)
File "/opt/miniconda/envs/tv/lib/python3.8/site-packages/casadi/casadi.py", line 47835, in set_initial
return _casadi.Opti_set_initial(self, *args)
RuntimeError: Error in Opti::set_initial [OptiNode] at .../casadi/core/optistack.cpp:125:
.../casadi/core/matrix_impl.hpp:252: Dimension mismatch. lhs is 8-by-1, while rhs is [14,1]
```
The error message indicates that `self.var_q` is dimension 8 and `self.init_data` is dimension 14, so it doesn't match.
For the h1 robot (single arm with 4 DoF), so the optimizing variable `self.var_q` is dimensionally correct. 
`self.init_data` is initialized [here](https://github.com/unitreerobotics/avp_teleoperate/blob/c337ef2b773f9fc04fbc44284f745ba05fa4c146/teleop/robot_control/robot_arm_ik.py#L93C9-L93C63), and theoretically the dimension here should also be the correct value 8.

**The problem lies in the dimension of the `motorstate` parameter passed to `ik_fun`, as it overrides `self.init_data`. In other words, you passed the wrong arm motor state variable `armstate` when you executed [here](https://github.com/unitreerobotics/avp_teleoperate/blob/c337ef2b773f9fc04fbc44284f745ba05fa4c146/teleop/teleop_hand_and_arm.py#L143) in teleop_hand_and_arm.py.**

On h1 branch, for time reasons, we only adapted the [robot_arm_ik.py](https://github.com/unitreerobotics/avp_teleoperate/blob/h1/teleop/robot_control/robot_arm_ik.py) inverse kinematics file, not the [robot_arm.py](https://github.com/unitreerobotics/avp_teleoperate/blob/h1/teleop/robot_control/robot_arm.py) control file and the others. This might be the problem: you passed the 7×2 DoF arm state of H1_2. (Since H1 was forked from the H1_2 branch.)

What you need to do is update the `robot_arm.py` file to make it compatible with H1. Then, in the `teleop_hand_and_arm.py` program, obtain the motor state values of the 8-DoF dual arms and pass them to `ik_fun`. Finally, publish the motor state results returned by `ik_fun` (`sol_q`) to H1 for control. The clearest functional flow of the code so far is available in the [g1 branch](https://github.com/unitreerobotics/avp_teleoperate/blob/g1/teleop/teleop_hand_and_arm.py), which you can refer to.

If you still have questions about using the above dds example scripts, please contact our technical support.

jluuser, 2024-11-22:
我之前也碰到过这个dds问题，我把unitree_hg这个库改为unitree_go这个库然后解决了

silencht, 2024-11-22:
> 我之前也碰到过这个dds问题，我把unitree_hg这个库改为unitree_go这个库然后解决了

是的，在这里重申一下：
unitree_hg是H1_2和G1机器人的消息类型（即最新的两款，最高配手臂自由度为7，电机数量保留至35个）
unitree_go是Go2和H1等机器人的消息类型（电机数量为20个）

closed 2024-12-05 by silencht