unitreerobotics/xr_teleoperate #14: Is there any error in avp_teleoperate-g1/teleop/robot_control/robot_arm_ik.py?

HomerIsAFool, 2024-11-18:
[image]
计算逆你运动学的solve_ik(self, left_wrist, right_wrist, current_lr_arm_motor_q = None, current_lr_arm_motor_dq = None)这一方法，返回值是否有问题？注释中表明了“return sol_q, sol_tauff”，但实际代码中返回的却是左臂当前的关节角度？（259行）
求各位指点。

HomerIsAFool, 2024-11-18:
[image]
同级目录下的robot_arm.py中调用这个方法时，返回的也是求解的q和tauff（334行）
[image]

silencht, 2024-11-18:
[CasADi Docs](https://web.casadi.org/docs/#extras)文档9.3节:
> 9.3. Extras
It may well happen that the solver does not find an optimal solution. In such cases, you may still access the non-converged solution through debug mode:
`opti.debug.value(x)`

我的理解是，当优化问题未收敛时，可以通过 debug mode 取得优化失败后最后一次的迭代结果值。但是这个值的可靠性可能是有限的：可能违反问题的约束条件、可能远离可行解区域、在某些情况下可能是数值不稳定的结果等等。
所以这里最终还是采取了：一旦本次优化问题未收敛，那么直接返回当前电机状态值（即维持当前状态不变）的策略。

如果您有更好的看法，欢迎讨论。

HomerIsAFool, 2024-11-18:
I got it, it sounds reansonable. 
Thanks a lot! :D

closed 2024-11-19 by silencht