unitreerobotics/xr_teleoperate #15: G1 Retargeting not working correctly in Isaac Gym

pk-scafo, 2024-11-19:
Hi, I'm running `teleop/teleop_test_gym.py` and able to get the simulation stream in AVP. However, the hand is not moving as expected. 

https://github.com/unitreerobotics/avp_teleoperate/blob/g1/teleop/teleop_test_gym.py

silencht, 2024-11-20:
The current isaac gym test really doesn't do what is expected. As an alternative, you can choose isaac gym under the h1_2 branch for testing. We will fix the issue as soon as possible. 
Thanks for your question :)

silencht, 2024-12-24:
Currently, **The 2.3 Test Environment** section in the `g1` branch is only used to verify the correctness of the environment installation, which may seem a bit excessive. There are indeed issues with this section, and we need some time to address the related content and code. Thank you for your attention!