unitreerobotics/xr_teleoperate #23: AVP teleoperation for g1

Baojch, 2024-12-19:
The  **2.3 Test Environment** section in the g1 branch seems to have some issues due to function modification. I suspect it might be related to the recent update. Could you please help review and address these? Thank you!

Baojch, 2024-12-20:
In order to run the test file, I add false in teleop_test_gym.py:219
`tv_wrapper = TeleVisionWrapper(False, img_shape, img_shm.name)`
and put ` ChannelFactoryInitialize(0)` out of `if-else` in teleop/robot_control/robot_hand_unitree.py.

But still has "[Dex3_1_Controller] Waiting to subscribe dds". Whether it means that I should connect to the real G1 robot?

silencht, 2024-12-24:
This issue is a duplicate of #15. Please follow and track progress there.

closed 2024-12-24 by silencht