unitreerobotics/xr_teleoperate #28: Inspire hands on Nvidia board

cixliv, 2025-02-04:
Hey there, been trying to get the Inspire hands on the H1-2 to work properly on an Nvidia NX board.

Serial to USB connection is connected to the correct connections for the Inspire Hands.
Verified the hands are receiving commands (the opening and closing example is working)
The wrist and arm data is coming in from the Vision Pro.
But the Inspire hand data from the Vision Pro does seem to be transferring or retargeted to the Inspire Hands. Any ideas?

endic-sam928281, 2025-02-04:
Hello, we tried to solve the issue.

This is what we did:

Implement Inspire hand controller and update the main loop to handle Inspire hands data from Vision Pro

You can review changes in this commit: https://github.com/endic-sam928281/unitreerobotics-avp_teleoperate-28/commit/8af04468afa72a9fa1b9a988f598dc6813681be9.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

cixliv, 2025-02-04:
> Hello, we tried to solve the issue.
> 
> This is what we did:
> 
> Implement Inspire hand controller and update the main loop to handle Inspire hands data from Vision Pro
> 
> You can review changes in this commit: [endic-sam928281@8af0446](https://github.com/endic-sam928281/unitreerobotics-avp_teleoperate-28/commit/8af04468afa72a9fa1b9a988f598dc6813681be9).
> 
> Caution
> 
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.
> 
> Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

This solution does not work for an Nvidia board. The retargeting only works on the Intel board.

silencht, 2025-02-06:
same as https://github.com/unitreerobotics/avp_teleoperate/issues/30

closed 2025-02-06 by silencht