unitreerobotics/xr_teleoperate #29: Error while running teleop_hand_and_arm.py regarding URDF.

Kshitij-Sky, 2025-02-05:
`python teleop_hand_and_arm.py 
Serving file:///home/robot/humanoid-remote-control/humanoid-remote-control/avp_teleoperate/teleop at /static
Visit: https://vuer.ai?grid=False
Traceback (most recent call last):
  File "teleop_hand_and_arm.py", line 140, in <module>
    teleoperator = VuerTeleop('inspire_hand.yml')
  File "teleop_hand_and_arm.py", line 115, in __init__
    self.left_retargeting = left_retargeting_config.build()
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/dex_retargeting/retargeting_config.py", line 139, in build
    robot_urdf = urdf.URDF.load(self.urdf_path)
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/dex_retargeting/yourdfpy.py", line 950, in load
    return URDF(robot=URDF._parse_robot(xml_element=xml_root), **kwargs)
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/dex_retargeting/yourdfpy.py", line 606, in __init__
    self._scene = self._create_scene(
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/dex_retargeting/yourdfpy.py", line 1248, in _create_scene
    matrix, _ = self._forward_kinematics_joint(j)
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/dex_retargeting/yourdfpy.py", line 1018, in _forward_kinematics_joint
    matrix = origin @ tra.rotation_matrix(q, joint.axis)
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/trimesh/transformations.py", line 373, in rotation_matrix
    M = np.diag([cosa, cosa, cosa, 1.0])
  File "<__array_function__ internals>", line 200, in diag
  File "/home/robot/anaconda3/envs/unitree_tv/lib/python3.8/site-packages/numpy/lib/twodim_base.py", line 295, in diag
    v = asanyarray(v)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (4,) + inhomogeneous part.
(unitree_tv) robot@AutodiscoveryPC001:~/humanoid-remote-control/humanoid-remote-control/avp_teleoperate/teleop$ /home/robot/anaconda3/envs/unitree_tv/lib/python3.8/multiprocessing/resource_tracker.py:216: UserWarning: resource_tracker: There appear to be 1 leaked shared_memory objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '
`

It seems there is an issue while building the urdf for the hands using the retargeting system. Could you please provide a solution?

silencht, 2025-02-06:
Please tell me which commit you are currently on in the repository. 

The `urdf_path` in the `inspire_hand.yml` file specifies the relative location of the URDF, and it should normally be correct.

Kshitij-Sky, 2025-02-06:
I'm in the latest commit for the h1 branch of the repo.

kisdia, 2025-02-07:
Correct path of urdf_path in the inspire_hand.yml file specifies the relative location of the URDF fixes this issue.

closed 2025-02-07 by Kshitij-Sky