unitreerobotics/xr_teleoperate #30: Getting the project to work for an Nvidia board for H1-2

cixliv, 2025-02-06:
Any plans on how to get the following to work on an Nvidia board? Been able to get it all to work so far until the Inspire hands on the H1-2.

silencht, 2025-02-06:
The older versions of the *dex-retargeting* library (like the 0.1.1 version used in this project) depend on the *SPAIEN* library to load and visualize URDF models. However, as you’ve noticed, *SPAIEN* doesn’t support ARM architecture. We tried compiling it from source on an ARM platform but couldn’t get it to work.  

Right now, the best option is to upgrade dex-retargeting to the latest version. But that brings a new problem: the new version uses `yourdfpy.py` to load URDF models, and its joint order seems different from *SPAIEN*’s, which can mess up the retargeting. We do plan to upgrade the *dex-retargeting* library, but we’re not sure when it’ll happen.

yzqin, 2025-02-06:
hi @silencht 

I am the developer for the `dex-retargeting` repo. The old version use SAPIEN for only urdf loading and kinematics computation, which is an overkill for SAPIEN as a simulator. That is why we change the computation backend from SAPIEN to Pinocchio, which comes with a native arch64 Python build and can be used on Jetson.

To solve the joint order issue you mentioned, the best way is to build a joint remapping on the fly, similar as [this one](https://github.com/dexsuite/dex-retargeting/blob/8632b2cab32e1b51ce379940c414a0f78332ff6b/example/vector_retargeting/render_robot_hand.py#L125). 

URDF (xml) is a tree structure, it does not have a convention for ordering. So it is always better to get the joint order based on joint names on the fly.

cixliv, 2025-02-07:
thanks @yzqin how you would recommend we implement the new joint mapping script? the one referenced is using sapien as well. we are specifically trying to get the inspire hands to remap to the new version of dex retargeting.

cixliv, 2025-02-07:
could you explain in a little more detail what you mean by building a joint remapping on the fly? what files would you modify in the Unitree project to adopt this new mapping? or would you put in a modified version of the file you provided?

I believe these are the files that handle the retargeting teleop/robot_control/robot_hand.py

or more specifically the h1-2 branch https://github.com/unitreerobotics/avp_teleoperate/blob/58beecac83f6536b0e6a5df6677727eb35a3d5b6/teleop/robot_control/robot_hand.py

yzqin, 2025-02-07:
hi @cixliv 

Yeah, the example I mentioned convert the joint order from `dex_retargeting` to `sapien`, which is used as a visualizer. 

On the real inspire robot, you can also do the same thing to convert the joint order from  `dex_retargeting` to `inspire hardware api`. Something like that:

```python
sapien_joint_names = [joint.get_name() for joint in robot.get_active_joints()]
inspire_api_joint_names = ['pinky_proximal_joint', 'ring_proximal_joint', 'middle_proximal_joint', 'index_proximal_joint',
                        'thumb_proximal_pitch_joint', 'thumb_proximal_yaw_joint']
retargeting_to_hardware = np.array([retargeting_joint_names.index(name) for name in inspire_api_joint_names]).astype(int)
```

You can change the `inspire_api_joint_names` based on the hardware API you are using. Maybe @silencht can provide some insight on the hardware side.

yzqin, 2025-02-07:
The reason to build a joint remapping on the fly is to solve the issue that different software use different joint orders, the software includes: dex-retargeting, unitree SDK or inspire hardware SDK, and sapien (if you are using simulator not real robot)

There are many ways to modify the current code to apply the new joint mapping. For example, you can change the control code [here](https://github.com/unitreerobotics/avp_teleoperate/blob/6a667c6057ecad34a348c1848f8cb6f392cddd3c/teleop/robot_control/robot_hand_inspire.py#L35) to apply this joint reordering or something else.

silencht, 2025-02-08:
Hi! Many many thanks for @yzqin 's support! 

Hi, @cixliv , The joint order in the IDL of the [Inspire Dexterous Hand documentation](https://support.unitree.com/home/en/H1_developer/Dexterous_hand) is shown in the table below.
- Right Hand

| Id    | 0     | 1    | 2      | 3     | 4          | 5              |
| ----- | ----- | ---- | ------ | ----- | ---------- | -------------- |
| Joint | pinky | ring | middle | index | thumb-bend | thumb-rotation |

- Left Hand

| Id    | 6     | 7    | 8      | 9     | 10         | 11             |
| ----- | ----- | ---- | ------ | ----- | ---------- | -------------- |
| Joint | pinky | ring | middle | index | thumb-bend | thumb-rotation |

The above describes the specific meanings of the 12 values in the control command publishing code: https://github.com/unitreerobotics/avp_teleoperate/blob/58beecac83f6536b0e6a5df6677727eb35a3d5b6/teleop/robot_control/robot_hand.py#L46-L51

------
@cixliv , after upgrading **dex-retargeting library**, you might consider adjusting the following code
https://github.com/unitreerobotics/avp_teleoperate/blob/58beecac83f6536b0e6a5df6677727eb35a3d5b6/teleop/teleop_hand_and_arm.py#L102-L114
to look like

```python
# So, MAYBE this code adjustment will work.
def step(self):
     head_mat, left_wrist_mat, right_wrist_mat, left_hand_mat, right_hand_mat = self.processor.process(self.tv)
     head_rmat = head_mat[:3, :3]

     left_wrist_mat[2, 3] +=0.45
     right_wrist_mat[2,3] +=0.45
     left_wrist_mat[0, 3] +=0.20
     right_wrist_mat[0,3] +=0.20

     left_qpos = self.left_retargeting.retarget(left_hand_mat[tip_indices])[[ 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]]
     right_qpos = self.right_retargeting.retarget(right_hand_mat[tip_indices])[[ 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11]]

     return head_rmat, left_wrist_mat, right_wrist_mat, left_qpos, right_qpos
```
**This is my test code below.**

```python
class VuerTeleop:
    def __init__(self, config_file_path):
        #......
        #......
        self.left_retargeting = left_retargeting_config.build()
        self.right_retargeting = right_retargeting_config.build()
        # LEFT
        # switch to OLD DEX-RETARGETING LIBRARY(0.1.1) output
        print([joint.get_name() for joint in self.left_retargeting.optimizer.robot.get_active_joints()])
        # ['L_thumb_proximal_yaw_joint', 'L_thumb_proximal_pitch_joint', 'L_thumb_intermediate_joint', 'L_thumb_distal_joint', 'L_index_proximal_joint', 'L_index_intermediate_joint', 'L_middle_proximal_joint', 'L_middle_intermediate_joint', 'L_ring_proximal_joint', 'L_ring_intermediate_joint', 'L_pinky_proximal_joint', 'L_pinky_intermediate_joint']
        
        # switch to NEW DEX-RETARGETING LIBRARY(0.4.6) output
        print(self.left_retargeting.joint_names)
        # ['L_index_proximal_joint', 'L_index_intermediate_joint', 'L_middle_proximal_joint', 'L_middle_intermediate_joint','L_pinky_proximal_joint', 'L_pinky_intermediate_joint', 'L_ring_proximal_joint', 'L_ring_intermediate_joint', 'L_thumb_proximal_yaw_joint', 'L_thumb_proximal_pitch_joint', 'L_thumb_intermediate_joint', 'L_thumb_distal_joint']
        
        # THIS is hardware dds idl command order
        self.left_inspire_api_joint_names = ['L_pinky_proximal_joint', 'L_ring_proximal_joint', 'L_middle_proximal_joint', 'L_index_proximal_joint', 'L_thumb_proximal_pitch_joint', 'L_thumb_proximal_yaw_joint']
        
        
        # RIGHT
        # OLD DEX-RETARGETING LIBRARY(0.1.1) output
        print([joint.get_name() for joint in self.right_retargeting.optimizer.robot.get_active_joints()])
        # ['R_thumb_proximal_yaw_joint', 'R_thumb_proximal_pitch_joint', 'R_thumb_intermediate_joint', 'R_thumb_distal_joint', 'R_index_proximal_joint', 'R_index_intermediate_joint', 'R_middle_proximal_joint', 'R_middle_intermediate_joint', 'R_ring_proximal_joint', 'R_ring_intermediate_joint', 'R_pinky_proximal_joint', 'R_pinky_intermediate_joint']
        
        # NEW DEX-RETARGETING LIBRARY(0.4.6) output
        print(self.right_retargeting.joint_names)
        # ['R_index_proximal_joint', 'R_index_intermediate_joint', 'R_middle_proximal_joint', 'R_middle_intermediate_joint', 'R_pinky_proximal_joint', 'R_pinky_intermediate_joint', 'R_ring_proximal_joint', 'R_ring_intermediate_joint', 'R_thumb_proximal_yaw_joint', 'R_thumb_proximal_pitch_joint', 'R_thumb_intermediate_joint', 'R_thumb_distal_joint']
		
        # THIS is hardware dds idl command order
[... 23 lines cut ...]

  # A smaller alpha means stronger filtering, i.e. more smooth but also larger latency
  low_pass_alpha: 0.8

right:
  type: vector
  urdf_path: inspire_hand/inspire_hand_right.urdf
  wrist_link_name: "R_hand_base_link"

  # Target refers to the retargeting target, which is the robot hand
  target_joint_names: ['R_thumb_proximal_yaw_joint', 'R_thumb_proximal_pitch_joint',
                       'R_index_proximal_joint', 'R_middle_proximal_joint', 
                       'R_ring_proximal_joint', 'R_pinky_proximal_joint', 
                       ]
  target_origin_link_names: [ "R_hand_base_link", "R_hand_base_link", "R_hand_base_link", "R_hand_base_link", "R_hand_base_link" ]
  target_task_link_names: [ "R_thumb_tip", "R_index_tip", "R_middle_tip", "R_ring_tip", "R_pinky_tip" ]
  scaling_factor: 1.5

  # Source refers to the retargeting input, which usually corresponds to the human hand
  # The joint indices of human hand joint which corresponds to each link in the target_link_names
  target_link_human_indices: [ [ 0, 0, 0, 0, 0 ], [ 4, 9, 14, 19, 24 ] ]

  # A smaller alpha means stronger filtering, i.e. more smooth but also larger latency
  low_pass_alpha: 0.8
```

silencht, 2025-02-08:
@cixliv 
Since I don't have the available Inspire Hand hardware for immediate use, I haven't actually tested the above program. If you have any further questions, please feel free to continue the communication under this issue. We will do our best to assist you.

cixliv, 2025-02-08:
hey @silencht and @yzqin thank you guys so much! I will try it out and get back to you. We got a basic version of it to work with very manual tweeting, but it seems a bit high latency and the thumb isn't working properly.

Thank you both for this I'll test it out and get back to you!

silencht, 2025-02-08:
In our [latest two commits](https://github.com/unitreerobotics/avp_teleoperate/commit/f41272134de08d8c23476824fca173d3b2f91707) on main branch, we have completed the upgrade of **dex-retargeting** library and the adaptation for Unitree DEX3-1. You may find these updates useful for reference. The adaptation for the Inspire Hand will require some waiting time.

Correll-Humanoid, 2025-04-02:
Thank you so much!

This does correctly map the joints for later versions, however the provided yaml does not close the joints all the way, I believe setting the scaling factor to 1 solves this.

Sometimes the pinky gets stuck down but I think that is just a sensing issue in the AVP, not a config issue.

closed 2025-04-25 by silencht