unitreerobotics/xr_teleoperate #32: teleop based on Quest 3/3S

wzds2015, 2025-02-14:
Is there such a feature to apply? Thanks!

silencht, 2025-02-14:
We have tried using hand tracking for teleoperation on the Meta-Quest 3 
(its system specifications are:
System version: 49829370066100510
Version: 62.0.0.273.343.570372087
Runtime version: 62.0.0.269.341.570372063
OS version: SQ3A.220605.009.A1).

Apart from being slightly less accurate compared to the Apple Vision Pro, there is no significant difference.

Initially, I followed the instructions [in this article](https://medium.com/@lazerwalker/how-to-easily-test-your-webvr-and-webxr-projects-locally-on-your-oculus-quest-eec26a03b7ee): using the ADB tool for port forwarding so that the Quest's browser could connect with the [Vuer server](https://github.com/vuer-ai/vuer) within avp_teleoperate. Later, I discovered that it’s possible to directly ignore the website’s warnings and forcibly log in to the URL: `https://your_avp_teleoperate_running_device_ip:8012/?ws=wss://your_avp_teleoperate_running_device_ip:8012` if you don’t mind the so-called security risks.

In the future, we plan to upgrade the [Vuer library](https://github.com/vuer-ai/vuer)  to support data from [Quest controllers](https://github.com/vuer-ai/vuer/issues/39), but there is no specific timeline for this yet.

silencht, 2025-02-14:
These are some of my notes on ADB reverse port forwarding for Quest. I hope they can be of some help to you.

1. Download [ADB tools](https://developer.android.com/tools/releases/platform-tools) and extract them.

```bash
(base) unitree@unitree:~/Downloads/platform-tools-latest-linux/platform-tools$ ls
adb  etc1tool  fastboot  hprof-conv  lib64  make_f2fs  make_f2fs_casefold  mke2fs  mke2fs.conf  NOTICE.txt  source.properties  sqlite3
```
2. Connect the XR device to the PC via USB and execute the following command:
```bash
$ sudo ./adb devices
[sudo] password for unitree: 
* daemon not running; starting now at tcp:5037
* daemon started successfully
List of devices attached
2G0YC1ZF9J0D9D	unauthorized
```
If `unauthorized` appears, it means the XR device has not been authorized. Put on the XR device and click "Allow" in the pop-up window asking for USB debugging permission. Then execute the command again:
```bash
$ sudo ./adb devices
List of devices attached
2G0YC1ZF9J0D9D	device
```
Start ADB port reverse forwarding by executing:
```bash
$ sudo ./adb -s 2G0YC1ZF9J0D9D reverse tcp:8012 tcp:8012
8012
```
You can verify the result using:
```bash
$ sudo ./adb -s 2G0YC1ZF9J0D9D reverse --list
UsbFfs tcp:8012 tcp:8012
```
Configure local HTTPS using a self-signed certificate, this step fuction same as [2.2.3 create certificate](https://github.com/unitreerobotics/avp_teleoperate)
```bash
$ openssl req -x509 -nodes -days 365 -newkey rsa:2048 -keyout key.pem -out cert.pem
```
If you want to use Wireless Network Reverse Port Forwarding:
```bash
$ sudo ./adb shell ifconfig wlan0
wlan0     Link encap:UNSPEC    Driver cnss_pci
          inet addr:192.168.123.101  Bcast:192.168.123.255  Mask:255.255.255.0 
          inet6 addr: fe80::5341:a567:4257:2c56/64 Scope: Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:136560 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:115802 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:3000 
          RX bytes:156741122 TX bytes:77421476
$ sudo ./adb tcpip 5566
restarting in TCP mode port: 5566
$ sudo ./adb connect 192.168.123.101:5566
connected to 192.168.123.101:5566
$ sudo ./adb devices
List of devices attached
192.168.123.101:5566	device
$ sudo ./adb -s 192.168.123.101:5566 reverse tcp:8012 tcp:8012
8012
$ sudo ./adb -s 192.168.123.101:5566 reverse --list
host-28 tcp:8012 tcp:8012
```

wzds2015, 2025-02-18:
> We have tried using hand tracking for teleoperation on the Meta-Quest 3 (its system specifications are: System version: 49829370066100510 Version: 62.0.0.273.343.570372087 Runtime version: [...]

Thanks. To make sure I understand: after making port forward and self-signed certificate. Everything else is exactly the same as the current code, as long as I specify the Quest device IP, correct?

silencht, 2025-02-20:
It should be like this. To clarify, you only need the device IP that is running the `avp_teleoperate` program in the Quest browser, not the IP of the Quest device itself.

Feel free to ask if you have any questions.

closed 2025-03-07 by silencht

Outlier-c, 2025-03-19:
I entered my URL into Quest3's Browser, but the page gets stuck on loading. I'm sure the other parts are fine because I can access the URL using Wolvic, enter the Vuer server, and see the G1 head camera's view (however, Wolvic freezes with a black screen after clicking "Enter VR", so it's unusable). Back in November last year, my Browser was still able to access Vuer normally.
I suspect it might be caused by a recent update to Quest's Browser. So, are there any other XR browsers that support avp_teleoperate?

csabahi, 2026-01-09:
> I want to use Wireless Network Reverse Port Forwarding， Are the following operations performed on the control computer（unitree@Host） or on the robot computer（unitree@PC2）?

Did you figure this out? I have the same question. I setup the Reverse Port Forwarding on my control computer for the sim teleoperate, not sure if I have to do the same on the robot computer or not for real robot teleoperation