unitreerobotics/xr_teleoperate #34: AVP camera stream

RiccardoZuppetti, 2025-02-18:
Hi, to which port on the H1 robot (not H1_2) should I connect the stereo-camera in order to have the stream into the VR? Eventually, it could be possible to exploit the built-in realsense?

silencht, 2025-02-20:
To use the stereo camera with the H1 robot, simply connect it to the USB port of the [Development Computing Unit (DCU) on the H1](https://support.unitree.com/home/en/H1_developer). Then, use `ssh` to log in the corresponding DCU and run the `image_server.py` program. (Note: `image_server.py` is not pre-installed on the DCU. You need to manually copy this file from the project directory to the DCU.)

**What does `image_server.py` do?**
It uses OpenCV’s (v4l2) API to read image streams from `/dev/video*` devices or Realsense cameras via the Realsense Python API. After reading the image data, it broadcasts the images over the local network on port 5555 using the ZMQ protocol. The `image_client.py` then receives this image data from port 5555 via ZMQ.

**For other camera brands?**
If you want to use a different brand of camera, you’ll need to create your own custom version of `image_server.py`. Write code to load the appropriate camera driver, read the images, and broadcast the data using the same ZMQ protocol.

**Using the H1’s built-in Realsense D435?**
Yes, you can use the built-in Realsense D435 camera on the H1 (it’s a monocular RGB camera). However, the visual feed in the Apple Vision Pro will be displayed as a flat image without stereo effects:
We handle the transmission of monocular RGB images through the following code in teleop_hand_and_arm.py:
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/teleop_hand_and_arm.py#L49-L61
The `BINOCULAR` variable is actually used here:

https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/teleop_hand_and_arm.py#L80
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/open_television/television.py#L15
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/open_television/television.py#L33-L36
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/open_television/television.py#L108-L128

You can refer to the [vuer (is a light-weight visualization toolkit for interacting with dynamic 3D and robotics data. It is VR and AR ready, and can be run on mobile devices.)](https://github.com/vuer-ai/vuer) for further assistance.

------   
Note: The h1 branch has only implemented a basic IK solver and is not fully adapted yet. There might be issues with joint mapping, such as in this [PR](https://github.com/unitreerobotics/avp_teleoperate/pull/21).

RiccardoZuppetti, 2025-02-20:
> Note: The h1 branch has only implemented a basic IK solver and is not fully adapted yet. There might be issues with joint mapping, such as in this https://github.com/unitreerobotics/avp_teleoperate/pull/21.

Actually I've tried to adapt everything to the specific use of the H1 adapting unitree_go within the control scripts. If I'll have something that is going to work maybe I'll open a PR for the H1 branch.

> Using the H1’s built-in Realsense D435?
Yes, you can use the built-in Realsense D435 camera on the H1 (it’s a monocular RGB camera). However, the visual feed in the Apple Vision Pro will be displayed as a flat image without stereo effects:
We handle the transmission of monocular RGB images through the following code in teleop_hand_and_arm.py

And in this case, how can I obtain the camera stream? I think that for my purpose I should not execute the `image_server.py` from the DCU probably (also because there the camera source refer to something that is connected to the DCU). Is there a topic on which can be retrieved the stream from the realsense?

> What does image_server.py do?
It uses OpenCV’s (v4l2) API to read image streams from /dev/video* devices or Realsense cameras via the Realsense Python API. After reading the image data, it broadcasts the images over the local network on port 5555 using the ZMQ protocol. The image_client.py then receives this image data from port 5555 via ZMQ.

So basically can be also attached a Realsense camera to the DCU? Apart from the built-in one. If yes, I assume that still the images are in a monocular format.

silencht, 2025-02-21:
The `class RealSenseCamera(object)`  is generic;
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/image_server/image_server.py#L10

it uses `pyrealsense2` to capture image streams from a RealSense camera. It can be the D405 mounted on the wrist or the D435 on the head, simply by adjusting the `config`:
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/image_server/image_server.py#L105-L137

------ 
There is indeed an issue. The RealSense D435 camera on the robot's head is currently connected to the Motion Control Computing Unit (PC 1) and is not yet publishing its image data stream in the system (this will be implemented in the future, but it is not available at the moment).  

So, if you want to use the built-in RealSense D435, you need to open the robot’s head cover and connect the RealSense D435 camera cable to the neck USB port of the DCU.

Inverse kinematics (IK) control is independent of the camera stream. The camera stream serves only two purposes: first, to provide a first-person view for assisting in teleoperating the robot, and second, to collect image data for training ai purposes. For safety reasons, you can first use MeshCat to test the correctness of your IK teleoperation program in a virtual environment.

silencht, 2025-02-25:
You can check the code here to perform unit testing of the inverse kinematics function using the MeshCat visualization tool.
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/robot_control/robot_arm_ik.py#L512-L560

If you think the test results in the visualization are good, you can follow this code to perform fixed trajectory testing on the real robot.
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/robot_control/robot_arm.py#L569-L626

Similarly, the dexterous hand code also has a unit test. You can run it or make small changes to test this function.
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/robot_control/robot_hand_unitree.py#L393-L459

The Vuer library is also there—it sends images to the XR headset and gets wrist and finger data from it.
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/open_television/television.py#L156-L178

silencht, 2025-02-25:
Oh, I forgot to mention, to run the code, just go to the folder with the files and run `python filename.py`.
If you have any other questions, feel free to post them below.

closed 2025-04-25 by silencht