unitreerobotics/xr_teleoperate #40: Page gets stuck on loading....

Outlier-c, 2025-03-19:
I entered my URL into Quest3's Browser, but the page gets stuck on loading. I'm sure the other parts are fine because I can access the URL using Wolvic, enter the Vuer server, and see the G1 head camera's view (however, Wolvic freezes with a black screen after clicking "Enter VR", so it's unusable). Back in November last year, my Browser was still able to access Vuer and support teleoperate normally.

I suspect it might be caused by a recent update to Quest's Browser. So, are there any other XR browsers that support avp_teleoperate with Quest3?

silencht, 2025-03-19:
my quest3's device information is write in this issue https://github.com/unitreerobotics/avp_teleoperate/issues/32
this is my vuer library test process:
### 1. Configure Environment

#### Set up local HTTPS with a self-signed certificate

```bash
$ openssl req -x509 -nodes -days 365 -newkey rsa:2048 -keyout key.pem -out cert.pem
```

#### Install Vuer environment

```bash
conda create -n vuer python=3.8
conda activate vuer
pip install -U 'vuer[all]==0.0.32-rc7'
```

### 2. Run the Program

#### Vuer Server

Create a file named `testvuer.py` with the following content:

```python
from vuer import Vuer, VuerSession
from vuer.schemas import Hands
from asyncio import sleep

# app = Vuer()
app = Vuer(host='0.0.0.0', cert="./cert.pem", key="./key.pem", queries=dict(grid=False), queue_len=3)

@app.add_handler("HAND_MOVE")
async def handler(event, session):
    print(f"Movement Event: key-{event.key}", event.value)

        
@app.spawn(start=True)
async def main(session: VuerSession):
    session.upsert @ Hands(fps=60, stream=True, key="hands", showLeft=True, showRight=True)
    while True:
        await sleep(0.033)
```

The PICO4U/QUEST3 headset and the computer running this server program must be on the same local network.  
Assume the computer's IP address is `192.168.123.10`.

Run the `vuertest.py` program:

```bash
(vuer) unitree@unitree:~/vuertest$ python vuertest.py 
Serving file:///home/unitree/vuertest at /static
Visit: https://vuer.ai?grid=False
```

#### Browser Client

In the PICO4U/QUEST3 browser, enter the following URL in the address bar:

```
https://192.168.123.10:8012/?ws=wss://192.168.123.10:8012
```

Ignore the HTTPS warning and proceed to the website.

At this point, the `vuertest.py` terminal output will show:

```bash
websocket is connected. id:27d0d4bf-0cee-4943-863a-74d06c1ce3b9
default socket worker is up, adding clientEvents 
Uplink task running. id:27d0d4bf-0cee-4943-863a-74d06c1ce3b9
```

Click **ENTER VR** at the bottom of the PICO4U/QUEST3 browser page to enter **VR mode**.

Now, the `vuertest.py` terminal will continuously output hand tracking data similar to the following:

```bash
Movement Event: key-hands {'keys': ['wrist', 'thumb-metacarpal', 'thumb-phalanx-proximal', 'thumb-phalanx-distal', 'thumb-tip', 'index-finger-metacarpal', 'index-finger-phalanx-proximal', 'index-finger-phalanx-intermediate', 'index-finger-phalanx-distal', 'index-finger-tip', 'middle-finger-metacarpal', 'middle-finger-phalanx-proximal', 'middle-finger-phalanx-intermediate', 'middle-finger-phalanx-distal', 'middle-finger-tip', 'ring-finger-metacarpal', 'ring-finger-phalanx-proximal', 'ring-finger-phalanx-intermediate', 'ring-finger-phalanx-distal', 'ring-finger-tip', 'pinky-finger-metacarpal', 'pinky-finger-phalanx-proximal', 'pinky-finger-phalanx-intermediate', 'pinky-finger-phalanx-distal', 'pinky-finger-tip'], 'leftLandmarks': [], 'leftHand': ExtType(code=0, data=b'\x00'), 'leftState': ExtType(code=0, data=b'\x00'), 'rightLandmarks': [[0.24259380996227264, 0.8467217683792114, -0.3405609428882599], [0.21547923982143402, 0.8566388487815857, -0.36539873480796814], [0.20332616567611694, 0.8535754680633545, -0.3906826376914978], [0.19521452486515045, 0.8456712365150452, -0.41622063517570496], [0.1884956657886505, 0.835878312587738, -0.4421296715736389], [0.22898007929325104, 0.8592796325683594, -0.3624369204044342], [0.23388302326202393, 0.8815991282463074, -0.4189493954181671], [0.22055953741073608, 0.863884687423706, -0.45003724098205566], [0.20920178294181824, 0.8483495712280273, -0.4581644833087921], [0.19728337228298187, 0.8319891095161438, -0.4638698399066925], [0.24299131333827972, 0.8525811433792114, -0.3625074625015259], [0.2539966106414795, 0.8712708950042725, -0.41819941997528076], [0.24114209413528442, 0.8536161780357361, -0.45790016651153564], [0.2282591611146927, 0.8379329442977905, -0.47167420387268066], [0.2156200110912323, 0.8224398493766785, -0.48657628893852234], [0.25312647223472595, 0.8448742032051086, -0.3619600534439087], [0.26520875096321106, 0.8549282550811768, -0.4152206778526306], [0.25495585799217224, 0.8380534648895264, -0.45311710238456726], [0.24459728598594666, 0.825562059879303, -0.4655803143978119], [0.2339121401309967, 0.8126782178878784, -0.47845539450645447], [0.26097145676612854, 0.834571361541748, -0.3596843183040619], [0.27606311440467834, 0.8342973589897156, -0.4129590094089508], [0.27137666940689087, 0.8235505819320679, -0.43700623512268066], [0.2635972797870636, 0.8138087391853333, -0.4481678903102875], [0.25401541590690613, 0.8022562861442566, -0.46032214164733887]], 'rightHand': [0.789176881313324, -0.613820493221283, 0.020597416907548904, 0, 0.5954859852790833, 0.7729519009590149, 0.2189563363790512, 0, -0.15032067894935608, -0.16052980720996857, 0.975517213344574, 0, 0.24259380996227264, 0.8467217683792114, -0.3405609428882599, 1], 'rightState': {'pinching': False}}
······
······
```

At this point, inside the VR headset, you should see 25 small cubes representing the positions of both hands.

Outlier-c, 2025-03-19:
Thank you for your quick response. I followed your suggestion, but I couldn't complete this step:_Ignore the HTTPS warning and proceed to the website.._ **My Quest3's Browser froze**, as shown in the image below (During the first attempt, there was a security warning which I ignored, but it still got stuck during loading). The screenshot of PyCharm running is shown below, and it doesn't display as expectation.

Also, I ve made sure that URL matches my computer's IP: 192.168.123.182.

[image]

[image]

Outlier-c, 2025-03-19:
I wonder if it's an issue with the Quest3 Browser. If your Quest3 Browser is up-to-date, it should also encounter this problem.

Outlier-c, 2025-03-19:
Plus：I'm also having this issue with another Quest3

silencht, 2025-03-20:
I’m very sorry, but I currently don’t have a Quest device. Additionally, I believe you may get better support by discussing this issue in the Vuer project or with Meta Quest Team.

Additionally, I can add that you can enable ADB log output while debugging to check for potential errors on the Quest device through the logs. The command is similar to:
```bash
(tv) unitree@unitree:~/Downloads/platform-tools-latest-linux/platform-tools$ sudo ./adb logcat -v time
```
Check the logs related to the time when you refreshed the browser and clicked the "Enter VR" button to see if any system errors occurred. If you find any errors, you can submit them to the Meta Quest team to seek a solution.

Outlier-c, 2025-03-20:
Thanks for your response and advice. I’ll contact Meta’s official team later for more details.

closed 2025-03-20 by Outlier-c

Outlier-c, 2025-03-31:
It was solved after I regenerated cert.pem and key.pem and reinstall rootCA.pem to my Quest3.

silencht, 2025-06-17:
> > 您好，黑屏的具体含义是无法刷新正常进入页面;还是已经进入Vuer页面，但是界面中间的图像未正常显示？ 另外，能否提供您启动过程中的终端输出信息以方便我们排查问题。
> 
> [image]
> 
> 已经进入了Vuer页面,界面中间的图像未正常显示，此外我在vision pro中虽然25 个小立方体的手部不能显示，但是随着我的手部移动head_rmat, left_wrist, right_wrist, left_hand, right_hand = tv_wrapper.get_data(),有数据变化

@Simease 应该是图像服务器/客户端没有正常通信，请查看image_server目录。有数据变化即可，25 个小立方体的手部不能显示不会影响手部数据跟踪。