unitreerobotics/xr_teleoperate #42: one method to **Perform Dual-arm Teleoperation in Motion Control Mode**

silencht, 2025-03-20:
**Note! This mode has not undergone large-scale testing, so users must ensure their own safety while using it. Blind operation or careless coding can also easily damage the robot!**

Before running the teleoperation program, you can first run and review the example program at [this link](https://support.unitree.com/home/en/G1_developer/arm_control_routine).

In other words, to perform dual-arm teleoperation in motion control mode (where the motion control program controls the lower body joints instead of directly locking them), you can achieve this through the DDS interface in [Advanced Motion => arm control routine](https://support.unitree.com/home/en/G1_developer/arm_control_routine).

The more specific approach is as follows:
1. Modify the line of code in 'avp_teleoperate/teleop/robot_control/robot_arm.py' from
https://github.com/unitreerobotics/avp_teleoperate/blob/20633f692d3b6d985ab12cb038cb05e817eabfe3/teleop/robot_control/robot_arm.py#L11 to
   ```python
   kTopicLowCommand = "rt/armsdk"
   ``` 
2. Additionally, at the start of the program and when exiting, you need to handle the **weight value** in order to mix the commands effectively. At the start of the teleoperation, gradually increase the weight value from 0 to 1.0; while exiting the teleoperation, gradually decrease the weight value from 1.0 to 0.

      The purpose can be referenced from this [webpage](https://support.unitree.com/home/en/G1_developer/arm_control_routine):
      > JointIndex::kNotUsedJoint of motor_cmd is the unused joint motor serial number. In the SDK, the value of this joint angular position is used as the transition weight of other motor rotations (weight in the above code). When weight changes from 0 to 1, the motor will gradually transition from the current position to the desired position. The faster the weight changes, the faster the motor's transition speed will be. Please refer to ["Sport Motion Service Interface"](https://support.unitree.com/home/en/G1_developer/sport_services_interface).

      > You can roughly understand it as:  
      `The actual command executed by the dual arms = motion control command * (1 - weight) + rt/arm_sdk user command * weight`
      > The motion control commands are:  
      > 1. Naturally hanging down when stationary.  
      > 2. A natural swinging position similar to human arms when walking.

      For example, This line of code directly sets the weight value of Unitree G1 to 1.0.:
    ```python
    self.msg.motor_cmd[G1_29_JointIndex.kNotUsedJoint0].q = 1.0;
    ```
      > Note that the above code only shows how to set the weight value. If there is a gap between the positions of the motion control commands and the user commands (published through rt/armsdk), suddenly changing the weight value may cause the robotic arm to move at high speed. Therefore, it is recommended to gradually adjust the weight value.

3. After robot powering on, do not enter debug mode, but instead enter **motion control mode**. Taking the Unitree G1 robot as an example, after powering on, press **L1 + A** to enter **damping mode**, then press **L1 + UP** to enter **preparation mode (Get Ready)**. Finally, press **R1 + X** to enter the main motion control program.
4. At this point, start the AVP teleoperation program (after modification). Both hands should be able to lift normally as before, and after pressing **r**, teleoperation should work properly.

If you have any questions about the above content, feel free to raise them under this issue.

schirane1, 2025-03-20:
Moving this here per @silencht's comment.

Hello- trying to build using this repo and wondering if scope is limited to arm movements only? if so, is there any workaround for walking?

Kidthe-72, 2025-04-13:
What should be done specifically for the second step？I cant let hands lift

silencht, 2025-05-20:
With the changes in the G1 remote control mode, some parts of the above documentation are now outdated. 
As described in [the latest Unitree documentation](https://support.unitree.com/home/en/G1_developer/remote_control), the button combinations for switching from startup to main control have been updated to:
`L2 + B` (damping mode) ==> `L2 + UP` (get ready mode) ==> `R1 + X` (main operation control mode).

Here is an `robot_arm.py' example code you can refer to 
(The key parts to look at are STEP1, STEP2, and STEP3) :

```python
import numpy as np
import threading
import time
from enum import IntEnum

from unitree_sdk2py.core.channel import ChannelPublisher, ChannelSubscriber, ChannelFactoryInitialize # dds
from unitree_sdk2py.idl.unitree_hg.msg.dds_ import ( LowCmd_  as hg_LowCmd, LowState_ as hg_LowState) # idl for g1, h1_2
from unitree_sdk2py.idl.default import unitree_hg_msg_dds__LowCmd_
from unitree_sdk2py.utils.crc import CRC

from unitree_sdk2py.idl.unitree_go.msg.dds_ import ( LowCmd_  as go_LowCmd, LowState_ as go_LowState)  # idl for h1
from unitree_sdk2py.idl.default import unitree_go_msg_dds__LowCmd_

# STEP 1
# kTopicLowCommand = "rt/lowcmd"
kTopicLowCommand = "rt/arm_sdk"
kTopicLowState = "rt/lowstate"

G1_29_Num_Motors = 35
G1_23_Num_Motors = 35
H1_2_Num_Motors = 35
H1_Num_Motors = 20
 

class MotorState:
    def __init__(self):
        self.q = None
        self.dq = None

class G1_29_LowState:
    def __init__(self):
        self.motor_state = [MotorState() for _ in range(G1_29_Num_Motors)]

class G1_23_LowState:
    def __init__(self):
        self.motor_state = [MotorState() for _ in range(G1_23_Num_Motors)]

class H1_2_LowState:
    def __init__(self):
        self.motor_state = [MotorState() for _ in range(H1_2_Num_Motors)]

class H1_LowState:
    def __init__(self):
        self.motor_state = [MotorState() for _ in range(H1_Num_Motors)]

class DataBuffer:
    def __init__(self):
        self.data = None
        self.lock = threading.Lock()

    def GetData(self):
        with self.lock:
            return self.data

    def SetData(self, data):
        with self.lock:
            self.data = data

class G1_29_ArmController:
    def __init__(self):
        print("Initialize G1_29_ArmController...")
        self.q_target = np.zeros(14)
[... 242 lines cut ...]
    kLeftShoulderPitch = 15
    kLeftShoulderRoll = 16
    kLeftShoulderYaw = 17
    kLeftElbow = 18
    kLeftWristRoll = 19
    kLeftWristPitch = 20
    kLeftWristyaw = 21

    # Right arm
    kRightShoulderPitch = 22
    kRightShoulderRoll = 23
    kRightShoulderYaw = 24
    kRightElbow = 25
    kRightWristRoll = 26
    kRightWristPitch = 27
    kRightWristYaw = 28
    
    # not used
    kNotUsedJoint0 = 29
    kNotUsedJoint1 = 30
    kNotUsedJoint2 = 31
    kNotUsedJoint3 = 32
    kNotUsedJoint4 = 33
    kNotUsedJoint5 = 34
```