unitreerobotics/xr_teleoperate #45: head cmaera POV ?

eddierex857, 2025-04-28:
I would like to ask what is the POV field of view of the camera on the head in your Apple Vision Pro remote control project. I am currently using a 125-degree camera, which will exceed the picture.

silencht, 2025-04-30:
> In addition, I have successfully used Quest3 to achieve remote operation, but every time I connect the host to Quest3, I need to connect it by wire first. Can I achieve wireless connection, similar to deploying trust certificates in Apple Vision Pro?

When testing with Pico4 Ultra, I didn’t use port forwarding. I just ignored the "Your connection is not private" warning and clicked "Proceed to 192.168.123.x (unsafe)" to access the page. You could try the same.

silencht, 2025-04-30:
> I would like to ask what is the POV field of view of the camera on the head in your Apple Vision Pro remote control project. I am currently using a 125-degree camera, which will exceed the picture.

We purchased the 125° distortion-free camera module from this [link](http://e.tb.cn/h.TaZxgkpfWkNCakg?tk=KKz03Kyu04u). According to the documentation provided by the seller, the field of view (FOV) is: horizontal 103°, vertical 86°, and diagonal 125°.  
During teleoperation, we usually use MJPEG format with a resolution of 2560×720. When recording training data, we switch to MJPEG format with a resolution of 1280×480.

closed 2025-05-06 by eddierex857