unitreerobotics/xr_teleoperate #46: Dexterous hands are not able to move properly

eddierex857, 2025-05-09:
Hello, I have completed the teleoperation with the G1 with the Timing Dexterous Hand, but I found that my two dexterous hands only follow the movement of my right hand, what should I do.
I have two terminal inputs on pc2: sudo ./inspire_hand -s /dev/ttyUSB1 and sudo ./inspire_hand -s /dev/ttyUSB2

silencht, 2025-05-09:
Hello, we will investigate this issue shortly. As soon as there is any update, we will notify you immediately.

jsgaobiao, 2025-05-10:
We meet the same problem on G1 with Inspire DFX Dexterous Hand, looking forward to your reply.   : )

jsgaobiao, 2025-05-10:
> Hello, we will investigate this issue shortly. As soon as there is any update, we will notify you immediately.
We meet the same problem on G1 with Inspire DFX Dexterous Hand, looking forward to your reply. : )

jsgaobiao, 2025-05-10:
In G1 Inspire example code [(here)](https://oss-global-cdn.unitree.com/static/0a8335f7498548d28412c31ea047d4be.zip)
inspire_ctrl.cpp.  When initialize serial port, two hands' cmd are send to the same port, maybe it should be modified like:

`class InspireRunner
{
public:
  InspireRunner()
  {
    //serial = std::make_shared<SerialPort>(param::serial_port, B115200);
    serial1 = std::make_shared<SerialPort>("/dev/ttyUSB1", B115200);
    serial2 = std::make_shared<SerialPort>("/dev/ttyUSB2", B115200);

    // inspire
    righthand = std::make_shared<inspire::InspireHand>(serial1, 1); // ID 1
    lefthand = std::make_shared<inspire::InspireHand>(serial2, 2); // ID 2
    ..............................
  }

eddierex857, 2025-05-10:
> 在 G1 Inspire 示例代码[（此处）](https://oss-global-cdn.unitree.com/static/0a8335f7498548d28412c31ea047d4be.zip)中inspire_ctrl.cpp。当初始化串口时，两只手的 cmd 都发送到同一个端口，也许应该修改如下：
> 
> class InspireRunner { public： InspireRunner（） { //serial = std：：make_shared（param：：serial_port， B115200）; serial1 = std：：make_shared（“/dev/ttyUSB1”， B115200）; serial2 = std：：make_shared（“/dev/ttyUSB2”， B115200）; // inspire righthand = std：：make_sharedinspire：：InspireHand（serial1， 1）; // ID 1 lefthand = std：：make_sharedinspire：：InspireHand（serial2， 2）; // ID 2.............................. } ................. }

Hello, may I ask if you have solved this problem? I rewrote it according to your suggestion, but I found that when running ./h1_hand_example, only the left hand can move.

jsgaobiao, 2025-05-10:
> Hello, may I ask if you have solved this problem? I rewrote it according to your suggestion, but I found that when running ./h1_hand_example, only the left hand can move.

There is a mistake above, where the right hand should be bound to ttyUSB2, the left hand is ttyUSB1;
Besides, try to change id. My G1 hands can work correctly now.
 
`righthand = std::make_sharedinspire::InspireHand(serial2, 1); 
lefthand = std::make_sharedinspire::InspireHand(serial1, 1); `

closed 2025-05-13 by eddierex857

silencht, 2025-05-13:
@eddierex857 May I ask if you have already resolved this issue? Due to the lack of necessary hardware, I have not yet started addressing this issue. If you already have a solution, could you please provide more information under this issue? We really appreciate your contribution!

eddierex857, 2025-05-13:
> [@eddierex857](https://github.com/eddierex857) May I ask if you have already resolved this issue? Due to the lack of necessary hardware, I have not yet started addressing this issue. If you already [...]

In fact, I rewrote the code as mentioned above, which can solve this problem.

In G1 Inspire example code [(here)](https://oss-global-cdn.unitree.com/static/0a8335f7498548d28412c31ea047d4be.zip)
inspire_ctrl.cpp. When initialize serial port, two hands' cmd are send to the same port, maybe it should be modified like:

class InspireRunner
{
public:
InspireRunner()
{
//serial = std::make_shared(param::serial_port, B115200);
serial1 = std::make_shared("/dev/ttyUSB1", B115200);
serial2 = std::make_shared("/dev/ttyUSB2", B115200);
// inspire
righthand = std::make_sharedinspire::InspireHand(serial2, 1); // ID 1
lefthand = std::make_sharedinspire::InspireHand(serial1, 1); // ID 2
..............................
}
.................
}

silencht, 2025-05-23:
The complete code of a properly working `inspire_ctrl.cpp` is as follows:

```c++
#include "inspire.h"
#include "param.h"

#include "dds/Publisher.h"
#include "dds/Subscription.h"
#include <unitree/idl/go2/MotorCmds_.hpp>
#include <unitree/idl/go2/MotorStates_.hpp>
#include <unitree/common/thread/recurrent_thread.hpp>

class InspireRunner
{
public:
  InspireRunner()
  {
    serial1 = std::make_shared<SerialPort>("/dev/ttyUSB1", B115200);
    serial2 = std::make_shared<SerialPort>("/dev/ttyUSB2", B115200);

    // If your left and right hand controls are reversed, you can swap the positions of `serial1` and `serial2` below.
    righthand = std::make_shared<inspire::InspireHand>(serial1, 1);
    lefthand = std::make_shared<inspire::InspireHand>(serial2, 1);

    // dds
    handcmd = std::make_shared<unitree::robot::SubscriptionBase<unitree_go::msg::dds_::MotorCmds_>>(
        "rt/" + param::ns + "/cmd");
    handcmd->msg_.cmds().resize(12);
    handstate = std::make_unique<unitree::robot::RealTimePublisher<unitree_go::msg::dds_::MotorStates_>>(
        "rt/" + param::ns + "/state");
    handstate->msg_.states().resize(12);

    // Start running
    thread = std::make_shared<unitree::common::RecurrentThread>(
      10000, std::bind(&InspireRunner::run, this)
    );
  }

  void run()
  {
    // Set command
    if(!handcmd->isTimeout())
    {
      for(int i(0); i<12; i++)
      {
        qcmd(i) = handcmd->msg_.cmds()[i].q();
      }
      righthand->SetPosition(qcmd.block<6, 1>(0, 0));
      lefthand->SetPosition(qcmd.block<6, 1>(6, 0));
    }

    // Recv state
    Eigen::Matrix<double, 6, 1> qtemp;
    if(righthand->GetPosition(qtemp) == 0)
    {
      qstate.block<6, 1>(0, 0) = qtemp;
    }
    else
    {
      for(int i(0); i<6; i++)
      {
        handstate->msg_.states()[i].lost()++;
      }
      // spdlog::debug("Failed to get right hand state");
    }
    if(lefthand->GetPosition(qtemp) == 0)
    {
      qstate.block<6, 1>(6, 0) = qtemp;
    }
    else
    {
      for(int i(0); i<6; i++)
      {
        handstate->msg_.states()[i+6].lost()++;
      }
      // spdlog::debug("Failed to get left hand state");
    }
    if(handstate->trylock())
    {
        for(int i(0); i<12; i++)
        {
            handstate->msg_.states()[i].q() = qstate(i);
        }
        handstate->unlockAndPublish();
    }
  }

  unitree::common::ThreadPtr thread;

  // inspire
  SerialPort::SharedPtr serial1;
  SerialPort::SharedPtr serial2;
  std::shared_ptr<inspire::InspireHand> lefthand;
  std::shared_ptr<inspire::InspireHand> righthand;
  Eigen::Matrix<double, 12, 1> qcmd, qstate;

  // dds
  std::unique_ptr<unitree::robot::RealTimePublisher<unitree_go::msg::dds_::MotorStates_>> handstate;
  std::shared_ptr<unitree::robot::SubscriptionBase<unitree_go::msg::dds_::MotorCmds_>> handcmd;
};

int main(int argc, char ** argv)
{
  auto vm = param::helper(argc, argv);
  unitree::robot::ChannelFactory::Instance()->Init(0, param::network);

  std::cout << " --- Unitree Robotics --- " << std::endl;
  std::cout << "  Inspire Hand Controller  " << std::endl;

  InspireRunner runner;
 
  while (true)
  {
    sleep(1);
  }
  return 0;
}
```

Since two serial control classes are constructed at once, the program only needs to be executed once in the terminal.