unitreerobotics/xr_teleoperate #48: could you develop more compatibility for inspire hand RH56DFTP?

chaingee, 2025-05-16:
avp_teleoperate only work on inspire hand RH56DFX, So I made some adjustment on teleop/robot_control/robot_hand_inspire.py to fit RH56DFTP
as below, but it's not working properly. please help me to use avp_teleoperate with RH56DFTP. thank you

```
# this file is legacy, need to fix.
from unitree_sdk2py.core.channel import ChannelPublisher, ChannelSubscriber, ChannelFactoryInitialize # dds
# from unitree_sdk2py.idl.unitree_go.msg.dds_ import MotorCmds_, MotorStates_ # OLD IDL, REMOVED
# from unitree_sdk2py.idl.default import unitree_go_msg_dds__MotorCmd_ # OLD IDL, REMOVED

# NEW IMPORTS for Inspire Hand SDK
from inspire_sdkpy import inspire_dds, inspire_hand_defaut

from teleop.robot_control.hand_retargeting import HandRetargeting, HandType # Assuming this remains the same
import numpy as np
# from enum import IntEnum # Old Enums for joint indexing might not be needed for new DDS messages
import threading
import time
from multiprocessing import Process, Array, Lock # Removed shared_memory as Array is used

inspire_tip_indices = [4, 9, 14, 19, 24] # Assuming this remains relevant for hand_retargeting
Inspire_Num_Motors = 6 # Number of motors per hand

# NEW DDS TOPIC NAMES (assuming these are the correct topics based on SDK examples)
kTopicInspireCtrlLeft = "rt/inspire_hand/ctrl/l"
kTopicInspireCtrlRight = "rt/inspire_hand/ctrl/r"
kTopicInspireStateLeft = "rt/inspire_hand/state/l"
kTopicInspireStateRight = "rt/inspire_hand/state/r"

class Inspire_Controller:
    def __init__(self, left_hand_array, right_hand_array, dual_hand_data_lock=None, dual_hand_state_array=None,
                 dual_hand_action_array=None, fps=100.0, Unit_Test=False, network_interface=""): # Added network_interface
        print("Initialize Inspire_Controller...")
        self.fps = fps
        self.Unit_Test = Unit_Test

        # Initialize DDS Channel Factory
        # This should ideally be called once per process.
        # If multiple controllers or DDS entities run in the same process, ensure this is handled.
        try:
            ChannelFactoryInitialize(0, network_interface)
            print(f"DDS ChannelFactory initialized with interface: '{network_interface if network_interface else 'default'}'")
        except Exception as e:
            print(f"Warning: DDS ChannelFactoryInitialize failed or already initialized: {e}")

        if not self.Unit_Test:
            self.hand_retargeting = HandRetargeting(HandType.INSPIRE_HAND)
        else:
            # Assuming a different config for unit tests if necessary
            self.hand_retargeting = HandRetargeting(HandType.INSPIRE_HAND_Unit_Test)

        # Initialize hand command publishers
        self.LeftHandCmd_publisher = ChannelPublisher(kTopicInspireCtrlLeft, inspire_dds.inspire_hand_ctrl)
        self.LeftHandCmd_publisher.Init()
        self.RightHandCmd_publisher = ChannelPublisher(kTopicInspireCtrlRight, inspire_dds.inspire_hand_ctrl)
        self.RightHandCmd_publisher.Init()

        # Initialize hand state subscribers
        self.LeftHandState_subscriber = ChannelSubscriber(kTopicInspireStateLeft, inspire_dds.inspire_hand_state)
        self.LeftHandState_subscriber.Init() # Consider using callback if preferred: Init(callback_func, period_ms)
        self.RightHandState_subscriber = ChannelSubscriber(kTopicInspireStateRight, inspire_dds.inspire_hand_state)
        self.RightHandState_subscriber.Init()

        # Shared Arrays for hand states ([0,1] normalized values)
        self.left_hand_state_array = Array('d', Inspire_Num_Motors, lock=True)
        self.right_hand_state_array = Array('d', Inspire_Num_Motors, lock=True)

        # Initialize subscribe thread
        self.subscribe_state_thread = threading.Thread(target=self._subscribe_hand_state_loop)
        self.subscribe_state_thread.daemon = True
        self.subscribe_state_thread.start()
[... 233 lines cut ...]
        # For this example, the control loop runs indefinitely.
        # We can just let it run and observe, or add a way to stop it.
        count = 0
        while True:
            time.sleep(1.0) # Main thread does something else or just waits
            # Simulate a slight change in human hand input to see retargeting work
            with mock_right_hand_input.get_lock():
                 # Make a noticeable change to one coordinate
                mock_right_hand_input[inspire_tip_indices[0]*3 + 1] = 0.1 + (count % 10) * 0.02
            
            with shared_lock:
                print(f"Logged State: {list(shared_state[:])}")
                print(f"Logged Action: {list(shared_action[:])}")
            count +=1
            if count > 300 : # Run for 5 minutes
                 print("Example finished.")
                 break

    except Exception as e:
        print(f"An error occurred in the example: {e}")
    finally:
        # In a real app, you'd signal the controller's processes/threads to stop cleanly.
        # Since they are daemons, they will exit when the main program exits.
        print("Exiting main program.")
```

silencht, 2025-05-20:
Could you describe how the above code behaves when it runs? I didn’t notice any obvious issues.

chaingee, 2025-05-22:
thank you for your advise, i finally succeeded with just a few adjustment.
:)

closed 2025-05-23 by chaingee

silencht, 2025-05-23:
If it's convenient, could you let me know how you eventually resolved the issue?

chaingee, 2025-05-23:
first I run /inspire_hand_ws/inspire_hand_sdk$ python example/Headless_driver_double.py 
and then 
/avp_teleoperate/teleop$ python teleop_hand_and_arm.py --arm=G1_23 --hand=inspire1
with below code ; robot_hand_inspire.py
```
# this file is legacy, need to fix.
from unitree_sdk2py.core.channel import ChannelPublisher, ChannelSubscriber, ChannelFactoryInitialize # dds
# from unitree_sdk2py.idl.unitree_go.msg.dds_ import MotorCmds_, MotorStates_ # OLD IDL, REMOVED
# from unitree_sdk2py.idl.default import unitree_go_msg_dds__MotorCmd_ # OLD IDL, REMOVED

# NEW IMPORTS for Inspire Hand SDK
from inspire_sdkpy import inspire_dds, inspire_hand_defaut

from teleop.robot_control.hand_retargeting import HandRetargeting, HandType # Assuming this remains the same
import numpy as np
# from enum import IntEnum # Old Enums for joint indexing might not be needed for new DDS messages
import threading
import time
from multiprocessing import Process, Array, Lock # Removed shared_memory as Array is used

inspire_tip_indices = [4, 9, 14, 19, 24] # Assuming this remains relevant for hand_retargeting
Inspire_Num_Motors = 6 # Number of motors per hand

# NEW DDS TOPIC NAMES (assuming these are the correct topics based on SDK examples)
kTopicInspireCtrlLeft = "rt/inspire_hand/ctrl/l"
kTopicInspireCtrlRight = "rt/inspire_hand/ctrl/r"
kTopicInspireStateLeft = "rt/inspire_hand/state/l"
kTopicInspireStateRight = "rt/inspire_hand/state/r"

class Inspire_Controller:
    def __init__(self, left_hand_array, right_hand_array, dual_hand_data_lock=None, dual_hand_state_array=None,
                 dual_hand_action_array=None, fps=100.0, Unit_Test=False, network_interface=""): # Added network_interface
        print("Initialize Inspire_Controller...")
        self.fps = fps
        self.Unit_Test = Unit_Test
        self.debug_counter = 0 # 디버깅 카운터 추가

        # Initialize DDS Channel Factory
        # This should ideally be called once per process.
        # If multiple controllers or DDS entities run in the same process, ensure this is handled.
        try:
            ChannelFactoryInitialize(0, network_interface)
            print(f"DDS ChannelFactory initialized with interface: '{network_interface if network_interface else 'default'}'")
        except Exception as e:
            print(f"Warning: DDS ChannelFactoryInitialize failed or already initialized: {e}")

        if not self.Unit_Test:
            self.hand_retargeting = HandRetargeting(HandType.INSPIRE_HAND)
        else:
            # Assuming a different config for unit tests if necessary
            self.hand_retargeting = HandRetargeting(HandType.INSPIRE_HAND_Unit_Test)

        # Initialize hand command publishers
        self.LeftHandCmd_publisher = ChannelPublisher(kTopicInspireCtrlLeft, inspire_dds.inspire_hand_ctrl)
        self.LeftHandCmd_publisher.Init()
        self.RightHandCmd_publisher = ChannelPublisher(kTopicInspireCtrlRight, inspire_dds.inspire_hand_ctrl)
        self.RightHandCmd_publisher.Init()

        # Initialize hand state subscribers
        self.LeftHandState_subscriber = ChannelSubscriber(kTopicInspireStateLeft, inspire_dds.inspire_hand_state)
        self.LeftHandState_subscriber.Init() # Consider using callback if preferred: Init(callback_func, period_ms)
        self.RightHandState_subscriber = ChannelSubscriber(kTopicInspireStateRight, inspire_dds.inspire_hand_state)
        self.RightHandState_subscriber.Init()

        # Shared Arrays for hand states ([0,1] normalized values)
        self.left_hand_state_array = Array('d', Inspire_Num_Motors, lock=True)
        self.right_hand_state_array = Array('d', Inspire_Num_Motors, lock=True)

        # Initialize subscribe thread
        self.subscribe_state_thread = threading.Thread(target=self._subscribe_hand_state_loop)
[... 218 lines cut ...]
        while main_loop_running:
            try:
                time.sleep(1.0) 
                # Simulate a slight change in human hand input
                with mock_right_hand_input.get_lock():
                    # Make a noticeable change to one coordinate (e.g., y-coord of thumb tip)
                    # inspire_tip_indices[0] is thumb tip, index 1 is y-coordinate
                    mock_right_hand_input[inspire_tip_indices[0]*3 + 1] = 0.1 + (count % 10) * 0.02 
                
                with shared_lock:
                    print(f"Cycle {count} - Logged State: {[f'{x:.3f}' for x in shared_state[:]]}, Logged Action: {[f'{x:.3f}' for x in shared_action[:]]}")
                count +=1
                if count > 3000 : # Increased run time for more observation
                    print("Example finished after 3000 cycles.")
                    main_loop_running = False
            except KeyboardInterrupt:
                print("Main loop interrupted. Finishing example.")
                main_loop_running = False

    except Exception as e:
        print(f"An error occurred in the example: {e}")
    finally:
        print("Exiting main program.")

```

SahilChande, 2025-07-21:
Hie, I have followed the given instructions in the thread above as @FranRFH , I want to connect the inspire hand to G1 robot instead of dex3.  Also i am using apple vision pro for teleop, i am able to do this with dex3 with no issue. I did installed and build inspire_hand_ws from  https://github.com/NaCl-1374/inspire_hand_ws without any issue. And i can access these from in xr_teleoprate ws
I tried to execute the commands and script as @chaingee 
"""
first I run /inspire_hand_ws/inspire_hand_sdk$ python example/Headless_driver_double.py
and then
/xr_teleoperate/teleop$ python teleop_hand_and_arm.py --arm=G1_23 --hand=inspire1
with below code ; robot_hand_inspire.py 
"""
But i was not able to launch the hand_and_arm.py successfully,
where i am doing wrong.
can someone please explain me steps to do as to inspire hand with G1 for my teleop

silencht, 2025-07-21:
@SahilChande 

Hi, original `--hand=inspire1` parameter is only compatible with [DFX inspire hand](https://support.unitree.com/home/en/G1_developer/inspire_dfx_dexterous_hand). 

And https://github.com/NaCl-1374/inspire_hand_ws repo is developed for [FTP inspire hand](https://support.unitree.com/home/en/G1_developer/inspire_ftp_dexterity_hand). 

Please ensure that your Inspire_Controller class is updated accordingly based on the [above information](https://github.com/unitreerobotics/xr_teleoperate/issues/48#issuecomment-2903133553).

FranRFH, 2025-07-25:
@SahilChande I read that you were able to remotely control the Unitree G1 robot with the Dex3-1 hands. Could you give me a step-by-step guide on how to do it? I need to control the G1 with the Dex5-1 hands.

FranRFH, 2025-07-25:
@chaingee /inspire_hand_ws/inspire_hand_sdk$ python example/Headless_driver_double.py, /avp_teleoperate/teleop$ python teleop_hand_and_arm.py --arm=G1_23 --hand=inspire1. Do you run these commands on the robot's computer or locally?

suzhelearning, 2025-07-31:
@chaingee @silencht markdown

### Issue: Robotic Arm Hand Not Functioning While Arm Movement Works

**Description:**
Hi, I have followed the given instructions in the thread above, but I'm encountering an issue with my robotic arm setup. While the main arm movement (g1) is working correctly, the hand (gripper) mechanism is not responding or functioning at all.

**Current Behavior:**
- Arm movement (g1): ✅ Working correctly
- Hand/gripper function: ❌ Not working
```
[Inspire_Controller] Control process has been closed.
Process Process-3:
Traceback (most recent call last):
  File "/home/midea/miniconda3/envs/xr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/midea/miniconda3/envs/xr/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/midea/workspace/xr/xr_teleoperate/teleop/robot_control/robot_hand_inspire.py", line 187, in control_process_loop
    raw_left_q_target_rad = self.hand_retargeting.left_retargeting.retarget(ref_left_value)[
  File "/home/midea/workspace/xr/xr_teleoperate/teleop/robot_control/dex-retargeting/src/dex_retargeting/seq_retarget.py", line 109, in retarget
    qpos = self.optimizer.retarget(
  File "/home/midea/workspace/xr/xr_teleoperate/teleop/robot_control/dex-retargeting/src/dex_retargeting/optimizer.py", line 81, in retarget
    objective_fn = self.get_objective_function(ref_value, fixed_qpos, np.array(last_qpos).astype(np.float32))
  File "/home/midea/workspace/xr/xr_teleoperate/teleop/robot_control/dex-retargeting/src/dex_retargeting/optimizer.py", line 447, in get_objective_function
    projected_vec = dir_vec * self.projected_dist[:, None]  # (6, 3)
ValueError: operands could not be broadcast together with shapes (5,3) (10,1) 
```

silencht, 2025-07-31:
@suzhelearning 
hi, could you run `simulaiton deploy` with inspire hand sucessful?

suzhelearning, 2025-08-01:
> [@suzhelearning](https://github.com/suzhelearning) hi, could you run `simulaiton deploy` with inspire hand sucessful?

NO, the same is as Physical Deployment,

le-wei, 2025-08-01:
@suzhelearning 
When testing the Inspire hand in the simulation, is it possible to display output logs to help identify the cause of issues? Please try using the latest code. Thank you.

amul1umt, 2025-08-28:
> [@chaingee](https://github.com/chaingee) /inspire_hand_ws/inspire_hand_sdk$ python example/Headless_driver_double.py, /avp_teleoperate/teleop$ python teleop_hand_and_arm.py --arm=G1_23 --hand=inspire1. Do you run these commands on the robot's computer or locally?
cc @chaingee @silencht 

This thread is extremely helpful. I want to as for further clarification. I am a newbie to Unitree world. Here is what I am trying to set up (any help is highly appreciated). I am working with G1 edu ultimate D, and I have ubuntu 22.04 installed on my PC.
- Connect my PC to robot's PC2 via ethernet. And, connect my PC to wifi. For installing the inspire_hand_ws repo, does that need to be installed on my ubuntu PC or robot's PC2? 
- For the correct network setup, do i need to connect robot's PC2 to wifi too (especially for teleoperation?) - the wifi adapter doesn't turn on (I even tried connecting a usb wifi adapter but it won't turn on) - i am assuming it is blocked for some reason. I tried all the commands with sudo rfkill to unblock it, but it won't work. The question is do I even need to connect the robot to wifi? for any updates, or installing anything on the robot - i guess it does need internet connection, but do I need to install anything? As much I understand, I only need to transfer image_server.py and run that - that runs alright.
- While installing inspire_hand_ws on robot's PC2, I run into cyclonedds issue with unitree_sdk2_python repo - i install cyclonedds as recommended in the unitree_sdk2_python library, and then install inspire_hand_sdk as given in the instruction. It installs fine without error, but when I try to run any of the example code, it gives me PyQt related errors, and I am unable to update PyQt5 (progresses to an extent and then gets stuck for hours). 
- While installing inspire_hand_ws on my PC - I am able to complete installation in inspire_hand_ws, and run example fine (python inspire_hand_sdk/example/dds_publisher.py), but when I go to xr_teleoperate and modify the inspire_robot_hand.py file as recommended above and run the command, it runs into numpy conflicts. 
what is the right set of sequence to follow to get this operational? 

Can anyone also share a few examples of publishing commands via say dds_publisher.py to have hands move?

silencht, 2025-08-29:
> Connect my PC to robot's PC2 via ethernet. And, connect my PC to wifi. For installing the inspire_hand_ws repo, does that need to be installed on my ubuntu PC or robot's PC2?

installed on your robot's PC2

> For the correct network setup, do i need to connect robot's PC2 to wifi too (especially for teleoperation?)

NO. PC2's WiFi is not need.

>  It installs fine without error, but when I try to run any of the example code, it gives me PyQt related errors, and I am unable to update PyQt5 (progresses to an extent and then gets stuck for hours).

please go to repo: https://github.com/NaCl-1374/inspire_hand_ws , and write a issue on this. I think you should use headless mode.

> While installing inspire_hand_ws on my PC - I am able to complete installation in inspire_hand_ws, and run example fine (python inspire_hand_sdk/example/dds_publisher.py), but when I go to [...]

numpy version always should be  `<2.0.0`

amul1umt, 2025-09-02:
@silencht Thank you, for looking at the other thread as well. I made past several of these errors but stuck on being able to move the arm and hand, and being able to see anything in XR screen.