unitreerobotics/xr_teleoperate #49: G1 with Quest 3 hand positioning not working properly

kuehnrobin, 2025-05-20:
Hello Together!

I implemented the avp_teleoperate setup on an G1 with 29 DoF and Dex3-1 hands using an Quest 3 instead of the AVP. I am communicating with the Quest 3 with an wired connection and an abd reverse port.

Sometimes the setup works really well, but sometimes the robot moves the arms to very strange positions, when starting the teleoperate_hand_and_arm.py, which do not correspond to the positions of the person with the VR-headset at all. This lead to the robots left arm selft colliding with the robots torso, when starting teleoperate_hand_and_arm.py today. Moreover, sometimes I am having the problem, that the thumbs are not properly recognized and do not move accordingly to the real thumbs.

Has anyone ever had similar problems? 
Dose avp_teleoperate run stable with the AVP or do similar issues occur?

Currently I am trying to figure out whether the problem is based in the IK and dex-retargeting or if the hand tracking of the quest 3 dosen't work as well as the AVP or if there are problems when accessing the hand tracking data from the Quest 3. 

Thank you guys for your help!

Best regards,

Robin

silencht, 2025-05-21:
### Question 1
Based on the information you've provided, it's still not possible to fully pinpoint the root cause of the issue.
When we use the Apple Vision Pro, we don't see any tracking errors. However, when using the PICO headset, we do occasionally notice both arms drifting in the same direction at the same time (is that similar to the issue you're seeing?).

[The wrist pose is used after being transformed into the local coordinate system of the head](https://github.com/unitreerobotics/avp_teleoperate/blob/728ffa0f9ff309e52f5cfbc16fea5e5c0e146785/teleop/open_television/tv_wrapper.py#L74-L102). It’s a relative pose, and can be expressed like this:
`T_head_wrist = inverse(T_world_head) * T_world_wrist = T_head_world * T_world_wrist`

I’m honestly not sure why this kind of drift happens. It might be caused by the difference in odometry / slam calculation principles between AVP and Quest / Pico.

That said, based on some articles and also from our own experience, the Apple Vision Pro does seem to provide higher accuracy and more stable tracking compared to other headsets.

> Reference documentation
> - https://www.researchgate.net/publication/384764536_Comparing_Tracking_Accuracy_in_Standalone_MR-HMDs_Apple_Vision_Pro_Hololens_2_Meta_Quest_3_and_Pico_4_Pro
> - https://dl.acm.org/doi/10.1145/3636534.3696215
> - https://developers.meta.com/horizon/design/hands

### Question 2
Besides the known issue of poor hand tracking accuracy on Quest 3, the current implementation is using the vector method from the dex-retargeting library. This approach solves inverse kinematics for each finger independently, without considering how fingers move in coordination.

So if you're aiming for gestures like fingertip pinching, the dex-pilot method may give better performance. We’re considering adding a configuration option to switch to that method in the future — but there’s no confirmed timeline yet.

kuehnrobin, 2025-05-21:
Hi, thank you for your fast response!

Question 1: Yes kind of, sometimes there is a drifting effect but sometimes when I start teleoperate_hand_and_arms.py the hand tracking seems to be completely of and the robot just raises the arms up to the sky, although the operator has his hands still in the starting pose.  In this cases the pose is about 40 cm different from the correct pose. This dose only happen sometimes, I would say about 30% of the time when I start the program, sometimes everything works fine (also about 30% of the time) and in the other cases, teleoperating the arms work fine, but Question 2 occurs and the thumbs move in a strange way or cannot be controlled at all, while the other fingers move fine.
The sub-optimal tracking, compared to the AVP could explain the finger tracking issues. The completely wrong position, which sometimes occurs when starting the problem could also come from this, but I never experienced this extreme positioning error when testing vr games on the quest 3.
I will try to get my hands on an AVP, but for now I have to work with the quest.

Have you ever experienced with using haptic gloves for VR to increase the hand tracking and get some haptic feedback while teleoperating? 

Question 2: Thank you for the advice! I will try out the dex-pilot method!

Thank you so much for your help!

silencht, 2025-05-21:
Based on your description, there is a relatively high probability of the issue occurring (around 30%). It is recommended to comment out all code related to `arm_ctrl` (to avoid hardware control and prevent potential damage to the robot). 

And, print the values of the variables `head_rmat`, `left_wrist`, `right_wrist`, `left_hand`, and `right_hand` obtained from `tv_wrapper.get_data()` to analyze the cause of the pose changes.

For a deeper investigation, you can also print, observe, and analyze the execution process of the `get_data` function in the `TeleVisionWrapper` class, located in [tv\_wrapper.py](https://github.com/unitreerobotics/avp_teleoperate/blob/main/teleop/open_television/tv_wrapper.py). Try to identify the root cause of the pose data issue as thoroughly as possible.

> Have you ever experienced with using haptic gloves for VR to increase the hand tracking and get some haptic feedback while teleoperating?

We haven't found a particularly suitable glove for now.