unitreerobotics/xr_teleoperate #54: How to improve the robot's ability to track hand speed？

yzhperseverance, 2025-06-17:
Currently, I can perform teleoperation relatively smoothly, but I’ve noticed that when I move my arms quickly, the robot is unable to track the speed of my hand accurately. I’ve tried increasing the update/control frequency, but it doesn’t seem to help. Do you have any suggestions?

silencht, 2025-06-17:
Hello! you can refer to this part of the code. Increasing the value of `velocity_limit` **appropriately** will make the dual-arm motion tracking faster. 
https://github.com/unitreerobotics/avp_teleoperate/blob/728ffa0f9ff309e52f5cfbc16fea5e5c0e146785/teleop/robot_control/robot_arm.py#L144-L178

The above control code was modified based on the code from [https://github.com/Dingry/BunnyVisionPro](https://github.com/Dingry/BunnyVisionPro).

------ 
Alternatively, you can also rewrite this part of the control code using a better upsampling or trajectory optimization algorithm. For example, the code in diffusion_policy repo is very beautiful written.
https://github.com/real-stanford/diffusion_policy/blob/main/diffusion_policy/common/pose_trajectory_interpolator.py

silencht, 2025-06-17:
To add on, you can also **appropriately** adjust the `kp` and `kd` values of the dual-arm motors to generate more torque for improving speed. 
https://github.com/unitreerobotics/avp_teleoperate/blob/728ffa0f9ff309e52f5cfbc16fea5e5c0e146785/teleop/robot_control/robot_arm.py#L64-L67

For the roles of the `kp` and `kd` parameters, you can refer to relevant documentation such as [https://support.unitree.com/home/en/Motor\_SDK\_Dev\_Guide](https://support.unitree.com/home/en/Motor_SDK_Dev_Guide).

closed 2025-07-14 by silencht