unitreerobotics/xr_teleoperate #56: 无法获取realsense相机的数据

BJHYZJ, 2025-06-17:
您好，感谢好工作的开源，我遇到一个问题，G1中似乎没有/dev/video*，但我在unitree 手机app中实际上是能看到相机的 rgb信息的，但运行image_server.py 报错，我还遇到了即使我安装了 librealsense之后，打开 realsense-viewer 发现不了设备的问题。

```
unitree@ubuntu:~$ cd image_server/
unitree@ubuntu:~/image_server$ python image_server.py 
{'fps': 30, 'head_camera_type': 'opencv', 'head_camera_image_shape': [480, 1280], 'head_camera_id_numbers': [0]}
[ WARN:0@0.072] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0@0.072] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index
[Image Server] Camera 0 Error: Failed to initialize the camera or read frames. Exiting...
[Image Server] Head camera 0 resolution: 0.0 x 0.0
[Image Server] Image server has started, waiting for client connections...
[Image Server] Head camera frame read is error.
[Image Server] The server has been closed.
unitree@ubuntu:~/image_server$ 

```

如果您能提供帮助，万分感谢！

silencht, 2025-06-18:
正常情况下，将realsense相机的线插到g1颈部的usb-c插口后，在jetson nx系统内使用`ls /dev/video` + `tab`键命令是可以看到6个video端口的:
```bash
(tv) unitree@ubuntu:~/image_server$ ls /dev/video
video0  video1  video2  video3  video4  video5
```
如果存在上述 video* 端口，那么只需要适当修改 `image_server.py`内相机id序号即可抓取图像。例如，直接使用v4l2驱动抓取rgb图像：
```python
if __name__ == "__main__":
    config = {
        'fps': 30,
        'head_camera_type': 'opencv',
        'head_camera_image_shape': [480, 1280],   #将此处的分辨率修改为realsense支持的分辨率
        'head_camera_id_numbers': [0],  #将此处的序号0尝试修改为其他video序号
       # 'wrist_camera_type': 'opencv',
       # 'wrist_camera_image_shape': [480, 640],  # Wrist camera resolution
       # 'wrist_camera_id_numbers': [2, 4],
    }

    server = ImageServer(config, Unit_Test=False)
    server.send_process()
```

如果无法找到`/dev/video*`端口，请尝试检查并告知我们您的具体接线情况。

BJHYZJ, 2025-06-18:
同时我也非常好奇为什么不直接用realsense的相机，而是额外外接一个双目相机呢

silencht, 2025-06-18:
> [@silencht](https://github.com/silencht) 是的，您说的没错，我额外插入一个相机到G1 颈部之后通过 `ls /dev/video + tab` 可以看到 6个 video端口。但实际上我是想获取 G1头部自带的 realsense d435i的图像信息，但它似乎和 直接 在 G1 颈部 插入一个相机不同，我无法直接 通过 `ls /dev/video + tab` 找到它，但我在unitree explore 手机 app里面可以看到这个图像信息。
> 
> 所以我想知道，如何获取G1 头部的图像信息呢

根据您的描述信息，可能您的g1设备是靠前的批次。其realsense相机是直接内部连接到pc1运控板。如果您确实要使用该设备，我建议您拆开g1头盖后，拔下相机自带线缆，然后使用新的线缆连接至颈部USB-C接口（该接口为Jetson NX开发板所有）。

> 同时我也非常好奇为什么不直接用realsense的相机，而是额外外接一个双目相机呢

可以直接使用realsense相机（目前发货的g1版本，就是将realsense相机的接线外接，用户可以手动将它插至Jetson NX接口上）。额外外接一个双目相机的目的是双目图像可以在VR头显设备中形成立体视角，沉浸感更强。