unitreerobotics/xr_teleoperate #57: 我在运行程序的时候一直显示[G1_29_ArmController] Waiting to subscribe dds...

douzi999, 2025-06-18:
我根据文档操作，xr和图像接收器已经配置好了，就在最后一步(tv) unitree@Host:~/avp_teleoperate/teleop$ python teleop_hand_and_arm.py  一直显示正在连接(tv) xc@xc:~/avp_teleoperate-main/teleop$ python teleop_hand_and_arm.py 
args:Namespace(arm='G1_29', frequency=30.0, hand=None, record=False, task_dir='./utils/data')
Image client has started, waiting to receive data...
Initialize G1_29_ArmController...
Serving file:///home/xc/avp_teleoperate-main/teleop at /static
Visit: https://vuer.ai?grid=False
[G1_29_ArmController] Waiting to subscribe dds...
[G1_29_ArmController] Waiting to subscribe dds...
[G1_29_ArmController] Waiting to subscribe dds...
[G1_29_ArmController] Waiting to subscribe dds...
[G1_29_ArmController] Waiting to subscribe dds...
然后手机端上会报错：底层固件通信故障
一直查不到原因，所以咨询怎么解决

silencht, 2025-06-18:
底层固件通信故障可能是由于切入Debug模式导致，属于正常现象。

持续输出`[G1_29_ArmController] Waiting to subscribe dds...`的原因是当前设备并未与机器人建立好dds通信。您可以先使用下面的手段进行排查：
1. 在终端中输入`cyclonedds ps`，如果输出一系列机器人的话题信息，说明是正常的；如果输出为空，说明并未在局域网内订阅到dds消息：那么请检查当前开发设备是否与机器人处于同一局域网下（可使用`ping 192.168.123.164`等命令检查）。
2. 第1步正常的情况下，在终端中输入`cyclonedds subscribe rt/lowstate`，如果输出持续不断地机器人所有关节数据信息，说明是正常的；如果输出为空，说明存在问题。

Funnn1k, 2025-06-20:
能ping通，但是cyclonedds ps返回为空，cyclonedds subscribe rt/lowstate返回 No types could be discovered over XTypes, no dynamic subsciption possible是什么原因造成的呢？

silencht, 2025-06-20:
@Funnn1k 
能ping通，说明确实在同一局域网内。`cyclonedds ps`返回为空，说明不能正确订阅到g1发布的dds消息。
建议您重启机器几分钟后，不要进行其他操作，再次尝试使用`cyclonedds ps`进行话题订阅进行检查。另外，使用Unitree App连接到机器人，查看各类服务状态是否正常。

Funnn1k, 2025-06-20:
重启后cyclonedds ps返回依然为空，Unitree App显示服务状态皆为正常。这种情况和pinocchio版本有关吗，用conda-forge目前默认的pinocchio版本是3.2.0

silencht, 2025-06-20:
@Funnn1k helloworld例程只是简单测试通信流程，需要同时运行publisher.py。请您查看一下https://github.com/unitreerobotics/unitree_sdk2_python/tree/master/example/g1 处的示例能否正常运行。如果还是无法通信，请提交工单，联系技术支持给您进一步排查dds通信问题。

Funnn1k, 2025-06-20:
@silencht 该示例可以跑通

silencht, 2025-06-23:
@douzi999 helloworld例程运行publisher和subscriber可以接收到信息：机器本身的DDS通信是没有问题的。cyclonedds ps返回依然为空：跨机器之间的DDS通信存在问题。
另，lowlevel能跑通的意思是？可以进行动作吗？如果lowlevel能跑通的话，那么通信本身应该是可以连接的。请提交工单联系进一步支持。

douzi999, 2025-06-23:
我只能运行成功helloworld例程，但是在运行g1下面的g1_audio_client_example程序，报错：(tv) xc@xc:~/unitree_sdk2_python/example/g1/audio$ python3 g1_audio_client_example.py wlx245a5f1248e6
[ClientStub] send request error. id: 5370914409302
debug GetVolume:  (3102, None)
[ClientStub] send request error. id: 5381036289043

douzi999, 2025-06-23:
> 啊，是这样的，目前暂不支持无线连接下的调试功能。 [image] 如果用户PC使用WiFi连接到PC2的无线网卡，虽然用户PC和PC2组合成了局域网，但是此时却和PC1产生了隔离（PC2使用有线网卡和PC1组建局域网）。

好的，非常感谢您的解答！

XiaoWZENG, 2025-06-28:
> [@silencht](https://github.com/silencht) [@douzi999](https://github.com/douzi999) 我们这个情况倒不是无线连接导致的，看到运行example的时候需要指定有线网卡，突然想到是不是因为我的host有两个以太网接口，导致通讯没有采用真正和机器人连接的那个接口，在代码里改了一下就解决了。

请问这个具体是如何解决的呢，我联网线也有这个问题能ping通但是订阅不到dds话题，同时主机pc2都能正常发送订阅。

closed 2025-07-18 by silencht