unitreerobotics/xr_teleoperate #59: 使用pico进行遥操作，g1的动作和我的动作是上下相反的

douzi999, 2025-06-24:
正在使用pico进行遥操作，但是g1的动作和我的动作是上下相反的，我双手向下运动，但是g1双手向上运动，并且我单独动一个手臂，g1没有任何动作，只能双臂一起运动，请问是代码还需要修改、操作上出了问题还是pico不是很适配导致的？（我只用了一个g1自带的realsense相机）

ronjian, 2025-06-27:
> This is a simulation environment compatible with this repository : https://github.com/unitreerobotics/unitree_sim_isaaclab. You can try using it first to rule out any hardware-related issues. When using the simulation environment, please make sure to switch to the newVuer test branch first.
> 
> 这是一个与本仓库兼容的仿真环境：[https://github.com/unitreerobotics/unitree_sim_isaaclab。](https://github.com/unitreerobotics/unitree_sim_isaaclab%E3%80%82) 您可以先用它来排除任何与硬件相关的问题。使用仿真环境时，请注意先切换到newVuer测试分支。

Is there any guide to run avp_teleoperate with unitree_sim_isaaclab?

silencht, 2025-06-28:
Install and launch the simulation environment according to the README of unitree_sim_isaaclab.
Then, simply start the program using the appropriate avp_teleoperate launch arguments that match the robot configuration used in the unitree_sim_isaaclab simulation.

for example:
```python
python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex1-Joint    --enable_gripper_dds --robot_type g129
```
g1 29dof + gripper simulation, you can start avp-teleoperate application with this command:
```python
python teleop_hand_and_arm.py --xr-mode=hand --ee=gripper --sim
```

closed 2025-07-14 by silencht