unitreerobotics/xr_teleoperate #64: H1 Teleoperation 误差问题

kaipengfang, 2025-07-03:
我的设备：
Meta Quest 3 + H1 edu 版 + Inspire Hands

问题：
1. 机器人动作和手始终有差距，双手并拢后，机器人双手难以并拢
2. Inspire Hands手指始终无法并拢，操作员食指和大拇指并拢，Inspire Hands食指和大拇指存在大约5 cm的距离
3. Manipulation困难，虽然机器人大致的动作可以模仿，但在距离上不够精准，总不能达到操作员所达到的位置

可能原因：
1. 没有精确的初始限位：虽然已经按照官网视频所示将左右手偏航电机调至限位开机，但是12,13,16,17这几个肩部电机的初始位置比较模糊
2. Meta Quest3 没有 Vision Pro精确
3. H1 edu版本双臂自由度少2个手腕自由度，导致IK无解或不精确

希望达到的效果：Open-Television中可以精确拿起耳塞，操作电钻等精细灵巧操作，网址：https://robot-tv.github.io/

不知道以上哪个问题最为严重，希望有一定的参考建议，谢谢!

silencht, 2025-07-07:
最新的Release版本，将手部重映射算法从Vector切换至DexPilot算法。DexPilot算法考虑了所有手指之间的约束。因此，相较于Vector算法（仅考虑单根手指的重映射），指尖对捏等动作会更加容易实现。请您尝试，看是否情况有所好转。

如果有新的问题请继续在此issue下回复，我将为您持续提供支持。

kaipengfang, 2025-07-14:
你好，手指确实有很大的改善，感谢！但我们发现另外一个问题，我们是老款H1原本单手臂4自由度，但加装Inspire Hands后手腕多了一个roll的自由度，但是我看代码中的H1是老款的，请问应该如何修改实现适配手腕滚动的控制？

silencht, 2025-07-15:
### teleop/robot_control/robot_arm_ik.py
```python
class H1_ArmIK:
    def __init__(self, Unit_Test = False, Visualization = False):
        np.set_printoptions(precision=5, suppress=True, linewidth=200)

        self.Unit_Test = Unit_Test
        self.Visualization = Visualization

        if not self.Unit_Test:
            self.robot = pin.RobotWrapper.BuildFromURDF('../assets/h1/h1_with_hand.urdf', '../assets/h1/')
        else:
            self.robot = pin.RobotWrapper.BuildFromURDF('../../assets/h1/h1_with_hand.urdf', '../../assets/h1/') # for test

        self.mixed_jointsToLockIDs = [
                                        "right_hip_roll_joint",
                                        "right_hip_pitch_joint",
                                        "right_knee_joint",
                                        "left_hip_roll_joint",
                                        "left_hip_pitch_joint",
                                        "left_knee_joint",
                                        "torso_joint",
                                        "left_hip_yaw_joint",
                                        "right_hip_yaw_joint",

                                        "left_ankle_joint",
                                        "right_ankle_joint",

                                        "L_index_proximal_joint",
                                        "L_index_intermediate_joint",
                                        "L_middle_proximal_joint",
                                        "L_middle_intermediate_joint",
                                        "L_ring_proximal_joint",
                                        "L_ring_intermediate_joint",
                                        "L_pinky_proximal_joint",
                                        "L_pinky_intermediate_joint",
                                        "L_thumb_proximal_yaw_joint",
                                        "L_thumb_proximal_pitch_joint",
                                        "L_thumb_intermediate_joint",
                                        "L_thumb_distal_joint",
                                        
                                        "R_index_proximal_joint",
                                        "R_index_intermediate_joint",
                                        "R_middle_proximal_joint",
                                        "R_middle_intermediate_joint",
                                        "R_ring_proximal_joint",
                                        "R_ring_intermediate_joint",
                                        "R_pinky_proximal_joint",
                                        "R_pinky_intermediate_joint",
                                        "R_thumb_proximal_yaw_joint",
                                        "R_thumb_proximal_pitch_joint",
                                        "R_thumb_intermediate_joint",
                                        "R_thumb_distal_joint",

                                        # "left_hand_joint",
                                        # "right_hand_joint"
                                    ]

        self.reduced_robot = self.robot.buildReducedRobot(
            list_of_joints_to_lock=self.mixed_jointsToLockIDs,
            reference_configuration=np.array([0.0] * self.robot.model.nq),
        )

        # self.reduced_robot.model.addFrame(
        #     pin.Frame('L_ee',
        #               self.reduced_robot.model.getJointId('left_elbow_joint'),
        #               pin.SE3(np.eye(3),
        #                       np.array([0.2605 + 0.05,0,0]).T),
        #               pin.FrameType.OP_FRAME)
        # )
[... 182 lines cut ...]

            sol_q = self.opti.debug.value(self.var_q)
            self.smooth_filter.add_data(sol_q)
            sol_q = self.smooth_filter.filtered_data

            if current_lr_arm_motor_dq is not None:
                v = current_lr_arm_motor_dq * 0.0
            else:
                v = (sol_q - self.init_data) * 0.0

            self.init_data = sol_q

            sol_tauff = pin.rnea(self.reduced_robot.model, self.reduced_robot.data, sol_q, v, np.zeros(self.reduced_robot.model.nv))

            logger_mp.error(f"sol_q:{sol_q} \nmotorstate: \n{current_lr_arm_motor_q} \nleft_pose: \n{left_wrist} \nright_pose: \n{right_wrist}")
            if self.Visualization:
                self.vis.display(sol_q)  # for visualization

            # return sol_q, sol_tauff
            return current_lr_arm_motor_q, np.zeros(self.reduced_robot.model.nv)
```

此时上述逆运动学类 解算出10个关节（原来是8个）。您需要将额外的两个手腕关节控制量发送到手腕电机。

另外，解算的10个关节已经不同于以前的8个，因此传递至`robot_arm.py`进行控制时，您需要注意将它的结果手动截断。

kaipengfang, 2025-07-15:
您好，我大致理解您的意思。但我主要不清除的是IK得到10个q后，如何发送给机器人的手腕电机？

我使用cyclonedds subscribe rt/lowstate 指令查看state，发现只有20个，不包括手腕的两个电机。是否意味着我的手腕电机没有开启？