unitreerobotics/xr_teleoperate #65: G1 Teleoperation + inspire hands RH56DFTP-2R/2L

FranRFH, 2025-07-04:
Hello, I am looking to teleoperate the Unitree G1 robot in which I follow the tutorial of avp_teleoperate and I have some doubts that are the following:
1- If or if it is necessary to have the 3 cameras? one that would be for the head and two others for the wrists.
2- Can I teleoperate the robot with the hands that I stated in the title?
3- The form of teleoperation with the Apple Vision Pro, is it necessary to have another tool for if I move a phalanx of a finger the robot do the same?

silencht, 2025-07-05:
> 1- If or if it is necessary to have the 3 cameras? one that would be for the head and two others for the wrists.

The head camera is necessary, while the wrist camera is used for data recording. In fact, if you place the PICO device around the neck (instead of wearing it on the head), the head camera may not even be required.

> Can I teleoperate the robot with the hands that I stated in the title?

Please refer to this https://github.com/unitreerobotics/avp_teleoperate/issues/48 for more details. 

> The form of teleoperation with the Apple Vision Pro, is it necessary to have another tool for if I move a phalanx of a finger the robot do the same?

No additional tools are required— the Apple Vision Pro provides wrist poses (for controlling the arms) and finger joint poses (for controlling the dexterous hands).

FranRFH, 2025-07-05:
Well, I find your information very useful but I'm going to ask you some
questions so that I can be sure:
1- So with the issue of the 3 cameras that you ask if or if I have to have
them to teleoperate?
2- If I teleoperate with Apple Vision Pro or Meta Quest or PICO4, I will
not need any more tools, because I know that Apple Vision Pro as you say
already makes a pose_tracking of the hand but the other two tools also?

silencht, 2025-07-17:
> Hi, I just tried the code you provide above to teleoperate the Unitree G1 robot with the DFTP hands but it is giving me an error like no moduled named inspire_sdkpy, so my question is the following. How do I install that package?

https://github.com/NaCl-1374/inspire_hand_ws

> I ask you a question, I want to be able to teleoperate the Unitree G1 robot with this type of haptic gloves https://www.senseglove.com/enhancing-virtual-reality-with-haptic-gloves-the-senseglove-nova/, is it possible?

Theoretically, it's feasible.
In practice, you need to consider the following aspects: the hand motion data output from the haptic gloves (including its data type), the retargeting algorithm, and the control code for the dexterous hand. We use the [dex-retargeting](https://github.com/dexsuite/dex-retargeting) library to handle this part.

If you plan to use the [dex-retargeting](https://github.com/dexsuite/dex-retargeting) library to solve this problem, you may need to think about how to convert the data from the motion capture glove into the input data format expected by the dex-retargeting library (simply put, it typically requires the position of each fingertip relative to the wrist).

You can refer to https://github.com/unitreerobotics/xr_teleoperate/issues/55 for more details.