unitreerobotics/xr_teleoperate #74: xr_teleoperate采集的数据集在IL_lerobot中使用的问题

PEAGREEABLB, 2025-07-11:
报错信息：
(unitree_lerobot) wcy@wcy:~/unitree_IL_lerobot$ python unitree_lerobot/utils/sort_and_rename_folders.py --data_dir $HOME/unitreedatasets/datatest
The folders have been successfully renamed.
(unitree_lerobot) wcy@wcy:~/unitree_IL_lerobot$ python unitree_lerobot/utils/convert_unitree_json_to_lerobot.py --raw-dir $HOME/unitreedatasets/datatest --repo-id your_name/g1_grabcube_double_hand --robot_type Unitree_G1_Dex3 --push_to_hub
Loading Cache Json:   0%|                                                                                                                                       | 0/16 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/wcy/unitree_IL_lerobot/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py", line 345, in <module>
    tyro.cli(json_to_lerobot)
  File "/home/wcy/anaconda3/envs/unitree_lerobot/lib/python3.10/site-packages/tyro/_cli.py", line 229, in cli
    return run_with_args_from_cli()
  File "/home/wcy/unitree_IL_lerobot/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py", line 326, in json_to_lerobot
    dataset = populate_dataset(
  File "/home/wcy/unitree_IL_lerobot/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py", line 277, in populate_dataset
    json_dataset = JsonDataset(raw_dir, robot_type)
  File "/home/wcy/unitree_IL_lerobot/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py", line 61, in __init__
    self._init_cache()
  File "/home/wcy/unitree_IL_lerobot/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py", line 95, in _init_cache
    with open(json_path, 'r', encoding='utf-8') as jsonf:
FileNotFoundError: [Errno 2] No such file or directory: '/home/wcy/unitreedatasets/datatest/episode_0000/audios/data.json'

采集数据的时候是在isaacsim中采集的，没有audio

PEAGREEABLB, 2025-07-11:
audio和depth文件夹都是空的
colors文件夹有数据

hengguo-boop, 2025-07-11:
convert_unitree_json_to_lerobot 目前没有用到 audio和depth 空的是正常的

FileNotFoundError: [Errno 2] No such file or directory:  对于这个问题 

因为你现在保存的路径是/home/wcy/unitreedatasets/datatest/ 你需要添加一个task的文件夹
数据集的构造方式大概是这样 $HOME/datasets/task_name/episode_*
```
datasets/                                     # Dataset folder
    task_name /   # Task name
    │
    ├── episode_0001                # First trajectory
    │    ├──colors/                       # Image information
    │    └──data.json                   # State and action information
    ├── episode_0002
    ├── episode_...
    ├── episode_xxx

```
你的运行命令应该是这样 
```
python unitree_lerobot/utils/convert_unitree_json_to_lerobot.py --raw-dir $HOME/unitreedatasets/datatest    --repo-id your_name/g1_grabcube_double_hand --robot_type Unitree_G1_Dex3 --push_to_hub
```

如果还存在问题，请随时提问。

PEAGREEABLB, 2025-07-11:
这是我修改之后的保存

closed 2025-07-11 by PEAGREEABLB