unitreerobotics/xr_teleoperate #76: vuer server often crash

congleetea, 2025-07-14:
vuer/server.py often crashed:
```
 assert self.CURRENT_WS_ID in self.vuer.ws, "Websocket session is missing."
AssertionError: Websocket session is missing.
```
Why the vuer server often crash when Running the server just seconds?

congleetea, 2025-07-16:
1. XR is Meta Quest3;
2. aiohttp version: 
 ```
> conda activate tv
> pip list | grep aiohttp
aiohttp                              3.10.5
aiohttp-cors                         0.8.1
```
3. xr_telooperate I used is 
```
*   commit c43facb0439d6c9b96798b8f4b899e03bcf3fa9f (HEAD -> main)
|\  Merge: 728ffa0 0e4a287
| | Author: silencht <silencht@qq.com>
| | Date:   Mon Jul 7 18:07:32 2025 +0800
| | 
| |     Merge pull request #66 from unitreerobotics/newVuer
| |     
| |     New vuer
```
4. the Wifi router is 2.4 GHz / 5 GHz, I will disable 2.4G to recheck. thx.

silencht, 2025-07-23:
@wtingsamDenso 
It seems that the error is not caused by the certificate issue. Therefore, I don't think installing the certificate will solve the problem, but you can give it a try.

And, Please try running the code from this repository [televuer](https://github.com/silencht/televuer) and check if it can capture data properly. This is the most core library for capturing XR data. It can eliminate the influence of other components in xr-teleoperate.

gawinwong, 2025-07-29:
> 1. XR is Meta Quest3;XR 是 Meta Quest3；
> 2. aiohttp version:  aiohttp 版本：
> 
> ```
> > conda activate tv
> > pip list | grep aiohttp
> aiohttp                              3.10.5
> aiohttp-cors                         0.8.1
> ```
> 
> 3. xr_telooperate I used is我使用的 xr_telooperate 是
> 
> ```
> *   commit c43facb0439d6c9b96798b8f4b899e03bcf3fa9f (HEAD -> main)
> |\  Merge: 728ffa0 0e4a287
> | | Author: silencht <silencht@qq.com>
> | | Date:   Mon Jul 7 18:07:32 2025 +0800
> | | 
> | |     Merge pull request #66 from unitreerobotics/newVuer
> | |     
> | |     New vuer
> ```
> 
> 4. the Wifi router is 2.4 GHz / 5 GHz, I will disable 2.4G to recheck. thx.路由器支持 2.4 GHz / 5 GHz，我将禁用 2.4G 后重新检查。谢谢。

> 1. XR is Meta Quest3;
> 2. aiohttp version:
>
> ```
> aiohttp                              3.10.5
> aiohttp-cors                         0.8.1
> ```
>
> 3. xr_telooperate I used is
>
> ```
> *   commit [...]

I'm using the same device as you and also encountered a WebSocket disconnection issue. Have you resolved it now?

gawinwong, 2025-07-29:
> 1. XR is Meta Quest3;
> 2. aiohttp version:
>
> ```
> aiohttp                              3.10.5
> aiohttp-cors                         0.8.1
> ```
>
> 3. xr_telooperate I used is
>
> ```
> *   commit [...]

I'm using the same device as you and also encountered a WebSocket disconnection issue. Have you resolved it now?

wtingsamDenso, 2025-07-29:
@silencht 
Just to update everyone here.
Hi, it turns out it does not matter if I uploaded  "mkcert/rootCA.pem" to AVP, it still works. 
The problem was due to the version of vuer. 0.0.60 does not work. If I used the version of vuer in h1 branch, it works. (Version was 0.0.3x)
> And, Please try running the code from this repository [televuer](https://github.com/silencht/televuer) and check if it can capture data properly. This is the most core library for capturing XR data. It can eliminate the influence of other components in xr-teleoperate. 

It works, however, this git [televuer](https://github.com/silencht/televuer) is quite difficult for testing, since when you do "Enter VR", you cant see anything on your VR screen. So I tested it by simply doing share memory of a normal camera.

silencht, 2025-07-30:
> Hi, it turns out it does not matter if I uploaded "mkcert/rootCA.pem" to AVP, it still works.
> The problem was due to the version of vuer. 0.0.60.

A very puzzling issue indeed. We’re testing our app with the Apple Vision Pro, and it behaves correctly under vuer 0.0.60. Our current approach is to use a self-signed certificate: when accessing the URL on port 8012, we ignore the HTTPS warning to force a connection to the server—exactly as described in the README for v1.2. Consequently, we’ve dropped the step of installing the certificate on the Vision Pro itself. [v0.5](https://github.com/unitreerobotics/xr_teleoperate/releases/tag/v0.5) is used 0.0.32rc7.

> It works, however, this git [televuer](https://github.com/silencht/televuer) is quite difficult for testing, since when you do "Enter VR", you cant see anything on your VR screen. So I tested it by simply doing share memory of a normal camera.

Yes, the interaction problem does exist. Once you enter VR and the data stream is captured, you can immediately remove the headset. By simply watching the terminal to see whether pose data keeps refreshing, you can tell whether the pipeline is running normally.