unitreerobotics/xr_teleoperate #80: 适配G1 (23 DoF)的3D打印件

ZhaoXiangXML, 2025-07-16:
现在https://github.com/unitreerobotics/xr_teleoperate/tree/main/hardware目录下的3D打印文件，在我的G1 (23 DoF)上不合适，其中：

1. 手腕部分太粗了
2. 头部的两个大螺丝孔的间距比机器人头部的两个螺丝孔间距大了10mm左右
3. 推荐的M4x14 screws拧不进螺丝孔

请问上述3D打印文件和螺丝推荐有更新吗？谢谢

zmingen, 2025-07-16:
我们也遇到了同样的问题G1(29 DOF), 头部相机支架尺寸不合适

Heven-Pan, 2025-07-24:
@ZhaoXiangXML 您好，请问您改的头部支架图纸可用嘛，想求一个改后图纸，谢谢！

silencht, 2025-07-24:
hello, eyeryone! Sorry to keep you waiting! The new camera mount download link is now live.

**reference image**
| New_binocular_camera_mount | New_binocular_camera_cover_plate |
|--------|-----------|
| [image] | [image] |
|[New_binocular_camera_mount.STEP](https://drive.google.com/file/d/11p7D8RLxE3OCz78Q4atTQbIrFoFU1o05/view?usp=drive_link)|[New_binocular_camera_cover_plate.STEP](https://drive.google.com/file/d/16lQySJeS3b_ptlyJp1Xsuo4BnlpHQdUf/view?usp=drive_link)|

@ZhaoXiangXML @zmingen @jkxdl @Heven-Pan