unitreerobotics/xr_teleoperate #83: I don't see in the terminal controller started,start main loop

FranRFH, 2025-07-16:
Hello, I am having a problem that when I launch the command python sim_main.py --device cpu --enable_cameras --task Isaac-PickPlace-Cylinder-G129-Dex3-Joint --enable_dex3_dds --robot_type g129. I don't see in the terminal the following controller started, start main loop..., I attach an image of my terminal

FranRFH, 2025-07-16:
I could access the software they give you, but it does not show me the image, here is an image.

[image]

le-wei, 2025-07-17:
@FranRFH Please click on the main Sim interface to enter the play (or running) state.

FranRFH, 2025-07-17:
Yes, I have the simulation running but I can't see anything. Attached image.

[image]

FranRFH, 2025-07-17:
Ok, thank you very much!!

closed 2025-07-18 by silencht