unitreerobotics/xr_teleoperate #85: Head Camera Mount Does Not Fit

SidPad, 2025-07-18:
Hello,
Thanks again for your contribution and support in teleoperation control of G1.

Previously, we had used the Head Camera Mount for the G1 robot around two months ago. Due to a mishap, we damaged the robot and about a week ago, we received an updated version of G1 (29 DoF). However, the same Head Camera Mount does not fit on this new G1. There seems to be a 
change in the distance between the screws under the G1's head. Is the Head Camera Mount still supported in the new G1? Or is there an updated .stl/.STEP file you can provide for the Head Camera Mount? If it is still not updated for the new version of G1, then if it is possible, it would be great if you can provide the .stl/.STEP file for the head itself so that we can model our own Head Camera Mount. Pictures are attached for reference.

[image]

[image]

Thanks again for your help in advance!
Please let me know if you need anything from my side.

Sid

silencht, 2025-07-18:
We will upload the new 3D printed bracket structure as soon as possible.

SidPad, 2025-07-22:
@le-wei @silencht  Thank you, please let me know once you have uploaded it!