unitreerobotics/xr_teleoperate #87: 使用brainco灵巧手时，握紧状态下会断联

minecraft-zzz, 2025-07-22:
在使用brainco灵巧手遥操作时，如果不受力可以正常运行。但是如果抓紧某个物品，比如矿泉水瓶，那么会在一两秒之后控制程序崩溃，提示如下：
`2025-07-22T06:25:50.804477Z  INFO bc_stark_sdk::utils::logging_desktop: src/utils/logging_desktop.rs:14: bc_stark_sdk version: 0.4.4
2025-07-22T06:25:50.804497Z  INFO bc_stark_sdk::stark::api: src/stark/api.rs:31: StarkSDK initialized, fw_type: V2Standard, protocol_type: Modbus, is_canfd: false
2025-07-22T06:25:50.804973Z  INFO bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:86: open serial port: "/dev/ttyUSB0", baudrate: 460800
2025-07-22T06:25:50.806050Z  INFO bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:98: serial port opened: "/dev/ttyUSB0"
[2025-07-22 14:25:50.868] [info] Slave id: 127 SKU Type: 1, Serial Number: BCXRR2157J250000A, Firmware Version: 0.0.10

2025-07-22T06:26:49.984419Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:497: write_multiple_registers(address: 1022, data: [129, 1000, 638, 1000, 510, 1000, 572, 1000, 607, 1000, 609, 1000]) timeout, retry 0/3
2025-07-22T06:26:50.035542Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:497: write_multiple_registers(address: 1022, data: [129, 1000, 638, 1000, 510, 1000, 572, 1000, 607, 1000, 609, 1000]) timeout, retry 1/3
2025-07-22T06:26:50.086465Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:497: write_multiple_registers(address: 1022, data: [129, 1000, 638, 1000, 510, 1000, 572, 1000, 607, 1000, 609, 1000]) timeout, retry 2/3
2025-07-22T06:26:50.137472Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 0/3
2025-07-22T06:26:50.187955Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 1/3
2025-07-22T06:26:50.239393Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 2/3`

仅使用右手，使用https://support.unitree.com/home/zh/G1_developer/brainco_hand文档中的程序测试时发现，灵巧手会先握紧，然后张开，然后断开连接。报错信息如下：
`(base) a@aa:~/下载/344ffd55d446431e8c69ca6f3047b3fd/brainco_hand_service/deploy/bin$ ./brainco_hand --id 127 --serial /dev/ttyUSB0
2025-07-22T09:33:19.561423Z  INFO bc_stark_sdk::utils::logging_desktop: src/utils/logging_desktop.rs:14: bc_stark_sdk version: 0.4.4
2025-07-22T09:33:19.561440Z  INFO bc_stark_sdk::stark::api: src/stark/api.rs:31: StarkSDK initialized, fw_type: V2Standard, protocol_type: Modbus, is_canfd: false
2025-07-22T09:33:19.561907Z  INFO bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:86: open serial port: "/dev/ttyUSB0", baudrate: 460800
2025-07-22T09:33:19.562935Z  INFO bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:98: serial port opened: "/dev/ttyUSB0"
[2025-07-22 17:33:19.618] [info] Slave id: 127 SKU Type: 1, Serial Number: BCXRR2157J250000A, Firmware Version: 0.0.10

2025-07-22T09:33:24.678971Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 0/3
2025-07-22T09:33:24.729960Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 1/3
2025-07-22T09:33:24.780919Z  WARN bc_stark_sdk::modbus::modbus_ctx: src/modbus/modbus_ctx.rs:359: read_input_registers(address: 2000, count: 24) timeout, retry 2/3`

Agnel-Wang, 2025-07-22:
Hi, Please clone the newest brainco sdk and try again.
```bash
git clone --recursive https://github.com/unitreerobotics/brainco_hand_service
```

we provide the **unitree_brainco_service** on github now.
see [brainco_hand_service](https://github.com/unitreerobotics/brainco_hand_service)

minecraft-zzz, 2025-08-11:
> Hi, Please clone the newest brainco sdk and try again.
> 
> git clone --recursive https://github.com/unitreerobotics/brainco_hand_service
> we provide the **unitree_brainco_service** on github now. see [brainco_hand_service](https://github.com/unitreerobotics/brainco_hand_service)
已经尝试了最新的sdk，还是有同样的问题