unitreerobotics/xr_teleoperate #92: Grip not working

Norite-dev, 2025-07-24:
Thanks, after the new commits everything works with the Pico 4 Ultra controllers. 

But how to make a grip? Tried all the buttons and triggers - doesn't work. All types of grippers are always in "open" state.

Norite-dev, 2025-07-24:
Dex3 hand. Can controll arms good with controllers on Pico 4 Ultra

https://github.com/user-attachments/assets/0bf96737-0891-441a-91fd-c1bf1002580b

thomasschichl, 2025-07-24:
Hi, I am facing the same issue. I used Meta Quest 3 and Neo Pico 4. Controllers are not working. Tracking is working but the buttons not. However, using handtracking is working and I kan grip objects as well. I guess the issue is that within the teleop_hand_and_arm.py script only considers certain toll (dex1, dex3, inspire) and control combinations (hand, controller)

[image]

I tried to adapt the script, however I failed ;)
Furthermore if I try a scene with the dex1 hand the dds communication is not working either...
tried this
python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex1-Joint    --enable_gripper_dds --robot_type g129
together with 
python teleop_hand_and_arm.py --xr-mode=controller --arm=G1_29 --ee=dex1 --sim

[image]

Furthermore, is it intended that there is no real 3d immersion?- I mean I only see a 2D screen through the NeoP ico 4 or Meta Quest 3 and the 2D screen is moving with the head... so actually no real VR...

I will continue adapting the script to enable the gripper buttons...

silencht, 2025-07-29:
@gawinwong 
ok, we will take this into consideration in the future. Thank your suggestion.

silencht, 2025-09-25:
@mohdammari97 maybe. but this take a while.