unitreerobotics/xr_teleoperate #94: Question for cant access to the URL

Heven-Pan, 2025-07-24:
您好，我本想进行实物部署，但是由于宇树jetson上目前没有网，所以先尝试进行启动仿真环境下的遥操作。
我目前isaac sim都成功配置，且能根据README中给到的命令正常运行，同时teleop_hand_and_arm.py也能正常运行。但我目前存在几个问题：

1.isaac sim运行后，终端没有输出controller started, start main loop...，我确定我是左键点击了，左键点击后输出的内容如下：
Creating window for environment.
[INFO]: Completed setting up the environment...
========= create image server =========
[Image Server] Initializing multi-image server from shared memory
[MultiImageReader] Shared memory opened: isaac_multi_image_shm
[Image Server] Multi-image server initialized on port 5555
[Image Server] Starting send_process from shared memory...
[Image Server] Multi-image publishing thread started
========= create image server success =========
========= create dds =========
[ResetPoseCmdDDS] DDS Node initialized
[ResetPoseCmdDDS] Output shared memory: psm_ffa2c815
[ResetPoseCmdDDS] Reset pose DDS node initialized
[DDSNodeManager] Node 'ResetPoseCmdDDS' registered
[ResetPoseCmdDDS] Setting up subscriber...
[ResetPoseCmdDDS] Reset pose command subscriber initialized
[ResetPoseCmdDDS] Subscriber setup successfully
[ResetPoseCmdDDS] Starting subscribe thread...
[ResetPoseCmdDDS] Subscribe thread started
[ResetPoseCmdDDS] Subscribe thread started
[ResetPoseCmdDDS] Node started successfully
[ResetPoseCmdDDS] Reset pose DDS communication started (subscribe only)
[SimStateDDS] DDS Node initialized
[SimStateDDS] Input shared memory: psm_8a4199b6
[SimStateDDS] Sim state DDS node initialized
[DDSNodeManager] Node 'SimStateDDS' registered
[SimStateDDS] Sim state publisher initialized
[SimStateDDS] Publish thread started
[SimStateDDS] Node started successfully
[SimStateDDS] Sim state DDS communication started (publish only)
========= create dds success =========

create action provider: dds...
ActionProvider init
enable_robot: g129
enable_gripper: False
enable_dex3: True
Starting G1 robot DDS subscriber...
[DDSNodeManager] Node 'G1RobotDDS' registered
[G1RobotDDS] Node is already running, dynamically add features...
[G1RobotDDS] Add subscribe function...
[G1RobotDDS] Create ChannelSubscriber...
[G1RobotDDS] Subscribe thread started
[G1RobotDDS] Subscribe thread started
[G1RobotDDS] G1 robot DDS communication started
start_g1_robot_subscriber_only success
[DDSNodeManager] Node 'Dex3DDS' registered
[Dex3DDS] Node is already running, dynamically add features...
[Dex3DDS] Add subscribe function...
[Dex3DDS] Hand command subscriber initialized
[Dex3DDS] Subscribe thread started
[Dex3DDS] Subscribe thread started
[Dex3DDS] Hand DDS communication started (subscribe only)
dex3_dds start_hand_subscriber_only success
[DDSActionProvider] DDS communication initialized
========= create controller =========
 control frequency: 500Hz
[SimpleController] set the action provider: DDSActionProvider
========= create controller success =========
performance analysis enabled, report every 500 steps
Note: The DDS in Sim transmits messages on channel 1. Please ensure that other DDS instances use the same channel for message exchange by setting: ChannelFactoryInitialize(1).
========= start controller =========
[DDSActionProvider] ActionProvider started
[SimpleController] the controller is started
========= start controller success =========

=== While loop execution frequency statistics ===
loop execution count: 97
running time: 10.08 seconds
overall average frequency: 9.62 Hz
moving average frequency: 9.62 Hz (last 97 times)
frequency range: 3.64 - 4415.06 Hz
average loop time: 103.95 ms
recent loop time: 103.95 ms
=============================

2.teleop启动后我无法访问URL，我通过ifconfig查看了wifi和与机器人连接的网线的ip，都尝试了都无法访问，甚至在自己的笔记本上和ubuntu上都尝试了没办法访问，不知道这个原因是什么。我觉得可能是我哪些操作不对。虽然我后续不使用仿真环境下遥操作，但对于URL的问题我想是否会影响到后续实物部署下访问URL有同样的问题

le-wei, 2025-07-24:
1、根据你提示的日志来看，sim是正常的运行的
2、你说的连不上url问题，有没有什么输出日志或者现象。如果说明更加详细更方便定位问题

silencht, 2025-07-25:
1. 您说的URL具体是指？
2. 无法查看本机的IP具体情况是怎样的？输入`ifconfig`后无法输出所有网卡的IP地址吗？
3. jetson nx虽然没有网络，但只需要将它连接到具备互联网的路由器局域网中，通过ssh远程登录后操作，此时它可以通过该局域网进行互联网链接。

Heven-Pan, 2025-07-25:
1. URL指的是quest 3 需要访问的网址
2. 本机IP可以正常显示，我这边和官方支持一起弄过，网线连接 G1的本机IP地址也改为192.168.123.xx，同时本机连接的wifi的ip地址也可以查看，我将这两个IP地址都分别用在网址上试了都是无法访问
3. 是的，jetson nx这边我是可以ssh，但是由于真实遥操下需要在jetson nx上运行image server，python环境缺少包，所以我需要联网，我通过G1 SDK开发指南使用nmcli配置了wifi，但是没有效果，但这一块和官方支持解决的是我后面可以通过USB直接供网

silencht, 2025-07-25:
@Heven-Pan 

> 2. 我将这两个IP地址都分别用在网址上试了都是无法访问

直接使用 https://github.com/silencht/televuer 进行测试，将本机PC连接到路由器局域网。然后使用VR设备输入本机IP的8012端口网址进行访问，记得使用https，而不是http。

> 3. 是的，jetson nx这边我是可以ssh，但是由于真实遥操下需要在jetson nx上运行image server，python环境缺少包，所以我需要联网，我通过G1 SDK开发指南使用nmcli配置了wifi，但是没有效果，但这一块和官方支持解决的是我后面可以通过USB直接供网

不需要使用jetson nx上的无线网卡进行供网。将具备互联网通信能力的网线接入G1颈部的RJ45网口后，jetson nx会自然通过该有线网卡进行互联网通信。