unitreerobotics/xr_teleoperate #96: vision pro显示问题

Sui1201, 2025-07-25:
我的visionpro中没有出现机器人的第一视角，只能看见自己的手，不知道是什么原因

silencht, 2025-07-25:
您好！

请提供unitree-sim-isaac-lab和xr-teleoperate的启动日志/截图以供我们分析。

Sui1201, 2025-07-25:
(base) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~$ conda activate unitree_sim_env
(unitree_sim_env) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~$ cd ~/unitree_sim_isaaclab
(unitree_sim_env) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~/unitree_sim_isaaclab$ python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex3-Joint --enable_dex3_dds --robot_type g129

[INFO][AppLauncher]: Using device: cpu
[INFO][AppLauncher]: Loading experience file: /home/icy/IsaacLab/apps/isaaclab.python.rendering.kit
Loading user config located at: '/home/icy/isaac-sim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/icy/isaac-sim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250725_143824.log
2025-07-25 06:38:24 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-07-25 06:38:24 [2ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/icy/IsaacLab/apps/rendering_modes' or '/home/icy/IsaacLab/apps/rendering_modes/config'
2025-07-25 06:38:25 [286ms] [Warning] [omni.datastore] OmniHub is inaccessible

|---------------------------------------------------------------------------------------------|
| Driver Version: 570.124.06    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 5090          | Yes: 0 |     | 32853   MB | 10de      | 0          |
|     |                                  |        |     |            | 2b85      | 6ce52b51.. |
|     |                                  |        |     |            | 2         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-67-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) Ultra 9 285K
| Cores: 24 | Logical Cores: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128269 | Free Memory: 117458
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
2025-07-25 06:38:25 [1,006ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2025-07-25 06:38:27 [2,379ms] [Warning] [omni.log] Source: omni.hydra was already registered.
2025-07-25 06:38:27 [2,611ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.
2025-07-25 06:38:27 [3,095ms] [Warning] [omni.replicator.core.scripts.extension] No material configuration file, adding configuration to material settings directly.
2025-07-25 06:38:28 [3,761ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/ambientOcclusion/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/directLighting/sampledLighting/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/indirectDiffuse/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/raytracing/cached/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/reflections/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/sceneDb/ambientLightIntensity'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/translucency/enabled'
2025-07-25 06:38:30 [5,428ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/viewTile/limit'
2025-07-25 06:38:30 [5,524ms] [Warning] [rtx.neuraylib.plugin] [IRAY:RENDER]   1.1   IRAY   rend warn : GPU 1 (NVIDIA GeForce RTX 5090) with CUDA compute capability 12.0 is unsupported by this version of iray photoreal. Please look for an update with your software provider.
2025-07-25 06:38:30 [5,527ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7508933632
2025-07-25 06:38:30 [5,527ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-07-25 06:38:30 [5,527ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7433845248
[... cut ...]
moving average frequency: 22.20 Hz (last 100 times)
frequency range: 11.29 - 30.22 Hz
average loop time: 49.92 ms
recent loop time: 45.04 ms
=============================

=== While loop execution frequency statistics ===
loop execution count: 409
running time: 20.03 seconds
overall average frequency: 20.42 Hz
moving average frequency: 20.14 Hz (last 100 times)
frequency range: 10.03 - 28.49 Hz
average loop time: 48.98 ms
recent loop time: 49.65 ms
=============================
[Performance] A:1.0ms, E:45.7ms, S:0.0ms, T:46.7ms

Sui1201, 2025-07-25:
(base) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~/unitree_sim_isaaclab$ conda activate tv
(tv) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~/unitree_sim_isaaclab$ cd ~/xr_teleoperate/teleop/
(tv) icy@icy-Dell-Pro-Max-Tower-T2-FCT2250:~/xr_teleoperate/teleop$ python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --sim --record
14:46:55:724892 INFO     args:                         teleop_hand_and_arm.py:71
                         Namespace(task_dir='./utils/d                          
                         ata', frequency=60.0,                                  
                         xr_mode='hand', arm='G1_29',                           
                         ee='dex3', record=True,                                
                         motion=False, headless=False,                          
                         sim=True)                                              
14:46:55:725820 INFO     Image client has started, waiting   image_client.py:134
                         to receive data...                                     
Serving file:///home/icy/xr_teleoperate/teleop at /static
Visit: https://vuer.ai?grid=False
14:46:55:730776 INFO     Initialize G1_29_ArmController...       robot_arm.py:63
14:46:56:065238 WARNING  [G1_29_ArmController] Waiting to       robot_arm.py:105
                         subscribe dds...                                       
14:46:56:065327 INFO     [G1_29_ArmController] Subscribe dds    robot_arm.py:106
                         ok.                                                    
14:46:56:066581 INFO     Current all body motor state q:        robot_arm.py:115
                         [ 5.16857579e-02  4.48987354e-03                       
                         2.41512172e-02 -1.65161341e-01                         
                           6.06996357e-01  8.95724446e-03                       
                         1.01913780e-01 -4.96951677e-03                         
                          -4.41718623e-02 -2.16807917e-01                       
                         1.03341639e-01 -8.41967702e-01                         
                          -3.27844718e-05 -1.98914622e-05                       
                         6.52301451e-03  1.31095955e-02                         
                          -8.61865701e-04  1.80389179e-04                       
                         9.85128433e-03 -1.40807009e-04                         
                           3.37531813e-03  9.62963441e-06                       
                         1.31124379e-02  8.71069613e-04                         
                          -1.98635113e-04  9.86438431e-03                       
                         1.24191691e-04  3.38724372e-03                         
                          -6.57966984e-06  0.00000000e+00                       
                         0.00000000e+00  0.00000000e+00                         
                           0.00000000e+00  0.00000000e+00                       
                         0.00000000e+00]                                        
                                                                                
14:46:56:066724 INFO     Current two arms motor state q:        robot_arm.py:116
                         [ 1.31095955e-02 -8.61865701e-04                       
                         1.80389179e-04  9.85128433e-03                         
                          -1.40807009e-04  3.37531813e-03                       
                         9.62963441e-06  1.31124379e-02                         
                           8.71069613e-04 -1.98635113e-04                       
                         9.86438431e-03  1.24191691e-04                         
                           3.38724372e-03 -6.57966984e-06]                      
                                                                                
14:46:56:066741 INFO     Lock all joints except two arms...     robot_arm.py:117
                                                                                
14:46:56:066854 INFO     Lock OK!                               robot_arm.py:137
                                                                                
14:46:56:067349 INFO     Initialize G1_29_ArmController OK!     robot_arm.py:145
                                                                                
 Target link human indices is provided in the DexPilot retargeting config, which is uncommon.
If you do not know exactly how it is used, please leave it to None for default.

 Target link human indices is provided in the DexPilot retargeting config, which is uncommon.
If you do not know exactly how it is used, please leave it to None for default.

14:46:58:655991 INFO     Initialize                     robot_hand_unitree.py:56
                         Dex3_1_Controller...                                   
14:46:59:094160 WARNING  [Dex3_1_Controller] Waiting to robot_hand_unitree.py:95
                         subscribe dds...                                       
14:46:59:094245 INFO     [Dex3_1_Controller] Subscribe  robot_hand_unitree.py:96
                         dds ok.                                                
14:46:59:102057 INFO     Initialize Dex3_1_Controller  robot_hand_unitree.py:103
                         OK!                                                    
                                                                                
[... cut ...]
[2025-07-25T06:46:59Z WARN  wgpu_hal::gles::adapter] Max vertex attribute stride unknown. Assuming it is 2048
14:54:06:503896 INFO     Program start signal          teleop_hand_and_arm.py:44
                         received.                                              

******************************************************************************
This program contains Ipopt, a library for large-scale nonlinear optimization.
 Ipopt is released as open source code under the Eclipse Public License (EPL).
         For more information visit https://github.com/coin-or/Ipopt

Sui1201, 2025-07-25:
您看一下，麻烦了

le-wei, 2025-07-25:
1、仿真有雪花问题可能与显卡有关系，请显示你的配置以及仿真的界面。
2、看不到机器人视角问题，请你确保进入visionpro中的浏览器时刷新页面并确保你允许跟踪手势和头部姿态。

Sui1201, 2025-07-25:
如图是我的显卡使用情况以及我在Vision pro中的截图，sim版本是4.5.0
[image]
[image]
[image]
[image]
[image]
[image]
[image]

le-wei, 2025-07-25:
1、针对sim渲染的问题：我们建议你使用isaacsim 命令打开软件，然后渲染一个场景试试是否一样的不清楚。目前我们需要两次这个情况使用的都是50系列的显卡，据我所知50系列显卡本身兼容性有待提高。建议你先使用isaacsim打开软件测试是否可以正常渲染。
2、visionpro使用情况，从你的图像上看，visionpro好像并没有连接成功，如果建立了连接是会出现图像的。而你的图像中好像没有接收到任何信息。请尝试查看网络通信是否正常。