unitreerobotics/xr_teleoperate #98: No VR Immersion

thomasschichl, 2025-07-26:
Dear community,

I have tried the XR_teleport respository with the Meta Quest 3 and Neo Pico 4 Ultra. I used the built in browsers form Meta qnd Pico as well as the Wolvic browser.
However, as soon as I enter the VR room by pressing hte VR Button in the browser the there is not VR immersion. I just get a 2D window and this 2 D window is moving according to the headset. This means if I turn my head to the right to have a better view at the box in the VR scene the widnows in the goggles is moving along asa well. This makes it extremly hard to work within the VR enviroment.

I tried different other webXR application and they work just find. I even created am sample webxr application in unity3D for testing and it worked just fine on my Meta and Pico goggles.

Is this behaviour intended?

I was facing a similar problem with XR goggles when they were using the spatial features and the goggles were set to single pass mode instead of multipass mode. However, I do not know whether this is even an option here as I did not find anything in regard to this within the python script of televuer.

[image]

Help would be appreciated!

le-wei, 2025-07-28:
@thomasschichl 
Did you click on the SIM window to activate rendering? If you have logs available, providing detailed log output would help better identify the cause of the issue.

silencht, 2025-07-28:
The G1's head is fixed and has no degrees of freedom. Consequently, when you turn your head while wearing a VR headset, the G1's viewpoint does not move accordingly.

Oliver9798, 2025-08-07:
same problem, did u solve?

owenonline, 2025-10-02:
Is there a way to enable the stereoscopic perspective in the vr headset? Even with the head unable to turn, only seeing a 2d image makes it a lot harder to teleoperate since I have no perception of depth.