unitreerobotics/xr_teleoperate #100: dex3 can communicate, but dex1 cannot, showing 'Waiting to subscribe dds'

gawinwong, 2025-07-28:
I launched a pickplace task for dex1 using unitree-sim-isaaclab, but it cannot communicate with xr_teleop, persistently showing 'Waiting to subscribe dds'. 

However, when starting a dex3 task, communication works (though the frame rate drops significantly). 

I noticed that dex1 was renamed from gripper—could the communication failure with dex1 be caused by mismatched topic names?

le-wei, 2025-07-28:
@gawinwong 
Please ensure that both SIM and `xr_teleop` use the same end-effector. For example, if you start the simulator with the following command:

```bash
python sim_main.py --device cpu --enable_cameras --task Isaac-PickPlace-Cylinder-G129-Dex1-Joint --enable_gripper_dds --robot_type g129
```

Then `xr_teleop` should be launched with:

```bash
python teleop_hand_and_arm.py --ee=dex1 --sim
```

Also, setting the Power mode to **Performance** may help improve the rendering frame rate.

gawinwong, 2025-07-28:
After reviewing the source code, I found that in `unitree-sim-isaaclab`, the topics corresponding to `enable_gripper_dds` are `rt/unitree_actuator/state` (publisher) and `rt/unitree_actuator/cmd` (subscriber).

However, in `xr_teleop`, the topics for Dex1 are `rt/dex1/left/cmd`, `rt/dex1/left/state`, `rt/dex1/right/cmd`, and `rt/dex1/right/state`. 

This mismatch in topic naming is the reason why communication fails when I replace the end effectors in both systems with Dex1.

le-wei, 2025-07-28:
@gawinwong 
Due to changes in the DDS implementation for `dex1` in the latest version of `xr_teleoperate`, the simulator was not initially updated to accommodate these changes. However, the simulator has now been adapted to work with the latest `xr_teleoperate`. Please make sure to use the latest version of the simulator.

gawinwong, 2025-07-29:
Thank you, the issue has been resolved after the update.