unitreerobotics/xr_teleoperate #106: Physical Teleoperation with Apple vision Pro and Unitree Dex-3 handsets

robotsforhumanity, 2025-07-29:
Hello, good afternoon, I am having many doubts in the documentation that you are giving to be able to teleoperate the robot in physical by a VR device. Could you help me in that step by step, because what I did is to pass the image_server.py to the robot, install all the necessary tools for teleoperation in the robot and nothing else. So now I don't know what I need to do. I would be very grateful for your help. Thank you very much.

silencht, 2025-07-30:
Here is the wiring diagram for your reference.
[image]

Please describe your situation—for example, share the startup logs. 

Also, have you successfully completed the simulated deployment?

robotsforhumanity, 2025-07-30:
What I want to know is the step by step of the commands to launch in the Unitree G1 robot with the dex3 hands to teleoperate with a VR device, because it is not very well explained in the documentation. I want to know if inside the robot I have to launch the teleop_hand_with_arm.py and with that you can teleoperate or need some other tool.

robotsforhumanity, 2025-07-30:
in robot launch, python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --motion --record, but it always keeps saying waiting for subscriber dds, so it makes me wonder if I don't have to launch this file that is in this path unitree_sdk2py/g1/arm/g1_arm_action_client.py

silencht, 2025-07-31:
hi, since you’re using Dex3, the only component that needs to be launched on the Jetson NX inside the G1 is the image server: `image_server.py`.

`python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --motion --record`  
should be run on **your PC (User PC)**, not on the robot.

if you use `--motion` param, please first read this [--motion Mode wiki](https://github.com/unitreerobotics/xr_teleoperate/wiki/%E2%80%90%E2%80%90motion-Mode)

The “waiting for subscriber DDS” message likely means your device is **not on the same LAN** as the G1.  
Please verify the IP configuration by following the guide at:  
https://support.unitree.com/home/en/G1_developer/quick_development

robotsforhumanity, 2025-07-31:
We followed the official Unitree documentation to configure the robot and development environment, specifically the tutorial at: https://support.unitree.com/home/en/G1_developer/quick_development. However, we noticed that the IP addresses shown there do not match the actual configuration of our robot. For instance, our robot uses the IP address 10.0.11.16 on the usb0 interface.

We tried different configurations. First, we connected the robot to the PC via Ethernet as shown in GitHub diagrams. When we do that, the usb0 interface automatically disables itself, which prevents communication between PC1 and PC2.

So instead, we kept the robot connected via usb0, and connected PC2 and the Apple Vision Pro to the same Wi-Fi network. On PC2, we launched the command
python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --motion --record.
Then, from the Apple Vision Pro browser, we opened the web interface using the link:
https://<PC2_IP>:8012?ws=wss://<PC2_IP>:8012,
and the interface loaded correctly.

Despite this, the Apple Vision Pro cannot ping the robot. Interestingly, the robot can ping the Apple Vision Pro. However, the teleoperation process gets stuck at the line waiting for subscriber dds. We used cyclonedds ps to inspect what was happening and confirmed that the node subscribes correctly to the topic. But still, no data is received, which leads us to believe it could be a routing, firewall, or DDS unidirectional communication issue.

We’re attaching screenshots and diagrams to better illustrate the configuration we tested and the results we obtained.

[image]

[image]

[image]

[image]

[image]

silencht, 2025-08-01:
The network cards of PC1 and PC2 of Robot G1 are both configured under the `192.168.123.x/24` subnet. Therefore, it is **necessary** to modify the IP address of the User PC according to the configuration specified in the official documentation. **Ensure** that the IP address of the network card on the User PC connected to the robot's network interface is within the `192.168.123.x` subnet. 

After the configuration is completed, the User PC should be able to ping `192.168.123.161` and `192.168.123.164`. At this point, the User PC should also be able to subscribe to all the topic data published by PC1 using the command `cyclonedds ps` (there are many items, not just the two you listed).

Normally, `python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --motion --record` should be launched on User PC. And,  Apple Vision Pro connect Router's WiFi (Router also connect User PC with cable). Then, AVP's safari enter `https://<User PC>:8012?ws=wss://<User PC_IP>:8012`.

P.S. To prevent the user's computer from blocking port 8012, please use `sudo ufw allow 8012` to open port 8012.

robotsforhumanity, 2025-08-01:
Hi, I wanted to let you know that I managed to teleoperate the robot using the Apple Vision Pro, and I’ll briefly explain the process in case you'd like to include it in the README or it’s useful for future development.

First, I accessed PC2 and saw that the usb0 interface was active with the IP address 10.6.1.10. When I tried to enable the Ethernet interface (eth0), usb0 was automatically disabled, which prevented communication between PC2 and PC1. To solve this, I manually configured the Ethernet interface with the IP 192.168.123.164. Once done, I confirmed there was a connection to PC1 by running the command cyclonedds ps, which showed over 200 active topics.

Then, I connected PC2 to a local Wi-Fi network and ran the script using the following command:
python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --motion --record
However, I noticed that no data was being received from the Dex-3 hands and the system kept displaying "waiting for subscriber dds...". To move forward, I temporarily commented out that part of the code, and the script then launched correctly.

Finally, I connected the Apple Vision Pro to the same network as PC2, opened the browser from the headset, and was able to control the robot from there. However, I still don’t have control over the finger movements.

Thank you so much for all your help, and if you could guide me with that last step (finger control), I’d greatly appreciate it.