unitreerobotics/xr_teleoperate #111: 如何在仿真遥操作中不使用腕部相机？

JunweiLiang, 2025-08-01:
我尝试把`pickplace_redblock_g1_29dof_inspire_joint_env_cfg.py`中的ObjectTableSceneCfg注释掉腕部相机，
```
@configclass
class ObjectTableSceneCfg(TableRedBlockSceneCfg):
    """object table scene configuration class
    
    inherits from G1SingleObjectSceneCfg, gets the complete G1 robot scene configuration
    can add task-specific scene elements or override default configurations here
    """
    
    # Humanoid robot w/ arms higher
    # 5. humanoid robot configuration 
    robot: ArticulationCfg = G1RobotPresets.g1_29dof_inspire_base_fix(init_pos=(-4.2, -3.7, 0.76),
        init_rot=(0.7071, 0, 0, -0.7071))

    # 6. add camera configuration 
    front_camera = CameraPresets.g1_front_camera()
    #left_wrist_camera = CameraPresets.left_inspire_wrist_camera()
    #right_wrist_camera = CameraPresets.right_inspire_wrist_camera()
```

但是在`teleop_hand_and_arm.py`中img_config把腕部相机也注释掉，VR获取的图片是黑的，请问如何解决呢？谢谢！
```
    if args.sim:
        img_config = {
            'fps': 30,
            'head_camera_type': 'opencv',
            'head_camera_image_shape': [480, 640],  # Head camera resolution
            'head_camera_id_numbers': [0],
           # 'wrist_camera_type': 'opencv',
           # 'wrist_camera_image_shape': [480, 640],  # Wrist camera resolution
            #'wrist_camera_id_numbers': [2, 4],
        }
```

le-wei, 2025-08-01:
不好意思，这种情况我们没有考虑到呢，你需要在teleop_hand_and_arm.py添加如图的代码：

[image]

JunweiLiang, 2025-08-01:
感谢回复！这个能work了

closed 2025-08-01 by JunweiLiang