unitreerobotics/xr_teleoperate #112: Only grids in VR view

ziwen-meng, 2025-08-01:
I am using meta quest 3 and following the instructions as follow:
~/unitree_sim_isaaclab$ python sim_main.py --device cpu  --enable_cameras  --task  Isaac-PickPlace-Cylinder-G129-Dex3-Joint --enable_dex3_dds --robot_type g129
in one terminal

and ~/xr_teleoperate/teleop/$ python teleop_hand_and_arm.py --xr-mode=hand --arm=G1_29 --ee=dex3 --sim --record in another terminal

on my head set i typed in the https address.

I successfully entered the vr mode but can only see blue grids not the robot...

ziwen-meng, 2025-08-01:
this is the output from the unitree isaac sim terminal
(unitree_sim_env) wendy@wendy-Legion-5-16IRX9:~/unitree_sim_isaaclab$ python sim_main.py --device cpu --enable_cameras --task Isaac-PickPlace-Cylinder-G129-Dex3-Joint --enable_dex3_dds --robot_type g129
[DDSManager] DDS system initialized
[DDSManager] DDSManager initialized
[INFO][AppLauncher]: Using device: cpu
[INFO][AppLauncher]: Loading experience file: /home/wendy/IsaacLab/apps/isaaclab.python.rendering.kit
Loading user config located at: '/home/wendy/isaacsim/kit/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/wendy/isaacsim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250801_170154.log
2025-08-01 08:01:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-08-01 08:01:54 [2ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/wendy/IsaacLab/apps/rendering_modes' or '/home/wendy/IsaacLab/apps/rendering_modes/config'
2025-08-01 08:01:54 [272ms] [Warning] [omni.datastore] OmniHub is inaccessible
2025-08-01 08:01:55 [742ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)
2025-08-01 08:01:55 [742ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) Graphics (RPL-S)

|---------------------------------------------------------------------------------------------|
| Driver Version: 575.64.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Laptop.. | Yes: 0 |     | 8188    MB | 10de      | 0          |
|     |                                  |        |     |            | 28e0      | 2af8a1dd.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) Graphics (RPL-S)        |        |     | 23849   MB | 8086      | 0          |
|     |                                  |        |     |            | a78b      | 86808ba7.. |
|     |                                  |        |     |            | 0         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-65-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12201001 (1.22.1.1)
| Processor: Intel(R) Core(TM) i7-14650HX
| Bare Metal Cores: 16 | Bare Metal Logical Cores: 32
| Available Cores:  24 
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31799 | Free Memory: 22763
| Total Page/Swap (MB): 0 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|
2025-08-01 08:01:55 [1,030ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2025-08-01 08:01:55 [1,030ms] [Warning] [gpu.foundation.plugin] Detected IOMMU is enabled. Running CUDA peer-to-peer bandwidth and latency validation.
Unidirectional P2P=Enabled Bandwidth (P2P Writes) Matrix (GB/s)
   D\D     0 
     0 198.12 
P2P=Enabled Latency (P2P Writes) Matrix (us)
   GPU     0 
     0   1.41 

   CPU     0 
     0   1.68 
2025-08-01 08:01:56 [1,841ms] [Warning] [omni.log] Source: omni.hydra was already registered.
2025-08-01 08:01:56 [2,134ms] [Warning] [omni.isaac.dynamic_control] omni.isaac.dynamic_control is deprecated as of Isaac Sim 4.5. No action is needed from end-users.
2025-08-01 08:01:57 [2,843ms] [Warning] [omni.replicator.core.scripts.extension] No material configuration file, adding configuration to material settings directly.
2025-08-01 08:01:58 [3,775ms] [Warning] [omni.kit.menu.utils.app_menu] add_menu_items: menu [<MenuItemDescription name:'New'>, <MenuItemDescription name:'Open'>, <MenuItemDescription name:'Re-open with New Edit Layer'>, <MenuItemDescription name:'Save'>, <MenuItemDescription name:'Save With Options'>, <MenuItemDescription name:'Save As...'>, <MenuItemDescription name:'Save Flattened As...'>, <MenuItemDescription name:'Add Reference'>, <MenuItemDescription name:'Add Payload'>, <MenuItemDescription name:'Exit'>] cannot change delegate
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/ambientOcclusion/enabled'
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/directLighting/sampledLighting/enabled'
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/indirectDiffuse/enabled'
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/raytracing/cached/enabled'
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/reflections/enabled'
2025-08-01 08:01:59 [5,383ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults/sceneDb/ambientLightIntensity'
[... cut ...]
========= start controller success =========
dds_manager: <dds.dds_master.DDSManager object at 0x7e1f3ef42ec0>
[g1_state] G1 robot DDS communication instance obtained
[Observations Dex3] DDS communication instance obtained

=== While loop execution frequency statistics ===
loop execution count: 90
running time: 10.06 seconds
overall average frequency: 8.95 Hz
moving average frequency: 8.95 Hz (last 90 times)
frequency range: 6.90 - 7231.56 Hz
average loop time: 111.77 ms
recent loop time: 111.77 ms
=============================

ziwen-meng, 2025-08-01:
[image]

silencht, 2025-08-01:
> 💥💥💥 NOTICE❗
> After simulation starts, click once in the window to activate it.
> The terminal will show: controller started, start main loop...

do you complete this step? because i dont see information `controller started, start main loop...` in your first teminal output.

ziwen-meng, 2025-08-01:
i/ve tried clicking.. same problem
***  Please left-click on the Sim window to activate rendering. ***

dds_manager: <dds.dds_master.DDSManager object at 0x748bc654aec0>
[DDSManager] object 'g129' not found, objects: dict_keys([])
[g1_state] G1 robot DDS communication instance obtained
g1_robot_dds is not initialized
[DDSManager] object 'dex3' not found, objects: dict_keys([])
[Observations Dex3] DDS communication instance obtained
========= create image server =========
[Image Server] Initializing multi-image server from shared memory
[MultiImageReader] Shared memory opened: isaac_multi_image_shm
[Image Server] Multi-image server initialized on port 5555
[Image Server] Starting send_process from shared memory...
[Image Server] Multi-image publishing thread started
========= create image server success =========
========= create dds =========
[g1_robot] Input shared memory: psm_4df1e49a
[g1_robot] Output shared memory: psm_302d3fb8
[g1_robot] G1 robot DDS node initialized
[DDSManager] register object 'g129' success (category: 未分类)
[dex3] Input shared memory: psm_7eec3744
[dex3] Output shared memory: psm_896e31dd
[dex3] Hand DDS node initialized
[DDSManager] register object 'dex3' success (category: 未分类)
[reset_pose_dds] Output shared memory: psm_670145b3
[reset_pose_dds] Reset pose DDS node initialized
[DDSManager] register object 'reset_pose' success (category: 未分类)
[sim_state_dds] Input shared memory: psm_8f701530
[sim_state_dds] Sim state DDS node initialized
[DDSManager] register object 'sim_state' success (category: 未分类)
[g1_robot] State publisher initialized (rt/lowstate)
[dex3] Hand state publisher initialized
[sim_state_dds] Sim state publisher initialized
[DDSManager] publish loop thread started
[DDSManager] manager started, managing 4 objects
[g1_robot] Create ChannelSubscriber...
[dex3] Hand command subscriber initialized
[reset_pose_dds] Reset pose command subscriber initialized
========= create dds success =========

create action provider: dds...
ActionProvider init
enable_robot: g129
enable_gripper: False
enable_dex3: True
[DDSActionProvider] DDS communication initialized
========= create controller =========
  - control frequency: 500Hz
[SimpleController] set the action provider: DDSActionProvider
========= create controller success =========
performance analysis enabled, report every 500 steps
Note: The DDS in Sim transmits messages on channel 1. Please ensure that other DDS instances use the same channel for message exchange by setting: ChannelFactoryInitialize(1).
========= start controller =========
[DDSActionProvider] ActionProvider started
[SimpleController] the controller is started
========= start controller success =========
dds_manager: <dds.dds_master.DDSManager object at 0x748bc654aec0>
[g1_state] G1 robot DDS communication instance obtained
[Observations Dex3] DDS communication instance obtained

ziwen-meng, 2025-08-01:
[image]

ziwen-meng, 2025-08-01:
[image]

le-wei, 2025-08-04:
@ziwen-meng 
I tested your workflow using the Apple Vision Pro and did not encounter any issues.
May I ask, when you press **'r'** in the `xr_teleoperate` terminal, do you see an image like the one shown below?
In my case, I experienced this on the Apple Vision Pro, and restarting the device resolved the issue.

[image]

ziwen-meng, 2025-08-04:
@le-wei Hi, I didn't see anything like that on my PC.

[image]

ziwen-meng, 2025-08-04:
I tried again and now I see the same problem as https://github.com/unitreerobotics/xr_teleoperate/issues/98

Oliver9798, 2025-08-07:
> see anything l

same as you, did u solve this issue?

lllzheng, 2025-09-16:
> 我再次尝试，现在看到与[#98相同的问题](https://github.com/unitreerobotics/xr_teleoperate/issues/98)

I also encountered the problem of Only grids in VR view. May I ask how you solved it