unitreerobotics/xr_teleoperate #115: 遥操网页丢失

jkxdl, 2025-08-05:
我搞了一个路由器，ip是192.168.31.1
1. 笔记本WIfi的IP：192.168.31.146
2. 笔记本跟G1通过网线链接，网线的IP：192.168.123.99，可以ping通G1
3. VR的IP：192.168.31.68。 （笔记本也能ping通V R的地址）
vision pro按要求输入网站后，能实时看到回传到vision pro的双目相机图像，但点击virtual reality，允许获取手部动作之后，vision pro黑屏看不到双目相机图像，同时代码报错：
websocket is connected. id: 776eeab4-2617-4b93-8c73-eeba1bd0e31e default socket worker is up, adding clientEvents
Uplink task running. id: 776eeab4-2617-4b93-8c73-eeba1bd0e31e websocket is now disconnected. Removing the socket.
WebSocket connection closed
Task exception was never retrieved
future: <Task finished name='Task-11' coro=<Vuer. downlink.< locals>. handler () done, defined at /home/robot/miniconda3/envs/tv/L ib/python3. 10/site-packages/vuer/server.py:618> exception=AssertionError ( 'Websocket session is missing.')>
Traceback (most recent call last):
File "/home/robot/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", Line 624, in handler raise e
File "/home/robot/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py"
, line 620, in handler
await self. socket_handler (vuer_proxy)
File "/home/robot/xr_teleoperate/teleop/televuer/src/televuer/televuer.py"，Line 219,in main_inage_binocular
session.upsert(
File "/home/robot/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 49, in _call_
return self.fn(*args, **kwargs)
File "/home/robot/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 199, in _upsert self @ Upsert(*element, to=to, strict=strict)
File "/home/robot/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 93, in _matmul_ assert self. CURRENT_WS_ID in self. vuer.ws, "Websocket session is missing."
AssertionError: Websocket, session is missing.
uplink: 776eeab4-2617-4b93-8c73-eeba1bd0e31e

silencht, 2025-08-05:
1. 仿真部署下可以运行吗？
2. 可以切换 vuer 库版本进行测试，比如切换至 0.0.32rc7版本。该仓库 git tag 为 v0.5。televuer仓库下可以直接修改 [pyproject.toml](https://github.com/silencht/televuer/blob/main/pyproject.toml) 的 dependencies

jkxdl, 2025-08-06:
网络带宽是否也会影响？我搭建的局域网带宽要小一些

silencht, 2025-08-06:
是的。路由器性能是一个重要因素，我们曾因为路由器性能降低引发了很多问题。

aruiplex, 2025-08-11:
您好，我遇到了差不多的问题，我用apple vision pro来访问链接的时候，显示safari浏览器打不开该网页，因为已丢失网络连接。但是我在同一个局域网下的其他设备中测试是有画面的（笔记本chrome进行测试）。

截图是笔记本的画面

[image]

这是vision pro的截图画面

[image]

我已经被这个问题困扰了至少4天了，感谢您的帮助。

jkxdl, 2025-08-11:
> 您好，严格按照步骤重新尝试过了，跳出了警告页面，然后仍然报一样的错：报错也确认继续后，还是显示丢失连接。我们手动安装证书后发现，在vision pro上装的edge浏览器可以打开，但是遥操无效，感觉应该是edge不支持xr功能。 [@silencht](https://github.com/silencht) 感谢！

你好，你的edge浏览器是否有提示授权获取手部动作

ning-mz, 2025-08-11:
> > 您好，严格按照步骤重新尝试过了，跳出了警告页面，然后仍然报一样的错：报错也确认继续后，还是显示丢失连接。我们手动安装证书后发现，在vision pro上装的edge浏览器可以打开，但是遥操无效，感觉应该是edge不支持xr功能。 [@silencht](https://github.com/silencht) 感谢！
> 
> 你好，你的edge浏览器是否有提示授权获取手部动作

没有提示，edge下，在台式上terminal 输入 r后，手就飞了

silencht, 2025-08-12:
@aruiplex 

1. 确保您所有设备均处于同一局域子网段下，以尽可能排除网络问题。比如将路由器设置为 `192.168.123.x`网段，wifi设置为2.4ghz与5ghz分离。
2. 忘记之前wifi链接，重新输入密码加入连接。
3. 使用Safari而不是Edge，尝试按照旧版README向Safari浏览器安装证书，您可参考git tag v0.5版本README中向apple vision pro中安装证书的步骤。
4. 有时重启apple vision pro可能会解决问题（我们仅遇到过几次，但确实不知道问题在哪里）