unitreerobotics/xr_teleoperate #117: 在运控模式下，没有遥控机器人，但机器人一直往前走

jkxdl, 2025-08-06:
机器人在常规运控模式下，运行指令：python teleop_hand_and_arm.py --ee=inspire1 --record --motion，输入r后开始遥操，没有使用R3 Controller控制机器人往前运动但机器人自动向前走

jkxdl, 2025-08-06:
之前没有出现这样的问题，后面刷过一次机，但代码完全没有改动

silencht, 2025-08-06:
没有遥操时，机器人会自动往前走吗？邀操时，机器人双臂在前方出现大幅度动作时重心靠前，为保证平衡，机器人确实会向前踏步。但如果一直往前走是不正常的。

jkxdl, 2025-08-06:
是否需要接触腰部锁定，在常规运控的三腰模式下机器人可以利用腰部关节平衡重心吗？G1的29dof版本具有此功能吗？

silencht, 2025-08-11:
> 是否需要接触腰部锁定，在常规运控的三腰模式下机器人可以利用腰部关节平衡重心吗？G1的29dof版本具有此功能吗？
> […](#)

您好！

请参考 https://support.unitree.com/home/zh/G1_developer/joint_motor_sequence 、https://support.unitree.com/home/zh/G1_developer/arm_control_routine、
https://support.unitree.com/home/zh/G1_developer/sport_services_interface

常规运控只控制腿部自由度保持平衡，用户接管可以接管腰部及上肢自由度。