unitreerobotics/xr_teleoperate #129: 使用PICO4Ultra遥操作网页界面经常断连接需要重新刷新。

siawxx, 2025-08-28:
你好，我使用PICO4Ultra VR设备遥操作机器人，路由器使用的是中兴的AX3000路由器，千兆wifi6,网络拓扑是将路由器接到计算机，VR通过wifi连接路由器，路由器网关是192.168.5.1，浏览器使用的是VR自带的浏览器，然后机器人是接入到计算机的另一块网卡，机器人与计算机直连。所有流程都正常，也可以正常操控机器人，但目前有几个问题：
1.在VR的浏览器刚打开网页后，点击visual reality全屏后，刚开始右眼是看不到图像的，要过几秒钟才能看到图像。
2.遥操作时画面经常卡顿，机器人也会卡顿。浏览器界面经常断开，需要重新刷新，这个非常频繁，导致无法正常操作机器人采集数据。浏览器断开时计算机xr_teleoperate项目终端会报错websocket is now disconnected. Removing the socket. WebSocket connection closed Connection closed due to Cannot write to closing transport Task exception was never retrieved future: <Task finished name='Task-8116' coro=<Vuer.downlink.<locals>.handler() done, defined at /home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py:618> exception=AssertionError('Websocket session is missing.')> Traceback (most recent call last): File "/home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 624, in handler raise e File "/home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 620, in handler await self.socket_handler(vuer_proxy) File "/home/sia05/xr_teleoperate/teleop/televuer/src/televuer/televuer.py", line 219, in main_image_binocular session.upsert( File "/home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 49, in __call__ return self.fn(*args, **kwargs) File "/home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 199, in _upsert self @ Upsert(*element, to=to, strict=strict) File "/home/sia05/miniconda3/envs/tv/lib/python3.10/site-packages/vuer/server.py", line 93, in __matmul__ assert self.CURRENT_WS_ID in self.vuer.ws, "Websocket session is missing." AssertionError: Websocket session is missing.
这个问题困扰我好几天了，请技术人员帮忙分析一下原因，非常感谢！

siawxx, 2025-08-28:
> 1. 这个现象正常，我们也遇到了。我目前怀疑是vuer库版本的问题。因为几秒后会恢复正常，因此还未测试更新版本。您可以尝试＞0.0.60版本。
> 
>     2. 该问题确实偶有出现。涉及到网络连接，可能出现问题的排查点比较多。除了我们暂时不能掌控的vuer库设置之外，我可以给出您一些建议：
> 
> 
>     * 重启所有设备（包括路由器），是的，这一般可以解决90%问题， orz。
> 
>     * 解除路由器双频合一特性，将WiFi设置为仅2.4Ghz与仅5Ghz频段。然后VR设备连接5Ghz频段热点
> 
>     * 将您的 PICO4U 从 Wi-Fi 切换到 ADB 反向端口转发。您可以参考 [teleop based on Quest 3 / Pico 4UE #32](https://github.com/unitreerobotics/xr_teleoperate/issues/32) 获取指导。通过 ADB 反向端口转发，用户电脑向PICO4U设备开放服务器端口（8012），从而进行数据的有线传输。

你好，目前我安装使用的就是0.0.60版本的vuer，而且现在不是偶尔出现浏览器断开连接，而是经常出现，基本几秒钟就会断开，而且频繁出现卡顿。我尝试过重启路由器和所有设备了，wifi也只连的5g频段，还是不能解决问题。
另外我们购买的是普通版的pico4ultra 不是企业版的 是否必须要使用企业版的？我看到issues里有人也反馈相关问题，回复的是安装4,0.28测试版浏览器，但这个版本必须要在企业版pico4u里安装，我的普通版的安装不上。是否有普通版能用的浏览器？

siawxx, 2025-08-28:
> 您好。在该issue下 [#63 (comment)](https://github.com/unitreerobotics/xr_teleoperate/issues/63#issuecomment-3022233778) 曾有用户反馈，非企业版也成功安装了4.0.28浏览器。您无法安装是提示了哪些错误？
你好，我这边点击安装之后进度条走到头，然后就直接提示无法安装，没有任何错误信息。

silencht, 2025-08-28:
请您尝试一下`adb install -d -r`等命令使用安装包。详情可与chatgpt等助手对话，尝试使用命令解除安装权限。
时间有点儿久了，具体命令我们没有具体记录。如果成功的话，如果您可以将过程贴到此issue，不胜感激。
另外，我们正在与Pico团队沟通，看能否将之前导致闪退的`bugfix`合并到最新浏览器分支。

siawxx, 2025-08-28:
> 请您尝试一下`adb install -d -r`等命令使用安装包。详情可与chatgpt等助手对话，使用命令解除安装权限。 时间有点儿久了，具体命令我们没有具体记录。如果成功的话，如果您可以将过程贴到此issue，不胜感激。 另外，我们正在与Pico团队沟通，看能否将之前导致闪退的`bugfix`合并到最新浏览器分支。

我一开始尝试直接adb install安装报错，后来我把之前系统里的新版删除掉就到4.0.28版本了，通过adb shell pm uninstall --user 0 com.pico.browser命令删除的，但是用这个版本浏览器仍然卡顿断连接，问题还是没有解决，是PICO VR设备的原因吗？另外我还尝试用chrome和edge浏览器，但是会报设备不支持google play服务，就无法操控机器人。还有VR界面边上的那些参数例如render,scene等需要调整吗？还是就用默认的就行？

silencht, 2025-08-28:
我们测试使用的企业版恢复出厂设置并更新至最新版后，浏览器版本默认在4.0.28（官方，非我们issue中提供的beta版本apk）。该版本使用同样是正常的。

VR界面边上的那些参数不需要调整。

另外，您可以尝试将Vuer版本降级至0.0.32rc7，或者0.0.40。

再次，可以尝试一下 adb 端口转发方案。

siawxx, 2025-08-29:
> 我们测试使用的企业版恢复出厂设置并更新至最新版后，浏览器版本默认在4.0.28（官方，非我们issue中提供的beta版本apk）。该版本使用同样是正常的。
> 
> VR界面边上的那些参数不需要调整。
> 
> 另外，您可以尝试将Vuer版本降级至0.0.32rc7，或者0.0.40。
> 
> 再次，可以尝试一下 adb 端口转发方案。

现在pico的系统更新到最新后，浏览器就是4.0.32版本了，4.0.28版本试了了也不行，Vuer版本在哪可以下载老版本的？adb端口转发方案正在准备尝试。

silencht, 2025-08-29:
我们最近采购的 PICO4 ultra（非企业版），系统版本号 5.14.4.u，浏览器版本 4.0.34 ，部署后可以正常使用。

> Vuer版本在哪可以下载老版本的

`televuer`仓库的pyproject.toml中可以进行修改。

siawxx, 2025-08-29:
> 我们最近采购的 PICO4 ultra（非企业版），系统版本号 5.14.4.u，浏览器版本 4.0.34 ，部署后可以正常使用。
> 
> > Vuer版本在哪可以下载老版本的
> 
> `televuer`仓库的pyproject.toml中可以进行修改。

你好，刚才试adb反向传输，sudo ./adb shell ifconfig wlan0这步看不到VR设备的ip地址，我这里输出的是这样的信息
wlan0     Link encap:UNSPEC    Driver cnss_pci
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:3953908 errors:0 dropped:0 overruns:0 frame:0 
          TX packets:641005 errors:0 dropped:0 overruns:0 carrier:0 
          collisions:0 txqueuelen:3000 
          RX bytes:5676358201 TX bytes:150071662 
没有inet addr:your_xr_device_ip  Bcast:192.168.123.255  Mask:255.255.255.0，请问应如何查看VR设备的IP地址？

silencht, 2025-08-29:
或许你可以登录路由器的管理后台，来确认路由器通过DHCP为PICO设备分配的ip地址。

silencht, 2025-08-29:
请咨询pico售后服务。重置pico出厂或许可以？

siawxx, 2025-09-01:
ping了下VR网络延迟不稳定，怀疑是路由器的问题，更换路由器后问题已解决，televuer版本改到最新的0.0.60也可正常操作。

siawxx, 2025-09-08:
> [@siawxx](https://github.com/siawxx) 能问下存在问题的路由器型号以及可用的路由器型号吗？

你好，我之前使用的是中兴的路由器，后来更换成华硕的小旋风pro路由器就好了。