unitreerobotics/xr_teleoperate #133: Direct Wireless Connection

maxvipon, 2025-08-29:
Hi there,

Could you please share the instructions for setting up the "wireless mode" as indicated in orange on the diagram from the README?

[image]

I tried creating a hotspot on the G1 dev-PC and connecting both the Host and the AVP to it, as well as connecting to the “G1_4801” hotspot, but neither worked.

silencht, 2025-09-05:
To learn more about the Unitree SDK’s network communication, please start by consulting this [wiki](https://github.com/unitreerobotics/xr_teleoperate/wiki/CycloneDDS-and-UnitreeSDK-(en)).

------

As long as the `xr-teleoperate` program runs on PC2 (Jetson NX), you only need to specify the DDS network interface param as eth0 to communicate normally with PC1. Either turn PC2’s wireless card into a hotspot or join it to the router’s Wi-Fi network. The user PC logs into PC2 over Wi-Fi via SSH and starts the xr-teleoperate program. The VR device communicates with the PC2 wireless network card via the WLAN, only passing the data to the `xr-teleoperate`. 

------ 

And, If the xr-teleoperate program is launched on the user’s own PC (instead of on PC2). I also have tried bridging pc2's wlan0 and eth0. However, due to various reasons such as hardware and drivers, I failed: ordinary data transmission is not a problem, but cyclonedds transmission delay is very large. I'm not sure if your understanding of the network is deep enough. If you have some research on computer networks, you may want to try bridging under a two-layer network yourself. If you can succeed, it is best to feedback the plan to the community. 

In addition, you can also try routing forwarding/proxy at the third IP layer. 

Finally, the simplest way is to create an application layer cyclonedds forwarding program in PC2. The program acts as a dds-relay: it receives data from the user PC, forwards it to PC1, and then relays PC1’s data back to the user PC.

maxvipon, 2025-09-17:
@silencht, thanks for the detailed reply. I read through the wiki, understood how the network is structured, and tried to bridge `eth0` and `wlo0` on PC2 (Orin NX). Unfortunately, this setup didn’t work for me — and it also felt risky since I could easily lose access to PC2 both over cable and Wi-Fi. So instead I went with installing xr_teleoperate directly on PC2.

Here’s what I ran into:

### **Problem 1: Installation on Orin NX (Ubuntu 20.04)**
Surprisingly, xr_teleoperate doesn’t build out of the box on the robot’s onboard computer.

I had to fix two issues:

1. **nlopt build failure** (during `dex-retargeting` submodule install)

<details><summary>Details</summary>
<p>

```bash
~/unitree_teleop/telop/robot_control/dex-retargeting$ pip install -e .
...
DEPRECATION: Building 'nlopt' using the legacy setup.py bdist_wheel mechanism, which will be removed in a future version. pip 25.3 will enforce this behaviour change. A possible replacement is to use the standardized build interface by setting the `--use-pep517` option, (possibly combined with `--no-build-isolation`), or adding a `pyproject.toml` file to the source tree of 'nlopt'. Discussion can be found at https://github.com/pypa/pip/issues/6334
  Building wheel for nlopt (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [190 lines of output]
      running bdist_wheel
      running build
      running build_ext
      cmake version 3.16.3

      CMake suite maintained and supported by Kitware (kitware.com/cmake).
      -- The C compiler identification is GNU 9.4.0
      -- The CXX compiler identification is GNU 9.4.0
      -- Check for working C compiler: /usr/bin/cc
      -- Check for working C compiler: /usr/bin/cc -- works
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Check for working CXX compiler: /usr/bin/c++
      -- Check for working CXX compiler: /usr/bin/c++ -- works
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Found PythonInterp: /home/unitree/miniconda3/envs/teleop/bin/python3.10 (found version "3.10.18")
      -- Found Python includes: /home/unitree/miniconda3/envs/teleop/include/python3.10
      -- Found Python libs: /home/unitree/miniconda3/envs/teleop/lib
      -- NLopt version 2.6.2
      -- Looking for dlfcn.h
      -- Looking for dlfcn.h - found
      -- Looking for getopt.h
      -- Looking for getopt.h - found
      -- Looking for unistd.h
      -- Looking for unistd.h - found
      -- Looking for string.h
      -- Looking for string.h - found
      -- Looking for strings.h
      -- Looking for strings.h - found
      -- Looking for inttypes.h
      -- Looking for inttypes.h - found
      -- Looking for memory.h
      -- Looking for memory.h - found
      -- Looking for stdlib.h
      -- Looking for stdlib.h - found
      -- Looking for stdint.h
      -- Looking for stdint.h - found
      -- Looking for time.h
      -- Looking for time.h - found
      -- Looking for sys/types.h
      -- Looking for sys/types.h - found
[... 248 lines cut ...]
```

### **Problem 2: Runtime performance**
After installation, teleop technically ran, but was unusable:
* Camera video stream had 10–30s latency
* Control commands ran at only 10–20 Hz

I saw similar lag on laptops used as Host PCs, while on a desktop everything was smooth. So it turned out xr_teleoperate is **veeeery CPU-sensitive**.

Profiling revealed that setting:

```bash
OMP_NUM_THREADS=1 python teleop_arms_and_hands.py --args
```

fixed the issue.

**Why it helps:**
* Libraries like OpenCV and PyTorch are highly optimized C++ code that aggressively use all available CPU cores by default.
* The Python side (responsible for streaming) couldn’t keep up under contention — memory bandwidth was saturated.
* By limiting HPC libraries to a single thread (OMP_NUM_THREADS=1), we freed up enough resources for Python code to handle data streaming in real time.

⸻

Hopefully this helps others who hit the same issues on G1's Orin NX or laptops.