unitreerobotics/xr_teleoperate #134: Black screen when trying to teleoperate G1EDU + RH56DFTP hand

amul1umt, 2025-08-29:
I have the network set up, I can ping PC2 and both hands' ip. G1 is directly connected to router via ethernet, and my PC as well directly connected. Meta quest 3 is connected to 5ghz wifi from the same network.

I have followed this https://support.unitree.com/home/en/G1_developer/inspire_ftp_dexterity_hand and this https://github.com/unitreerobotics/xr_teleoperate/issues/48 to set up my environment.

xr_teleoperate is set up on my pc. inspire_hand_ws from https://github.com/NaCl-1374/inspire_hand_ws is also set up on my PC.

I have copied image_server.py to PC2 and started the server. I am only using head camera. There are a number of camera IDs. I have commented out wrist camera related lines in the image_server.py, and changed the camera ID to check which one is correctly sending images to image_client. ID 2 in this case. All other /dev/video<id> return error - some how this one also shows monochrome stream (not sure what is wrong). Can anyone please let me know how to get this corrected to rgb stream?

Then, as recommended in [#48](https://github.com/unitreerobotics/xr_teleoperate/issues/48) by [@chaingee](https://github.com/chaingee) I updated "robot_hand_inspire.py", start `/inspire_hand_ws/inspire_hand_sdk$ python example/Headless_driver_double.py` and run t`eleop$ python teleop_hand_and_arm.py --arm=G1_29 --hand=inspire1`

I get the signal about modbus being connected from first command (see screenshot below), and the second command also runs to make connection. I hear some movement - the very first time i ran this, the hands opened up. And, have not done anything since then. Then i hit r, and go to my Vuer URL: https://192.168.123.101:8012/?ws=wss://192.168.123.101:8012

The screen appears dark. I am suspecting it is not able to get the right inputs from the camera or there is a camera ID mismatch. I have tried to dig in many places but I cannot find anything.. It appears the same in my meta quest 3 as well. Appreciate any help in getting past this issue

[image]

[image]

[image]

[image]

silencht, 2025-09-02:
please notice these code:
https://github.com/unitreerobotics/xr_teleoperate/blob/988431ef738b59edfd0dc2b8023ac2facd460b3d/teleop/teleop_hand_and_arm.py#L73-L93

amul1umt, 2025-09-02:
@silencht I have that part edited already to the right camera id :(
See below. I still get black screen. I understand I don't have wrist cameras connected, but as much I understand there are multiple cameras in the head mounted one, and I only get one of those to show up in in the image client.
`   # image client: img_config should be the same as the configuration in image_server.py (of Robot's development computing unit)
    if args.sim:
        img_config = {
            'fps': 30,
            'head_camera_type': 'opencv',
            'head_camera_image_shape': [480, 640],  # Head camera resolution
            'head_camera_id_numbers': [2],
            #'wrist_camera_type': 'opencv',
            #'wrist_camera_image_shape': [480, 640],  # Wrist camera resolution
            #'wrist_camera_id_numbers': [2, 4],
        }
    else:
        img_config = {
            'fps': 30,
            'head_camera_type': 'opencv',
            'head_camera_image_shape': [480, 1280],  # Head camera resolution
            'head_camera_id_numbers': [2],
            #'wrist_camera_type': 'opencv',
            #'wrist_camera_image_shape': [480, 640],  # Wrist camera resolution
            #'wrist_camera_id_numbers': [2, 4],
        }
`

amul1umt, 2025-09-02:
ok, one issue was related to image shape, so after fixing that I am able to control hands, but arm movement is not happening, and I still see the monochrome output from image_server. I think that maybe a realsense driver issue that is defaulting to infrared camera and should be fixable. but any suggestion on getting the arm movement correctly working will be appreciated.

bethalageetachandraraju, 2025-09-09:
i think, the issue you are facing is because of wrong camera ID, this help to solve the monochrome image
        'head_camera_type': 'realsense',
        'head_camera_image_shape': [480, 640],  # Head camera resolution
        'head_camera_id_numbers': ["************"],  # RealSense serial number
        # Comment out wrist camera to avoid resource conflict
        # 'wrist_camera_type': 'realsense',
        # 'wrist_camera_image_shape': [480, 640],  # Wrist camera resolution
        # 'wrist_camera_id_numbers': ["************"],  # Wrist RealSense camera