unitreerobotics/xr_teleoperate #139: H1 Dual Arm Teleoperatoin in --motion mode

brelkh, 2025-09-09:
Hi @silencht,

I am unable to run the arm teleoperation in motion mode for the H1_2. When the robot is in standing mode (L1 + up) or motion mode (R2 + x), I run the teleop script and before pressing "r", the arms already begin to jitter and vibrate. On the other hand, the teleoperation runs fine when the H1 is in debug mode (L2+R2) and I am using the meta quest 3 and Vuer V0.0.40.

I have read the [motion mode wiki](https://github.com/unitreerobotics/xr_teleoperate/wiki/%E2%80%90%E2%80%90motion-Mode) and I see that STEP 2 and STEP 3 in robot_arm.py are written for the G1. For example,
```
    def _ctrl_motor_state(self):
        # STEP 2
        self.msg.motor_cmd[G1_29_JointIndex.kNotUsedJoint0].q = 1.0;
```

Is the motion mode able to work on the H1_2 or would I need to make further modifications?

silencht, 2025-09-09:
`--motion` is only support g1. 

so, H1_2 need some modifications like g1. yes, about Modification steps, please refer to [motion mode wiki](https://github.com/unitreerobotics/xr_teleoperate/wiki/%E2%80%90%E2%80%90motion-Mode)

brelkh, 2025-09-09:
@silencht alright, thank you for the quick response!

PR #141 (2025-09-17): Add motion mode support for H1_2
Changes:
- Add motion_mode parameter to H1_2_ArmController constructor
- Use kTopicLowCommand_Motion publisher when motion_mode is enabled
- Add motion control logic in motor state control loop
- Add gradual weight transition during homing sequence for motion mode
- Update main teleop script to pass motion mode flag to H1_2_ArmController

Other Notes:
- This fixes #139
- Based on changes in #116 and advice from [motion mode wiki](https://github.com/unitreerobotics/xr_teleoperate/wiki/%E2%80%90%E2%80%90motion-Mode)
- Tested successfully on the H1_2
closed 2025-09-17 by silencht