unitreerobotics/xr_teleoperate #144: Quest only the grid could be seen

lllzheng, 2025-09-16:
I connect to the host via a wired connection using quest and enable the port service via adb
`adb devices 
 adb reverse tcp:8012 tcp:8012
`
Then, I ran python and accessing https://127.0.0.1:8012 on the quest side, only the grid could be seen, but the first-person perspective of the robot could not be opened. How can this be solved
```
sim_main.py \
--device cpu  \
--enable_cameras  \
--task $TASK \
--enable_dex3_dds \
--robot_type g129
``` 
and 
```
python teleop_hand_and_arm.py \
--xr-mode=controller \
--arm=G1_29 \
--ee=dex3 \
--sim \
--record
```

lllzheng, 2025-09-18:
I've solved this problem, but now I'm encountering a new one. When connecting with quest, `xr-mode=hand` can remotely operate G1 in isaacsim. However, when set to `xr-mode=controller`, the simulation G1 cannot be remotely operated with the quest controller. What's going on here

silencht, 2025-09-22:
hi.

so, how did you solve this problem?

The controller can only drive the dex1/gripper—did you perhaps select the wrong dex-hand model by mistake?

zhou442, 2025-09-28:
请问您是如何解决仅有网格的问题呢