unitreerobotics/xr_teleoperate #147: Resetting G1 to Default Position after Saving an Episode

KumoLiu, 2025-09-19:
**Description:**

I am encountering a problem with resetting G1 to its default position after saving an episode. In simulation mode, there is an automatic reset after each episode save. However, when teleoperating the real robot during AVP, there seems to be no reset function available.
https://github.com/unitreerobotics/xr_teleoperate/blob/main/teleop/teleop_hand_and_arm.py#L291

**Steps Taken:**

I attempted to use ctrl_dual_arm_go_home to move the robot to the default reset position, but it's not working as expected.
Questions:

- How can I reset G1 to the default position after saving an episode while teleoperating the real robot? Otherwise each episode, I can not ensure the start state the same.
- How can I modify the default resetting pose, such as adjusting the elbow to be slightly higher?

**Expected Behavior:**

The robot should reset to the default position after each episode save in both simulation and real robot teleoperation modes.
There should be an option to customize the default reset pose.
Any guidance or solutions to address these issues would be greatly appreciated. Thank you!

silencht, 2025-09-22:
Hi. 

> How can I reset G1 to the default position after saving an episode while teleoperating the real robot? Otherwise each episode, I can not ensure the start state the same.

There's no built-in "home" move after saving data during physical deployment—the wearer must manually return both arms to the start/end pose. We think only need the two poses to be roughly the same each run.

> How can I modify the default resetting pose, such as adjusting the elbow to be slightly higher?

maybe you could refer these code:
https://github.com/unitreerobotics/unitree_IL_lerobot/blob/7f9b3f40598cdbad9c9c13d0313c8b8ea790343e/unitree_lerobot/eval_robot/eval_g1.py#L84-L98
or 
https://github.com/silencht/televuer/blob/34f4475fca12166d2c52f2469385a851f614fd4e/src/televuer/tv_wrapper.py#L128-L139

KumoLiu, 2025-09-22:
Hi @silencht, thanks for the quick response and share the link, will try it later.

I have a quick question: could you please guide me on how to modify the meta controller to ensure compatibility with Dex3? Specifically, I'm looking to set various combinations of grasp positions and enable the controller to effectively manage Dex3. Thanks in advance!